Klipper3d/klipper #1: Core X/Y Support

TinHead, 2017-03-12:
Hello Kevin,

I'd like to give Klipper a try on my core xy build but I have trouble understanding the movement code which I would need to adapt on the python side.

Is there any chance I could get a quick example to get me on the right track?

Thank you,
Razvan

KevinOConnor, 2017-03-13:
There have been several requests for corexy in the last couple of weeks.  I uploaded a preliminary development branch "work-corexy-20170306" for corexy.  The code is currently very rough and likely wont work great.  If you're feeling brave enough to test, then please provide feedback on the results.  (Be sure to select "kinematic: corexy" in the config.)  For the first few tests, be ready to issue an M112 (emergency stop) or pull the power if things don't go well.

TinHead, 2017-03-13:
Perfect! I'll check it out and get back to you.

Thanks,
Razvan

TinHead, 2017-03-13:
Oookeeey so I did install the firmware on my olde Arduino Mega 1280, figured out baud rate is 115200 on that one and fired up the daemon. 

I hit the error below ... not sure where to look for oid:

INFO:root:Starting Klippy...
INFO:root:Git version: 'v0.3.0-100-gb07bdd1'
INFO:root:CPU: 4 core Intel(R) Atom(TM) CPU D525   @ 1.80GHz
INFO:root:Building C code module
INFO:root:Start printer at Mon Mar 13 20:26:56 2017 (1489429616.5 387921.5)
INFO:root:===== Config file =====
INFO:root:[stepper_x]
INFO:root:step_pin = ar54
INFO:root:dir_pin = ar55
INFO:root:enable_pin = !ar38
INFO:root:step_distance = .01
INFO:root:endstop_pin = ^ar2
INFO:root:homing_speed = 50.0
INFO:root:position_min = 0
INFO:root:position_endstop = 0
INFO:root:position_max = 200
INFO:root:
INFO:root:[stepper_y]
INFO:root:step_pin = ar60
INFO:root:dir_pin = ar61
INFO:root:enable_pin = !ar56
INFO:root:step_distance = .01
INFO:root:endstop_pin = ^ar15
INFO:root:homing_speed = 50.0
INFO:root:position_min = 0
INFO:root:position_endstop = 0
INFO:root:position_max = 200
INFO:root:
INFO:root:[stepper_z]
INFO:root:step_pin = ar46
INFO:root:dir_pin = ar48
INFO:root:enable_pin = !ar62
INFO:root:step_distance = .01
INFO:root:endstop_pin = ^ar19
INFO:root:position_min = 0.1
INFO:root:position_endstop = 0.5
INFO:root:position_max = 200
INFO:root:
INFO:root:[extruder]
INFO:root:step_pin = ar26
INFO:root:dir_pin = ar28
INFO:root:enable_pin = !ar24
INFO:root:step_distance = .0022
INFO:root:nozzle_diameter = 0.400
INFO:root:filament_diameter = 1.750
INFO:root:heater_pin = ar10
INFO:root:sensor_type = ATC Semitec 104GT-2
INFO:root:sensor_pin = analog13
INFO:root:control = pid
INFO:root:pid_kp = 22.2
INFO:root:pid_ki = 1.08
INFO:root:pid_kd = 114
INFO:root:min_temp = 0
INFO:root:max_temp = 250
INFO:root:
INFO:root:[heater_bed]
INFO:root:heater_pin = ar8
INFO:root:sensor_type = EPCOS 100K B57560G104F
INFO:root:sensor_pin = analog14
INFO:root:control = watermark
INFO:root:min_temp = 0
INFO:root:max_temp = 130
INFO:root:
INFO:root:[fan]
INFO:root:pin = ar9
INFO:root:
[... 76 lines cut ...]
INFO:root:Receive: 6 387922.431518 387922.424759 49: seq: 19, identify_response offset=1040 data="\x02\xb3GCl\xf5\xc4\x0c+\xf8\xa4.\x9e\xea\xca'\x0b\xa3\xc8\xe3\x88\xc1\x83\xd0\x17\x1f\xc49\xee\x81\xd7\x08,&\n\xbe\x8a;Li"
INFO:root:Receive: 7 387922.438522 387922.431736 49: seq: 1a, identify_response offset=1080 data='\x0c]\xafE8Hb\x0f\xb7\xb69?\xf1\xe7DA\x99\x88\xf3\x9dM.\x9e\x02\x15\xf8.)(;\xda\x8a\xffFR\xce\xbe=\\\x1a'
INFO:root:Receive: 8 387922.445528 387922.438738 49: seq: 1b, identify_response offset=1120 data='\x8bU:\x14\x1f\rD\x9f?j\xfd"I\t\x1c\xb2\xa1\n\xc8A:\x91\xe8\xfa&\x0eS\x83N\xa2,\xe4\x18\xce\x86d:\xd0w\x88'
INFO:root:Receive: 9 387922.452522 387922.445757 49: seq: 1c, identify_response offset=1160 data="CI\xb9\xf1g\x05\xdb\xe8fC\xa3'\xa9\x842\x8f\x1c\x87A\x0b\x14c\x1b\x1e\xf6C\xc4\x15i9[\x98\x80\xbe\x02\x82%\xe5\xb4P"
INFO:root:Receive: 10 387922.459527 387922.452743 49: seq: 1d, identify_response offset=1200 data='\x81\xee\xd0\x89\xd0\x95\x10\xa1c\xb3\x13\x9e\xa2~(\xbc\xe9\x06\xaa\x86\xc7Y\rz(\r\x98\xd1\xc3\xf9\x1f_(\xa1\x9f\x00\x03\x8dh!'
INFO:root:Receive: 11 387922.466521 387922.459742 49: seq: 1e, identify_response offset=1240 data='\xc2\x08\xdce\xcf\x8ag\xd9)\xcf\xb2\xd3\xe6.;\xbb\xabk~\x9ag\xfc,+x!\xf8j\x91\xf1\xec\xd4\xca;\xb97\x9d\xbf\xbaR'
INFO:root:Receive: 12 387922.473518 387922.466785 49: seq: 1f, identify_response offset=1280 data='\xe3\xe1e\xe3\xe6\xe5\xf5\xab\x8b\xd7\xe2\xa7\x8b\xb7/\x90\x88\xf3E\x9ee\x88\x81;\x9d\xb8\xba\xf8\x83\x84\x19\xf48\xde\xed.\xdf\x927\xae'
INFO:root:Receive: 13 387922.480521 387922.473779 49: seq: 10, identify_response offset=1320 data='U\x8d\xe4\xf9\x8ap\x97\xaf\x7f\xbb\xfcE\xfc|\xfd\xf2\rA\x97\x19\xfd\xd1\xfd\xf2\xf6\xe2\xf6F\xdc\xbc\xbd\xbay\x03\xfc7/q=_'
INFO:root:Receive: 14 387922.484524 387922.480741 19: seq: 11, identify_response offset=1360 data=',g\x9f>\xfd\x03f\xf2J\xfa'
INFO:root:Receive: 15 387922.488524 387922.484727 9: seq: 12, identify_response offset=1370 data=''
INFO:root:Receive: 16 387922.503525 387922.500939 12: seq: 13, uptime high=10 clock=2143533111
INFO:root:Receive: 17 387922.559525 387922.556263 12: seq: 14, status clock=2144417527 status=1
INFO:root:Receive: 18 387922.560546 387922.556784 9: seq: 15, config is_config=0 crc=0 move_count=0
INFO:root:Receive: 19 387922.610534 387922.607865 7: seq: 16, is_shutdown static_string_id=9
ERROR:root:MCU error during connect
Traceback (most recent call last):
  File "klippy/klippy.py", line 172, in connect
    self.mcu.connect()
  File "/home/razvan/klipper/klippy/mcu.py", line 441, in connect
    self._send_config()
  File "/home/razvan/klipper/klippy/mcu.py", line 532, in _send_config
    self._shutdown_msg,))
error: Firmware error during config: Invalid oid type

I think it might be related to the mega 1280 vs 2650 I'm waiting on one of those...

KevinOConnor, 2017-03-13:
On Mon, Mar 13, 2017 at 11:32:34AM -0700, TinHead wrote:
> Oookeeey so I did install the firmware on my olde Arduino Mega 1280, figured out baud rate is 115200 on that one and fired up the daemon. 
> 
> I hit the error below ... not sure where to look for oid:
[...]
> INFO:root:Receive: 19 387922.610534 387922.607865 7: seq: 16, is_shutdown static_string_id=9
> ERROR:root:MCU error during connect
> Traceback (most recent call last):
>   File "klippy/klippy.py", line [...]

Thanks, there was an ordering problem with the initial config messages
which should be fixed now.  Can you pull the branch again and retry?

> I think it might be related to the mega 1280 vs 2650 I'm waiting on one of those...

There's no gain to the 2560 vs the 1280 - indeed the 1280 is probably
faster.

-Kevin

TinHead, 2017-03-13:
OK still oid error: 

`python klippy/klippy.py config/example-corexy.cfg 
INFO:root:Starting Klippy...
INFO:root:Git version: 'v0.3.0-102-g150e51e-dirty'
INFO:root:CPU: 4 core Intel(R) Atom(TM) CPU D525   @ 1.80GHz
INFO:root:Start printer at Mon Mar 13 21:05:24 2017 (1489431924.6 390229.7)
INFO:root:===== Config file =====
INFO:root:[stepper_x]
INFO:root:step_pin = ar54
INFO:root:dir_pin = ar55
INFO:root:enable_pin = !ar38
INFO:root:step_distance = .01
INFO:root:endstop_pin = ^ar2
INFO:root:homing_speed = 50.0
INFO:root:position_min = 0
INFO:root:position_endstop = 0
INFO:root:position_max = 200
INFO:root:
INFO:root:[stepper_y]
INFO:root:step_pin = ar60
INFO:root:dir_pin = ar61
INFO:root:enable_pin = !ar56
INFO:root:step_distance = .01
INFO:root:endstop_pin = ^ar15
INFO:root:homing_speed = 50.0
INFO:root:position_min = 0
INFO:root:position_endstop = 0
INFO:root:position_max = 200
INFO:root:
INFO:root:[stepper_z]
INFO:root:step_pin = ar46
INFO:root:dir_pin = ar48
INFO:root:enable_pin = !ar62
INFO:root:step_distance = .01
INFO:root:endstop_pin = ^ar19
INFO:root:position_min = 0.1
INFO:root:position_endstop = 0.5
INFO:root:position_max = 200
INFO:root:
INFO:root:[extruder]
INFO:root:step_pin = ar26
INFO:root:dir_pin = ar28
INFO:root:enable_pin = !ar24
INFO:root:step_distance = .0022
INFO:root:nozzle_diameter = 0.400
INFO:root:filament_diameter = 1.750
INFO:root:heater_pin = ar10
INFO:root:sensor_type = ATC Semitec 104GT-2
INFO:root:sensor_pin = analog13
INFO:root:control = pid
INFO:root:pid_kp = 22.2
INFO:root:pid_ki = 1.08
INFO:root:pid_kd = 114
INFO:root:min_temp = 0
INFO:root:max_temp = 250
INFO:root:
INFO:root:[heater_bed]
INFO:root:heater_pin = ar8
INFO:root:sensor_type = EPCOS 100K B57560G104F
INFO:root:sensor_pin = analog14
INFO:root:control = watermark
INFO:root:min_temp = 0
INFO:root:max_temp = 130
INFO:root:
INFO:root:[fan]
INFO:root:pin = ar9
INFO:root:
INFO:root:[mcu]
INFO:root:serial = /dev/ttyUSB0
[... 73 lines cut ...]
INFO:root:Receive: 6 390230.560617 390230.553838 49: seq: 1f, identify_response offset=1040 data="\x02\xb3GCl\xf5\xc4\x0c+\xf8\xa4.\x9e\xea\xca'\x0b\xa3\xc8\xe3\x88\xc1\x83\xd0\x17\x1f\xc49\xee\x81\xd7\x08,&\n\xbe\x8a;Li"
INFO:root:Receive: 7 390230.567625 390230.560837 49: seq: 10, identify_response offset=1080 data='\x0c]\xafE8Hb\x0f\xb7\xb69?\xf1\xe7DA\x99\x88\xf3\x9dM.\x9e\x02\x15\xf8.)(;\xda\x8a\xffFR\xce\xbe=\\\x1a'
INFO:root:Receive: 8 390230.574617 390230.567904 49: seq: 11, identify_response offset=1120 data='\x8bU:\x14\x1f\rD\x9f?j\xfd"I\t\x1c\xb2\xa1\n\xc8A:\x91\xe8\xfa&\x0eS\x83N\xa2,\xe4\x18\xce\x86d:\xd0w\x88'
INFO:root:Receive: 9 390230.581621 390230.574884 49: seq: 12, identify_response offset=1160 data="CI\xb9\xf1g\x05\xdb\xe8fC\xa3'\xa9\x842\x8f\x1c\x87A\x0b\x14c\x1b\x1e\xf6C\xc4\x15i9[\x98\x80\xbe\x02\x82%\xe5\xb4P"
INFO:root:Receive: 10 390230.588615 390230.581849 49: seq: 13, identify_response offset=1200 data='\x81\xee\xd0\x89\xd0\x95\x10\xa1c\xb3\x13\x9e\xa2~(\xbc\xe9\x06\xaa\x86\xc7Y\rz(\r\x98\xd1\xc3\xf9\x1f_(\xa1\x9f\x00\x03\x8dh!'
INFO:root:Receive: 11 390230.595624 390230.588847 49: seq: 14, identify_response offset=1240 data='\xc2\x08\xdce\xcf\x8ag\xd9)\xcf\xb2\xd3\xe6.;\xbb\xabk~\x9ag\xfc,+x!\xf8j\x91\xf1\xec\xd4\xca;\xb97\x9d\xbf\xbaR'
INFO:root:Receive: 12 390230.602620 390230.595899 49: seq: 15, identify_response offset=1280 data='\xe3\xe1e\xe3\xe6\xe5\xf5\xab\x8b\xd7\xe2\xa7\x8b\xb7/\x90\x88\xf3E\x9ee\x88\x81;\x9d\xb8\xba\xf8\x83\x84\x19\xf48\xde\xed.\xdf\x927\xae'
INFO:root:Receive: 13 390230.609613 390230.602847 49: seq: 16, identify_response offset=1320 data='U\x8d\xe4\xf9\x8ap\x97\xaf\x7f\xbb\xfcE\xfc|\xfd\xf2\rA\x97\x19\xfd\xd1\xfd\xf2\xf6\xe2\xf6F\xdc\xbc\xbd\xbay\x03\xfc7/q=_'
INFO:root:Receive: 14 390230.613618 390230.609841 19: seq: 17, identify_response offset=1360 data=',g\x9f>\xfd\x03f\xf2J\xfa'
INFO:root:Receive: 15 390230.617616 390230.613826 9: seq: 18, identify_response offset=1370 data=''
INFO:root:Receive: 16 390230.635620 390230.632404 12: seq: 19, uptime high=19 clock=419717909
INFO:root:Receive: 17 390230.690622 390230.687516 12: seq: 1a, status clock=420599797 status=1
INFO:root:Receive: 18 390230.691615 390230.688037 9: seq: 1b, config is_config=0 crc=0 move_count=0
INFO:root:Receive: 19 390230.741633 390230.739019 7: seq: 1c, is_shutdown static_string_id=9
ERROR:root:MCU error during connect
Traceback (most recent call last):
  File "klippy/klippy.py", line 172, in connect
    self.mcu.connect()
  File "/home/razvan/klipper/klippy/mcu.py", line 444, in connect
    self._send_config()
  File "/home/razvan/klipper/klippy/mcu.py", line 535, in _send_config
    self._shutdown_msg,))
error: Firmware error during config: Invalid oid type
`
looks like it does not like analog14 if I read this correct ...

KevinOConnor, 2017-03-13:
On Mon, Mar 13, 2017 at 12:07:46PM -0700, TinHead wrote:
> OK still oid error: 
[...]
> INFO:root:Receive: 16 390230.635620 390230.632404 12: seq: 19, uptime high=19 clock=419717909
> INFO:root:Receive: 17 390230.690622 390230.687516 12: seq: 1a, status clock=420599797 status=1 [...]

Can you confirm you ran FIRMWARE_RESTART after updating the host
software?  At first glance, it looks like the firmware was still in an
error state from the last attempt.

-Kevin

TinHead, 2017-03-13:
OK you got me on that :) 
Reset of the Mega did it :)

TinHead, 2017-03-13:
I can confirm comms work fine now got octoprint talking to the daemon and daemon to the mega.

Commands work fine too.

Going to test movement next I have to fix some wiring first.

TinHead, 2017-03-13:
Quick question: does klipper support dual steppers on z with separate drivers?

KevinOConnor, 2017-03-13:
On Mon, Mar 13, 2017 at 03:17:37PM -0700, TinHead wrote:
> Quick question: does klipper support dual steppers on z with separate drivers?

Klipper doesn't currently have support for multiple extruders.  I
don't think it would be difficult to add, but I don't have the
hardware and there doesn't seem to be a standard for multiple
extruders.

-Kevin

TinHead, 2017-03-13:
I did not mean multiple extruders, I mean dual stepper motors for an axis driven at the same time using separate drivers, sort of port mirroring.

KevinOConnor, 2017-03-13:
On Mon, Mar 13, 2017 at 03:35:54PM -0700, TinHead wrote:
> I did not mean multiple extruders, I mean dual stepper motors for an axis driven at the same time using separate drivers, sort of port mirroring.

Ah, sorry.  It's not currently supported, but it is not difficult to
add.  Out of curiosity though, why use two drivers instead of just
wiring both motors to a single driver?

-Kevin

TinHead, 2017-03-13:
You get to control the current per motor. Also both motors and driver run cooler then when wired in parallel.

TinHead, 2017-03-15:
Update: 
I'm blocked by hardware for now, once I get that solved I should be able to do some more testing.

MrJulz, 2017-03-16:
I can confirm the CoreXY kinematics are working for me.

TinHead, 2017-03-19:
Hi, 

Finally figured out my hardware issues, got x/y moving and homing. One thing I have noticed is that the following sequence fails atm:

1. Home x/y
2. Move x/y to random location
3. Home again => results in hitting x switch and bail with firmare error 

To get it back to work I have to issue clear_shutdown..

I still have to get Z going so that axis is not homed in this case, could it be related?

I'll get back with more once I get the Z going.

KevinOConnor, 2017-03-19:
On Sun, Mar 19, 2017 at 01:47:02AM -0700, TinHead wrote:
> Hi,
>
> Finally figured out my hardware issues, got x/y moving and homing. One thing I have noticed is that the following sequence fails atm:
>
> 1. Home x/y
> 2. Move x/y to random location
> 3. Home again [...]

What version are you on?  There were several fixes and your
description above sounds like one of the initial bugs that were fixed
(latest is 1e3f245f).

-Kevin

TinHead, 2017-03-19:
Ok seems I have missed some commits there gonna pull and let you know ....

TinHead, 2017-03-20:
Issue still happens, I'm on latest:
`git pull
remote: Counting objects: 64, done.
remote: Total 64 (delta 33), reused 33 (delta 33), pack-reused 31
Unpacking objects: 100% (64/64), done.
From https://github.com/KevinOConnor/klipper
   150e51e..1e3f245  work-corexy-20170306 -> origin/work-corexy-20170306
   0f2478b..5ff2d5a  master               -> origin/master
Updating 150e51e..1e3f245
Fast-forward
 klippy/corexy.py      |  6 +++-
 klippy/gcode.py       | 86 ++++++++++++++++++++++++++++++++++-----------------
 klippy/klippy.py      |  2 ++
 klippy/mcu.py         |  6 ++--
 klippy/serialhdl.py   |  2 +-
 klippy/stepcompress.c |  8 +++--
 klippy/stepper.py     | 14 ++++++---
 klippy/toolhead.py    |  9 ++++--
 8 files changed, 90 insertions(+), 43 deletions(-)
`

Here is the terminal from Octoprint:
`Recv: !! Must home axis first: 10.000 0.000 0.000 [0.000]
Recv: ok
Recv: ok
Send: G91
Send: G28 X0 Y0
Recv: ok
Send: G90
Send: M105
Recv: ok
Recv: ok
Recv: ok T:23.5 /0.0 B:23.2 /0.0
Send: M105
Recv: ok T:22.8 /0.0 B:23.6 /0.0
Send: G91
Send: G1 X10 F6000
Recv: ok
Send: G90
Recv: ok
Recv: ok
Send: G91
Send: G1 X10 F6000
Send: G90
Recv: ok
Recv: ok
Recv: ok
Send: G91
Send: G1 X10 F6000
Recv: ok
Send: G90
Recv: ok
Recv: ok
Send: G91
Send: G1 X10 F6000
Recv: ok
Send: G90
Recv: ok
Recv: ok
Send: G91
Send: G1 X10 F6000
Send: G90
Recv: ok
Recv: ok
Recv: ok
Send: M105
Recv: ok T:23.4 /0.0 B:23.4 /0.0
Send: G91
Send: G1 Y10 F6000
Send: G90
[... 84 lines cut ...]
Read 395306.820259: 'G90\n'
Read 395307.562189: 'G91\n'
Read 395307.568688: 'G1 X10 F6000\n'
Read 395307.571273: 'G90\n'
Read 395308.361218: 'G91\n'
Read 395308.368412: 'G1 X10 F6000\n'
Read 395308.371651: 'G90\n'
Read 395309.565106: 'M105\n'
Read 395310.550810: 'G91\n'
Read 395310.558436: 'G1 Y10 F6000\n'
Read 395310.559735: 'G90\n'
Read 395311.327508: 'G91\n'
Read 395311.328364: 'G1 Y10 F6000\n'
Read 395311.331484: 'G90\n'
Read 395312.109319: 'G91\n'
Read 395312.110176: 'G1 Y10 F6000\n'
Read 395312.116138: 'G90\n'
Read 395312.917586: 'G91\n'
Read 395312.919320: 'G1 Y10 F6000\n'
Read 395312.922268: 'G90\n'
Read 395314.566786: 'M105\n'
Read 395315.608195: 'G91\n'
Read 395315.611707: 'G28 X0 Y0\n'
Read 395315.622809: 'G90\n'
`

KevinOConnor, 2017-03-20:
On Mon, Mar 20, 2017 at 02:20:07AM -0700, TinHead wrote:
> Issue still happens, I'm on latest:

Thanks for testing.  Can you attach the full klipper.log file?  If
it's large, can you gzip it before attaching.

-Kevin

TinHead, 2017-03-20:
You are welcome I'll test further, got tired of the mainstream firmwares:)

Unfortunatelly I cannot provide the full log right now I will tomorrow, hope it's still there ... Else I'll replicate it again. I don't think it is a switch issue...

TinHead, 2017-03-30:
OK back with an update:
- I can no longer replicate homing issue so I guess it was an issue on my part
- everything seems to work I actually managed to print some tests -- looking good :+1: 
- sometimes I've seen errors: Recv: // Firmware shutdown: adc out of range however not sure what to correlate with ... did not happen during my test prints so not sure ... will get back to you if I figure it out 

All in all things are pretty smooth I'll stick with Klipper at least for now :)

TinHead, 2017-03-30:
I think we can close... maybe merge to master?

KevinOConnor, 2017-03-30:
Lets leave this one open until it is merged.  I need to rebase the changes and I'll likely ask you to retest with the merged version.  I hope to get to that in a few days.

The "adc out of range" indicates the firmware detected one of the thermistors (either bed or extruder) had a temperature outside of the config's min_temp/max_temp.  I recommend you investigate that.

TinHead, 2017-03-30:
OK Let me know I'll test and get back to you, if any new issue is found I'll open a new ticket once in master I hate handling multiple issues in the same tread :)

KevinOConnor, 2017-03-31:
On Thu, Mar 30, 2017 at 10:08:14AM -0700, TinHead wrote:
> OK Let me know I'll test and get back to you, if any new issue is
> found I'll open a new ticket once in master I hate handling multiple
> issues in the same tread :)

Thanks.  Can you pull the corexy branch again and verify everything
still works?  This is rebase'd to master (so you'll need to use "git
fetch; git checkout work-corexy-20170306").

Merging with master pulled in some important bug fixes, so you'll want
to use this code regardless.

Once I have confirmation on the recent changes I'll commit the corexy
code into master.

-Kevin

KevinOConnor, 2017-04-03:
I received that basic functionality works with the merged corexy, so I've committed this to master.  Thanks for testing.

closed 2017-04-03 by KevinOConnor