Klipper3d/klipper #4: i2c support and multiple mcu's

Kenzu, 2017-04-21:
Hey

Would it be posible to use i2c for data transfer between python code and mcu?
And maybe then and support for more mcu's on one bus. Ex an mcu for z and y steppers, then an mcu for x and e stepper + heater/temp.
Then we only have to use 6 wires for all data/power on a 3d printer.
12-24v+, gnd, sda, scl, 3.3/5v+ and lowlevel gnd.

Nice work by the way :-) really like to idea to move things to a powerfull armv7

Jesper

KevinOConnor, 2017-04-22:
On Fri, Apr 21, 2017 at 09:04:09AM -0700, Kenzu wrote:
> Would it be posible to use i2c for data transfer between python code and mcu?
> And maybe then and support for more mcu's on one bus. Ex an mcu for z and y steppers, then an mcu for x and e stepper + [...]

Thanks.

It should be possible to support multiple MCUs from a single host.
There is some work that would need to be done - mostly calculating
clock drift between micro-controllers and synchronizing them
accordingly.

It should also be possible to use i2c for data transfers, but that
code would need to be written.  For new hardware, however, I suspect
using a USB bus, rs-485, or even SPI would be easier to implement and
be more robust.

-Kevin

KevinOConnor, 2017-10-06:
Support for multiple mcus is now present.  There is no immediate support for i2c, but that would need to be hardware specific anyway.

closed 2017-10-06 by KevinOConnor