Klipper3d/klipper #5: Exception in flush handler

flaminggoat, 2017-04-22:
Hi, I've been trying klipper on my delta printer but it often stops halfway through a print. I have attached the relevant part of the log.
[log.txt](https://github.com/KevinOConnor/klipper/files/949046/log.txt)

KevinOConnor, 2017-04-22:
Thanks for reporting. Can you post the full log along with the gcode file in use at the time of the error?

-Kevin

flaminggoat, 2017-04-22:
Full log, error around line 2000
[klippy.txt](https://github.com/KevinOConnor/klipper/files/949056/klippy.txt)
Gcode
[bearing5.txt](https://github.com/KevinOConnor/klipper/files/949058/bearing5.txt)
Config
[printer.txt](https://github.com/KevinOConnor/klipper/files/949059/printer.txt)

KevinOConnor, 2017-04-22:
On Sat, Apr 22, 2017 at 08:07:08AM -0700, Theo Hussey wrote:
> Full log, error around line 2000
> [klippy.txt](https://github.com/KevinOConnor/klipper/files/949056/klippy.txt)
> Gcode
> [bearing5.txt](https://github.com/KevinOConnor/klipper/files/949058/bearing5.txt)
> Config
> [printer.txt](https://github.com/KevinOConnor/klipper/files/949059/printer.txt)

It looks like there are two separate issues here - the first is an
error from the firmware (Message encode error) and that seems to
precipitate an error in the host (Internal error in stepcompress).

The firmware error is surprising.  Can you provide the contents of the
firmware build ( tar cfz ~/klipper-out.tgz ~/klipper/out/ )?

-Kevin

flaminggoat, 2017-04-22:
I saw message encode error in an earlier print that failed in the octoprint log, it scrolled off the top before i was able to copy it. Firmware build attached.
[out.zip](https://github.com/KevinOConnor/klipper/files/949150/out.zip)

KevinOConnor, 2017-04-22:
On Sat, Apr 22, 2017 at 10:40:04AM -0700, Theo Hussey wrote:
> I saw message encode error in an earlier print that failed in the octoprint log, it scrolled off the top before i was able to copy it. Firmware build attached.
> [out.zip](https://github.com/KevinOConnor/klipper/files/949150/out.zip)

Could you try reproducing with some additional debugging?

The steps are:

git fetch
git checkout origin/work-encode-error-20170422
make clean
make
make flash FLASH_DEVICE=/dev/ttyACM0

Then post the full log if an error occurs again.

-Kevin

KevinOConnor, 2017-04-22:
On Sat, Apr 22, 2017 at 04:08:35PM -0400, Kevin O'Connor wrote:
> On Sat, Apr 22, 2017 at 10:40:04AM -0700, Theo Hussey wrote:
>
> Could you try reproducing with some additional debugging?
>
> The steps are:
>
> git fetch
> git checkout origin/work-encode-error-20170422
> make [...]

Unfortunately, that debugging code was incorrect.  I've pushed up a
new version - the steps to use it are the same as above (git should
report version 4c02310781b3dd4dc83baa2b080b1e026e3e2008) now.

-Kevin

flaminggoat, 2017-04-23:
It stopped in a print again but i think it is a different error.
[image]

Tahutipai, 2017-04-23:
Are you running Klipper on a RPi or some other hardware?
Do you have a good quality well shielded USB cable?

I had the same error previously, while using a USB extension cable.  I haven't seen the error since removing the extension cable. Could be coincidence, though I have not seen that error since.

KevinOConnor, 2017-04-23:
On Sat, Apr 22, 2017 at 05:43:20PM -0700, Theo Hussey wrote:
> It stopped in a print again but i think it is a different error.
> [image]

That is indicative of a hang in the output code - trying to transmit
debug information is painful because the error is being reported in
the transmit code itself.

My best guess is that your setup (which I think is an RPi 1?) is
causing the comms to be very bursty and that may be triggering some
previously undiscovered defect.  The code definitely should not be
responding this way.

If you want to run another test - grab the latest code from the
work-encode-error-20170422 and send the log.  (The latest code on that
branch should be more resilient with its debugging.)

Separately, I'll try to setup a comms stress test to see if I can
reproduce locally.

-Kevin

KevinOConnor, 2017-04-23:
On Sat, Apr 22, 2017 at 05:50:55PM -0700, Tahutipai wrote:
> Are you running Klipper on a RPi or some other hardware?
> Do you have a good quality well shielded USB cable?
>
> I had the same error previously, while using a USB extension cable.  I haven't seen the [...]

If you have a way of reproducing the error, then that would be
helpful.  Grab the latest code on the work-encode-error-20170422 and
send the log if the error occurs.  The error looks to be a software
issue, so it will need to be tracked down.

-Kevin

KevinOConnor, 2017-04-25:
On Sat, Apr 22, 2017 at 09:40:34PM -0400, Kevin O'Connor wrote:
> Separately, I'll try to setup a comms stress test to see if I can
> reproduce locally.

I have tried several simulations in an attempt to reproduce the issue.
Unfortunately, I am unable to reproduce any abnormal events.

Further debugging information would help - best would be to grab the
latest code from the work-encode-error-20170422 branch (same commands
as above) and send the klippy.log and out/klipper.elf files if the
problem occurs.

-Kevin

KevinOConnor, 2017-07-01:
The command has change significantly since this bug report.  If there is still errors after updating to the latest code, then please re-open this issue.

closed 2017-07-01 by KevinOConnor