Klipper3d/klipper #6: any chance for adding multiple extruders

deafboxer, 2017-04-25:
With the host on the RPi, it seems like adding multiple extruders would be a simple thing (coming from a guy whose coding limits are minimal).

Matt

Tahutipai, 2017-04-25:
Hi Matt,

Actually this is under process, would you be interested as a beta tester?
Would you mind sharing what type of multiple extruder you have, the mechanics of how it works? (if it's off-the-shelf hardware, a link would be great)

deafboxer, 2017-04-25:
Absolutely, I've spent months will no progress trying to get a second
extruder added to a modded Press i3 clone, Hictop.

Short back story, my roommate bought a HICTOP KIT (single extruder, heated
and fan as outputs all on a 20 A power supply running an incorrectly
configured marlin) two years ago, assembled it, then had no idea what to do
next. He figured it was like a plug and play device. I moved in, being a
hobbyist mechanical/electrical engineer, spotted several, several flaws in
the design. Rebuilt his kit from scratch and realized you pay for what you
get for. I got the printer working after making upgrades to the z axis to
use lead screws, added a second power supply to handle the heated bed.
Built a metal case to allow room for the power supplies to sit underneath,
Insulated the under side of the bed to keep the heat up top. It worked but
with Marlin but the results were less then impressive, and slow. I printed
out a dual extruder holder for officall all metal original hexagon RepRap
J-heads. I got Octopi all setup, firmware loaded on my ramps 1.4, they
connect and there is no issues there. Went to edit the configuration file
for my specific build and to my suprise there was no option to set up a
dual extruder. After the extensive work I did in rebuiling the whole
printer, I would hate to resort back to using a Marlin branch,
MarlinKimbra, to get both extruders working.

As far as mechanics go, x, y are belt driven with tension tightens. Z
axises are powered through one driver using leaded screws. I am off
boarding the heat bed to either use a Moslet to trigger the heat bed from a
second 15A power supply, or even disable it in firmware and build an
Arduino to control, set its temperature independent of slicer software.

If dual extruders are coming, I am more then happy to assist speed up that
process and we stick with Klipper limited to one for the time being. My
Python is iffy at best, but Arduino code I get, so software wise I don't
know how much help I could be.

Sorry for long response, being thorough.

Matt

Tahutipai, 2017-04-25:
Thanks, thorough is good!
Just to clarify, with so many variations out there: Do you have two separate hotends(nozzles), and a dedicated stepper motor for each hotend, or are some of these components shared? 

Assuming you have two nozzles, are they fixed in position (same height), or does one of them move so it can be below the other one?

e.g here is a video of what Marlin calls a "switching_extruder", which has two hotends sharing one stepper motor, and a servo that rotates the hotend's so always one is lower than the other.
https://youtu.be/9aekOiyL1qY?t=1m1s

P.S Perhaps inadvertently, I think you just coined what I shall acquire as my favourite expression of the month: "You pay for what you get."  Can be very true, thanks for that :-)

deafboxer, 2017-04-26:
I've raised that servo video, and I know I could figure out how to get 2-8
extruders working on marlin using relays and other similar solutions. My
interest in klipper is in speed/printing quality. Most people who read your
project summary don't understand algorithms, I do and understand off
loading these cpu intensive algorithms an 8, even 32 bit, processor would
be overwhelmed by to a raspberry pi which then just tells ramps what to do
has a huge advantage.

I have to reprap ESD V6 hotels, with bowden extruders Both my hotends are
same height, only offset by the X axis. My first attempt at using marlin
for dual extruders resulted in offset in steps per minute per each extruder
motor. Advanced marlin configuration can be overwhelming which let me to
see alt firmwares, hence, klipper. I'm going to finish building the
mechanical side of the machine first and hopefully you will have something
I can test for you.

Just an upgrade of what I am going to my PrusaShit stock kit to make it
compete with the big boys.

-as mentioned, z axis lead screws in place of metal rods
-printer bed top thinly coated with silicone silicone (tail pipe sealer),
topped with high temp black silicone, thinly, just enough to get the PEI to
adhere.
-printer bed bottom coated with 8 mil of kapton cut to fit within 1/2 inch
of border. Several high temp silicone oven mats with a honey come design to
trap in air. 2 mil of Kapton on top, 1/4 sheet of neoprene glued with high
temp silicone on top of kapton,  then whole bottom surface covered in 1/4
thick silicone cover whole surface plus all edges. There is a cut out to
access the wiring. Almost all heat is kept on top, uses less amps.

Z lead screws to replace threaded rods. Top mounted bearings to keep z axis
slides straight.

Elevate platform made from t_slot steel bars to raise printer to put power
and ramps board below printer. Fans fitted with panty hose to keep dust
out.

All wires cut to size, routed, and recrimped.

Two identical Nema 17 motors to drive both extruders.

I hate bed leveling as the springs start to sag so future plans are to
install SMD two axis gyroscopes to the bed with a red, yellow and green led
attached. Red, that corner is too high, yellow, that corner is too low.
Green, level.

Would also like it implement laser distance sensors in place of ends tops
and to be used for autocalibaration/autobedleveling. (distance plans)

Where have you gotten in terms of adding multi extruder support? Demo code?
Huge problem with Marlin is that you can't specify individual steps per
extruder. You can only define one and that's what the 2nd extruder is
assigned, often with huge discrepancies despite being the same motor.

Tahutipai, 2017-04-26:
Interesting, thanks for the details, esp. your build oven design.
The smd gyro idea for levelling known-flat build plates is intriguing too.

FYI I'd be curious as to the success levels with both nozzles fixed at the same height - I've heard of this causing significant problems - print-job collisions - but I've never tried it personally so am not in a position to comment.  
For it to work, I would presume the Purge and Wipe functions would need to be near impeccable?

The current multi-extruder is in very early beta (you would be the first tester), although I am not currently aware if it lacking any function.  I will be submitting to Kevin for code review shortly (this week), and, once he has had a chance to review, and deems it ready for public testing, your testing would be greatly appreciated.  Yes individual extruder control is supported, thanks to Kevin's prior work.

KevinOConnor, 2017-04-26:
Hi.

On Tue, Apr 25, 2017 at 05:09:49PM -0700, deafboxer wrote:
> Where have you gotten in terms of adding multi extruder support? Demo code?

I personally don't have multi-extruder hardware, but I would like to
see support added to Klipper.  I don't think it will be an issue to
support your hardware (separate heaters, separate motors, seperate
temperature sensors).  Given Tahutipai's comments, I suspect it's a
week or two out from general testing.

I would recommend you verify single extruder support first.  We'll
want to make sure that is rock solid before testing multi.

> Huge problem with Marlin is that you can't specify individual steps per
> extruder. You can only define one and that's what the 2nd extruder is
> assigned, often with huge discrepancies despite being the same motor.

That should not be an issue in Klipper.

Cheers,
-Kevin

deafboxer, 2017-04-26:
Keep you posted

MDM63, 2017-05-21:
Here is another volunteer tester for multiple (2 at the moment) extruder's. I have a prusa3i style (Started as a Tevo Tarantula) printer with two ESD 6V hotends side by side on single carriage, with seperate heaters, temp sensors, etc. fed by two bowden style extruder's.

KevinOConnor, 2017-05-22:
The work-multiextrude-20170429 branch has test code for multiple extruders. It currently only supports fixed offset extruders. Testing feedback would be appreciated. To use it, checkout the branch ( git fetch ; git checkout work-multiextrude-20170429 ) and update the printer config file ( nano ~/printer.cfg ) with an additional "extruder1" config section. The example.cfg file has some documentation on the new parameters.

The distance offset between the first extruder and second extruder should either be specified in the slicer config or specified using an M206 command in the activate_gcode parameter of the extruder config. For example, if the second extruder has a 20mm offset on the X carriage then the config might look something like:
```
[extruder]
...
activate_gcode: M206 X0
...

[extruder1]
...
activate_gcode: M206 X20
...
```
Please let us know of any successes or failures.

MDM63, 2017-05-25:
Thanks Kevin!
I will give this a try tomorrow and will report back with the results. Is there somewhere a official discussion forum or such to ask questions, report issues, successes and post configurations?

KevinOConnor, 2017-05-25:
On Thu, May 25, 2017 at 12:07:51PM -0700, MDM63 wrote:
> Thanks Kevin!
> I will give this a try tomorrow and will report back with the results. Is there somewhere a official discussion forum or such to ask questions, report issues, successes and post configurations?
> 

Great!

I've been using the RepRap forum:
http://forums.reprap.org/read.php?147,667655

I wish I had a better alternative though, as that forum isn't
particularly easy to navigate and I don't think it gets much traffic.

-Kevin

MDM63, 2017-05-27:
HI. I had bunch on non firmware related issues, so I have not yet printed anything. I got finally today the printer working, but it still needs to be calibrated. I have registered to reprap forums, but it seams that the approval process takes it's time. I will post there in future, so we can keep this clutter free. I just wanted to let you know I am actively involved, in the time constraints that I have (ie. my other projects, family life etc.).

KevinOConnor, 2017-07-01:
I'm curious if there has been any tests (successful or failing) with the Klipper multiple extruder code.  The Klipper master branch now has almost all the patches for basic multiple extruder support.  I'd like to close this issue if all is working well.

deafboxer, 2017-07-01:
I hope to have printer up and running by end of 4 day weekend. Had to
rebuild single extruder printer to reprint multi extruder carrier. Haven't
had time to rebuild. For testing should I use master branch or
multiextruder branch?

MDM63, 2017-07-01:
I have managed ~10 (mostly calibration cubes, nothing big) dual extruder prints with out klipper related issues. I am sorry that I have not been able to do more prints, but I have been quite busy with other commitments and have had hardware issues with my printer. These should be more or less solved in couple of weeks, so I can commit more time to this project. As of now, IMHO this issue can be closed. 

@deafboxer 
You can use the master branch.

Tahutipai, 2017-07-18:
@MDM63 Would you be willing to share your process/toolchange script/purge&wipe procedures etc?

Are there any votes for setting up a dedicated place to discuss multi-extruder printing w/ Klipper?

deafboxer, 2017-07-30:
I posted this on reprap forum but activity is slow there. Trying to
calibrate the one head before proceeding to test second. Hardware is
finally back up. Here is my dilemma... (one note worthy lesson that took me
hours to resolve, install-octopi.sh will not work if downloaded as zip on
another computer and extracted to Pi. Need to chown root:root and chmod to
755 before it will install /tmp/printer.)

My error below is from a fresh install using git.

I can connect, but in Octoprint terminal typing 'status' gives me a message
about printer not being ready. Trying resrart doesnt help. here is
klippy.log file.

Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l',
'/tmp/klippy.log']
Git version: 'v0.4.0-104-g981c536-dirty'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.9 (default, Sep 17 2016, 20:26:04) \n[GCC 4.9.2]'
Start printer at Sun Jul 30 03:37:08 2017 (1501385828.3 11.5)
===== Config file =====
[stepper_x]
step_pin = ar54
dir_pin = ar55
enable_pin = !ar38
step_distance = .0125
endstop_pin = ^ar3
position_endstop = 0
position_max = 200
homing_speed = 50

[stepper_y]
step_pin = ar60
dir_pin = !ar61
enable_pin = !ar56
step_distance = .0125
endstop_pin = ^ar14
position_endstop = 0
position_max = 200
homing_speed = 50

[stepper_z]
step_pin = ar46
dir_pin = ar48
enable_pin = !ar62
step_distance = 0.00025
endstop_pin = ^ar18
position_endstop = 0.5
position_max = 200

[extruder]
step_pin = ar26
dir_pin = ar28
enable_pin = !ar24
step_distance = .002
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = ar10
sensor_type = EPCOS 100K B57560G104F
sensor_pin = analog13
control = pid
pid_kp = 22.2
pid_ki = 1.08
pid_kd = 114
min_temp = 0
max_temp = 250

[heater_bed]
heater_pin = ar8
sensor_type = EPCOS 100K B57560G104F
sensor_pin = analog14
control = watermark
min_temp = 0
max_temp = 130

[fan]
pin = ar9

[mcu]
serial = /dev/ttyUSB0
pin_map = arduino

[printer]
kinematics = cartesian
max_velocity = 300
max_accel = 3000
max_z_velocity = 5
max_z_accel = 100

=======================
Starting serial connect
Stats 11.6: gcodein=0 print_time=0.000 buffer_time=0.000 print_stall=0
mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000
Stats 12.6: gcodein=0 print_time=0.000 buffer_time=0.000 print_stall=0
bytes_write=16 bytes_read=0 bytes_retransmit=53 bytes_invalid=0 send_seq=3
receive_seq=1 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.800
ready_bytes=0 stalled_bytes=0 last_clock=0 last_clock_time=0.000
mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000
Stats 13.6: gcodein=0 print_time=0.000 buffer_time=0.000 print_stall=0
bytes_write=32 bytes_read=0 bytes_retransmit=86 bytes_invalid=0 send_seq=5
receive_seq=1 retransmit_seq=4 srtt=0.000 rttvar=0.000 rto=1.600
ready_bytes=0 stalled_bytes=0 last_clock=0 last_clock_time=0.000
mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000
Stats 14.6: gcodein=0 print_time=0.000 buffer_time=0.000 print_stall=0
bytes_write=48 bytes_read=0 bytes_retransmit=86 bytes_invalid=0 send_seq=7
receive_seq=1 retransmit_seq=4 srtt=0.000 rttvar=0.000 rto=1.600
ready_bytes=0 stalled_bytes=0 last_clock=0 last_clock_time=0.000
mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000
Stats 15.6: gcodein=0 print_time=0.000 buffer_time=0.000 print_stall=0
bytes_write=64 bytes_read=0 bytes_retransmit=143 bytes_invalid=0 send_seq=9
receive_seq=1 retransmit_seq=7 srtt=0.000 rttvar=0.000 rto=3.200
ready_bytes=0 stalled_bytes=0 last_clock=0 last_clock_time=0.000
mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000
Stats 16.6: gcodein=0 print_time=0.000 buffer_time=0.000 print_stall=0
bytes_write=80 bytes_read=0 bytes_retransmit=143 bytes_invalid=0
send_seq=11 receive_seq=1 retransmit_seq=7 srtt=0.000 rttvar=0.000
rto=3.200 ready_bytes=0 stalled_bytes=0 last_clock=0

KevinOConnor, 2017-07-30:
On Sun, Jul 30, 2017 at 07:25:28PM +0000, deafboxer wrote:
> I posted this on reprap forum but activity is slow there. Trying to
> calibrate the one head before proceeding to test second. Hardware is
> finally back up. Here is my dilemma... (one note worthy [...]

I'd recommending following the instructions listed at:
https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md

If the installation is customized, it's difficult to diagnose any
problems.

> My error below is from a fresh install using git.
> 
> I can connect, but in Octoprint terminal typing 'status' gives me a message
> about printer not being ready. Trying resrart doesnt help. here is
> klippy.log file.
> 
> Starting Klippy...
[...]
> Stats 16.6: gcodein=0 print_time=0.000 buffer_time=0.000 print_stall=0
> bytes_write=80 bytes_read=0 bytes_retransmit=143 bytes_invalid=0
> send_seq=11 receive_seq=1 retransmit_seq=7 srtt=0.000 rttvar=0.000
> rto=3.200 ready_bytes=0 stalled_bytes=0 last_clock=0

Does the log file end here?  If not, please post the entire log file.

Also, what does the output of "make flash FLASH_DEVICE=/dev/ttyUSB0"
report?

-Kevin

deafboxer, 2017-07-30:
Full log file. output of make flash initially connects but times out. i can get the exact output in the hour. 

Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.4.0-104-g981c536-dirty'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.9 (default, Sep 17 2016, 20:26:04) \n[GCC 4.9.2]'
Start printer at Sun Jul 30 03:37:08 2017 (1501385828.3 11.5)
===== Config file =====
[stepper_x]
step_pin = ar54
dir_pin = ar55
enable_pin = !ar38
step_distance = .0125
endstop_pin = ^ar3
position_endstop = 0
position_max = 200
homing_speed = 50

[stepper_y]
step_pin = ar60
dir_pin = !ar61
enable_pin = !ar56
step_distance = .0125
endstop_pin = ^ar14
position_endstop = 0
position_max = 200
homing_speed = 50

[stepper_z]
step_pin = ar46
dir_pin = ar48
enable_pin = !ar62
step_distance = 0.00025
endstop_pin = ^ar18
position_endstop = 0.5
position_max = 200

[extruder]
step_pin = ar26
dir_pin = ar28
enable_pin = !ar24
step_distance = .002
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = ar10
sensor_type = EPCOS 100K B57560G104F
sensor_pin = analog13
control = pid
pid_kp = 22.2
pid_ki = 1.08
pid_kd = 114
min_temp = 0
max_temp = 250

[heater_bed]
heater_pin = ar8
sensor_type = EPCOS 100K B57560G104F
sensor_pin = analog14
control = watermark
min_temp = 0
max_temp = 130

[fan]
pin = ar9

[mcu]
serial = /dev/ttyUSB0
pin_map = arduino

[... 1101 lines cut ...]
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 202, in connect
    self.mcu.connect()
  File "/home/pi/klipper/klippy/mcu.py", line 457, in connect
    self.serial.connect()
  File "/home/pi/klipper/klippy/serialhdl.py", line 71, in connect
    self.serialport, self.baud, timeout=0)
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/serial/serialutil.py", line 236, in __init__
    self.open()
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/serial/serialposix.py", line 286, in open
    self._update_dtr_state()
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/serial/serialposix.py", line 612, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
IOError: [Errno 5] Input/output error

deafboxer, 2017-07-30:
here is output from make flash.

pi@octopi:~/klipper $ make flash FLASH_DEVICE=/dev/ttyUSB0
  Flashing /dev/ttyUSB0 via avrdude
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer

avrdude done.  Thank you.

src/avr/Makefile:27: recipe for target 'flash' failed
make: *** [flash] Error 1

KevinOConnor, 2017-07-30:
On Sun, Jul 30, 2017 at 02:59:50PM -0700, deafboxer wrote:
> here is output from make flash.
>
> pi@octopi:~/klipper $ make flash FLASH_DEVICE=/dev/ttyUSB0
>   Flashing /dev/ttyUSB0 via avrdude
> avrdude: stk500v2_ReceiveMessage(): timeout
> avrdude: [...]

The micro-controller must be flashed before one can proceed to
configuring Klipper.  Some things to try:

1 - make sure the klipper software isn't running.  Try "sudo service
    klipper stop" and make sure "ps auxw | egrep -i 'klippy.py[ ]'"
    does not show any output.  Then retry the above flash command.

2 - make sure /dev/ttyUSB0 is the appropriate device name for the
    printer's serial port.  (The "/dev/ttyACM0" is also very popular.)
    Also, make sure the printer is up and running and connected.

Finally, if the install was previously attempted under a different
username (or from a different installation directory), then be aware
that some configuration files may be stale and pointing to the old
installation site.

-Kevin

deafboxer, 2017-08-09:
Found root cause of error. I was using a USBTiny to program the AtMega2056,
which it doesn't support and only is able to program part of the hex.
Received my USBasp today and will give it a go tonight this week. Lots to
calibrate first.

cfelicio, 2017-08-29:
Yes, I'm interested in testing this out, on a multiple extruder / single nozzle setup. I have a ramps 1.4 board, with both extruder 0 / extruder 1 connected to their respective steppers. :-)

KevinOConnor, 2017-09-01:
On Tue, Aug 29, 2017 at 10:59:49PM +0000, cfelicio wrote:
> Yes, I'm interested in testing this out, on a multiple extruder /
> single nozzle setup. I have a ramps 1.4 board, with both extruder 0
> / extruder 1 connected to their respective steppers. :-)

Great!  Is it just a matter of them sharing the same nozzle and nozzle
heater?  Or, is there some kind of servo activation required to change
filament?  Is there a specific filament change sequence, or can one
just start stepping the second extruder motor?

In general, in order to test a new setup with Klipper, the first step
would be to install the code (as described in the Installation
document), and configure the printer with a single extruder.  Once
that's up and running, we can work on getting the second extruder
working.

-Kevin

cfelicio, 2017-09-01:
It's just a matter of sharing the same nozzle / heater, all the filament switching logic is done on the gcode.

The code will likely be similar to what you already have for multiple extruders, with the following differences:

- Nozzle offset is always 0 (as you have a single nozzle)
- No heater control / reporting for the 2nd extruder, as you have a single heater

There are some users currently running Klipper already on the D Bot group, I will try to start with their default configuration this weekend, and see if I can get it to work. 

Feedback so far is very positive, with users reporting an amazing speed up compared to Marlin!

KevinOConnor, 2017-09-03:
On Fri, Sep 01, 2017 at 09:30:28PM +0000, cfelicio wrote:
> It's just a matter of sharing the same nozzle / heater, all the filament switching logic is done on the gcode.
>
> The code will likely be similar to what you already have for multiple extruders, with [...]

If it's just a matter of sharing the heater, then I think the python
code can also just share the heater.  Could you try a test with the
work-multiextrude-20170429 branch:

git fetch ; git checkout origin/work-multiextrude-20170429

Then update your config with an "extruder1" section and set
"shared_heater: extruder" in its config section - see
config/example-extras.cfg for details.

Unfortunately, I don't have a good way to test this.  Let me know how
it goes (success or failure).

-Kevin

KevinOConnor, 2017-10-06:
Basic support for multiple extruders is now present.

closed 2017-10-06 by KevinOConnor