Klipper3d/klipper #9: Cartesian Z Probe/ Z EndStop

superg3, 2017-05-11:
Hello from Spain,
I'm very interested in your firmware because you are improving the kinematics. I installed Klipper but I could not find how setup the z-profe for homing, Probe is working like a Z endstop but I dont know how set it to home in the center of the bed.
I`m availabe as beta tester as I can have your firmware installed permanently. Also I would like to donate if you accept donations.
Kind regards,
Marcos Lemos.

KevinOConnor, 2017-05-12:
Thanks!  Unfortunately the support for z-probe hardware has not been written yet.  It is on the Todo list.

Does your printer also have regular endstops?  If so, you might be able to manually level your bed until z-probe hardware is supported in Klipper.

If you'd like to beta test z-probe then please post the /tmp/klipper.log file.  However, I don't think I'll be able to look at writing the generic bed leveling code for a couple of months.

Spartus, 2017-10-27:
Hi Kevin, I just want to check in since several months have passed and I'm hoping to see news of mesh leveling support one day. Is it still on your radar?

Thanks for the awesome project.

KevinOConnor, 2017-10-27:
On Fri, Oct 27, 2017 at 02:30:48PM +0000, Spartus wrote:
> Hi Kevin, I just want to check in since several months have passed and I'm hoping to see news of mesh leveling support one day. Is it still on your radar?
> 
> Thanks for the awesome project.

Thanks.  I just finished assembling a new delta printer that I
purchased.  So, I do plan to work on probe support and delta
auto-calibration.  Not sure about mesh leveling, but I do hope to have
some improved probe support in the coming weeks.

-Kevin

MrWu-at-msv, 2018-01-12:
For who want use Z sensor as standard Z end-stop to doing home at center you can easily use this g-code instead of G28:
G28 X0 Y0; home
G0 X110 Y110; at center of your print bed or where you want
G28 Z0;

i don't know if is possible to set up a offset on printer.cfg but this is a fast work around

KevinOConnor, 2018-01-23:
The work-probe-20170609 test branch now has code that allows the probe to be used in place of a normal z endstop.  (It also has homing_override code that can be used to position the head prior to Z probing.)

cd ~/klipper ; get fetch ; git checkout origin/work-probe-20170609 ; sudo service klipper restart

I don't have a good way to test this - please let me know your results (success or failure).

KevinOConnor, 2018-01-29:
I've merged the work-probe-20170609 branch into the master branch.

closed 2018-01-29 by KevinOConnor