Klipper3d/klipper #10: Fan support

outlookhazy, 2017-05-12:
I'm very excited by this project - it makes perfect sense to me. I just finished working my way through the first pass of configuration. I may be missing something, but I don't see much in the way of fan support. I found the "Extruder print fan" - is this referring to the part-cooling fan or the extruder transition fan? In either case - is the other supported?
I attempted to manually enable an always-on fan at startup using the custom command support, but apparently no matter what I do I can't get the formatting correct there... not sure what exactly it's expecting.

KevinOConnor, 2017-05-12:
Thanks.  The "[fan]" section in the config file is for the print cooling fan.  If you have a cpu or nozzle cooling fan, then it can be turned on via the custom block in the mcu section.  There is an example of this in the config/makergear-m2-2012.cfg file.  (Note the "set_pwm_out" command.)  If it is not working for you, then please post the /tmp/klippy.log file.

outlookhazy, 2017-05-12:
Noted, I figured there must be a way - sounds like i'm on the right track.  

As a feature request, I'd love to be able to tie that fan to the extruder. Actually, being able to tie a fan to an arbitrary thermistor would be nice - especially if it was structured such that an arbitrary number of fans could be tied to each. I haven't dug into the code so _that_ might not be feasible - but if each fan were its own object/class subscribing to temperature update events - then each fan could be configured in a section similar to an extruder/tool. 

I.E. fan1 is on pin5 in PWM mode, observing thermistor 1 with a PID profile, min/target/shutdown temps etc  
fan2 is on pin 6 in digital mode, observing thermistor 1 with a hysteresis profile min/target/max/transition

Usage wouldn't really be specific to 'fan' in this scenario, it would be operating the same way as an extruder or heat bed might - except not controlled via G-Code necessarily.

KevinOConnor, 2017-05-22:
Implementing a PID algorithm for the fan would only require python host code changes. If you're interested in implementing this, look at klippy/fan.py (for the code to define a fan) and klippy/heater.py (for code that implements a PID).

I've added an item to the Todo list for moving the definitions for auxiliary fans from the custom block to additional config sections. Defining a config section with something like "[aux_fan CPU]" would be nicer than having to navigate the custom block.

Can you confirm that you've been able to get your fan working? I will close this issue if so.

outlookhazy, 2017-05-22:
Yes, I got the fan working using the custom commands as suggested. I started looking into modding the code to allow extruder to have its own fan definition loaded from config, but got sidetracked rewriting OpenDACT...

closed 2017-05-23 by KevinOConnor