Klipper3d/klipper #11: Delta Angle Corrections Support

outlookhazy, 2017-05-29:
Diving into the kinematics code isn't on my to-do list, but it _looks_ like the tower angles are hard-coded in one spot only. Am I correct in thinking that's the only place an adjustment would need to be made for a tower angle correction? (and if so, that adding that as a config setting would be super-trivial?)

I moved away from klipper for awhile because calibration was becoming a pain on this new machine... since then I've been working on my own rewrite of an OpenDACT-eque application that uses the Escher3D code for delta calibration. Using that I have my settings (tentatively) flawless with Repetier. All the adjustments of which I think I can copy over to klipper _except_ the delta angle corrections...

KevinOConnor, 2017-05-29:
If you are saying the towers are not exactly at 210, 330, and 90 degrees, then yes - making that configurable is trivial.

outlookhazy, 2017-05-30:
Yes, nothing is perfect ;-) 
The calibration routine I'm using reverse engineers the angle error (or at least uses that as a logical equivalent to remove other error)

KevinOConnor, 2017-06-05:
On Mon, May 29, 2017 at 05:03:39PM -0700, Charlie wrote:
> Yes, nothing is perfect ;-) 
> The calibration routine I'm using reverse engineers the angle error (or at least uses that as a logical equivalent to remove other error) 

I added an "angle" config setting on a new development branch
work-delta-20170605 (git fetch ; get checkout work-delta-20170605 ).
Can you confirm it works as expected?

-Kevin

outlookhazy, 2017-06-05:
I _think_ I'll have time to check it out tonight. I can make sure nothing explodes, but i'm not sure how to guarantee that it's working as expected since the change is so subtle, especially with no z-probe

outlookhazy, 2017-06-08:
I haven't forgotten about this, I discovered some more dimensional/calibration/repeat-ability issues with my rig while printing some prototype parts for work and have been working to get those nailed down before I try to evaluate this. I don't want to introduce extra variables (my angular correction is only .738 degrees on A and 1.402 degrees on B)

outlookhazy, 2017-06-09:
It's definitely not exploding, but I can't for the life of me get my accuracy dialed in like I have done with Repetier which is **really** frustrating, because the movements with klipper are _**so**_ much smoother!  

I wish I was more comfortable with python :-| 
I feel like basic G30 support would be very similar to the home routine... doing this with paper after writing an app that calibrates with up to 50 points repeated N iterations is torture - and apparently the settings I generated using that don't precisely translate to klipper.

outlookhazy, 2017-06-09:
Dragged out the dial indicator and got it close enough to start printing. I'd tentatively call the angle correction working, don't think I'll be _certain_ until we have G30 - I just can't get repeatable enough measurements.

KevinOConnor, 2017-06-09:
On Thu, Jun 08, 2017 at 07:10:49PM -0700, Charlie wrote:
> It's definitely not exploding, but I can't for the life of me get my accuracy dialed in like I have done with Repetier which is **really** frustrating, because the movements with klipper are [...]

Can you point me to a concise description of the delta calibration
algorithm?  I could probably assist with the code.

-Kevin

outlookhazy, 2017-06-09:
The algorithm itself isn't the problem, but basically it works like this:  

- Initial machine parameters are entered including
  - machine homed height
  - diagonal rod length
  - delta radius
  - angle adjustments a/b/c
  - endstop offsets a/b/c

Next, a series of points on the bed are probed (using those initial settings), this generates a list of height deviations from 'expected' value - I personally just normalize them all to the center height.

Now the algorithm itself. It will take those initial values and probed height offsets and attempt to determine the 'actual' machine dimensions - this might technically be saccadic gradient descent, but honestly I just rewrote the thing in C# from javascript - I'm no math guy. It runs through a specified number of factors (radius, angles, offsets etc), making adjustments until the test data error is sufficiently reduced. Once done, it spits out new values for the requested number of factors. (i.e. 3 factors will spit out  only endstop adjustments, 4 factors adds delta radius, 6 factors adds adjustments to tower angle, and 7 factors also attempts to adjust diagonal rod length).

My implementation repeats this process as many times as requested, updating firmware after each run and then re-testing the points. I'm also applying a random rotation to the test points on each iteration to avoid overfitting inconsistencies in the bed.  

So the issue with klipper in this scenario is a couple things. Initially it was that I couldn't include the delta angle corrections that I had determined using other firmware. Now that this has been addressed, I have discovered other problems such as my offset adjustments using different units, then the homing behavior being slightly different (i.e. different max height). Following those, it looks like the kinematics in general just aren't quite the same. This may be down to rounding errors in the Repetier firmware - I don't know. 

What I found is that it looks like I will need slightly different adjustments in klipper than I did before. That in itself is no big deal (theoretically) I can just re-measure using klipper and enter new adjustments relative to its behavior. The rub is that I don't have a good way to quickly and efficiently generate that height delta map without the ability to probe the bed (G30). I can jog the head around using the terminal, but accurately measuring those height differences and iterating the process is _super_ slow and unreliable. What I was doing last night was jogging the effector to a series of known points and plunging against an inverted dial indicator one point at a time - plugging those numbers into a spreadsheet to determine a height map, and then manually running that data through the calibration routine. Unfortunately those measurements aren't terribly accurate due to the human factor and the cheap indicator I have. It's also **way** slower. Using my application, I can probe 50 points, run the calibration routine, adjust firmware, and be ready to do another iteration in just over a minute or two with my own graphing of progress to see when the numbers stop improving (usually after only a couple iterations). Manually, it takes probably 15 minutes to do 6 points - and even after doing that probably 5 times my dimensional accuracy is still off - which means my measurements are probably suspect - or i'm swapping a sign somewhere in my conversions into klipper dimensions - it's hard to determine when each test takes so long.

outlookhazy, 2017-06-09:
My python comment was basically in reference to wanting to help out by adding **basic** G30 support, which I guess would require a config section with 
- pin definition
- offset x, y, z, definitions
- probe start distance definition
- probe speed

sending G30 would cause a relative move to start position [offset x, offset y, offset z + probe start distance]
the head would then relative move down until defined pin triggers, save current height, return to start position, then report nozzle height via serial (measured height + offset z + start distance). 
The routine itself is pretty simple without start/stop script support, repeated iterations with averaging, etc that other firmwares have or any kind of baked in bed-leveling. The issue is that it's not _quite_ the same as homing as it's motor independent, and I'm not well versed enough in python to try tackling it. Luckily the code looks pretty modular and flexible from what I've seen, so i'm sure it'll show up at some point when you have time.

KevinOConnor, 2017-06-09:
On Fri, Jun 09, 2017 at 08:48:36AM -0700, Charlie wrote:
> What I found is that it looks like I will need slightly different
> adjustments in klipper than I did before. That in itself is no big
> deal (theoretically) I can just re-measure using klipper and [...]

I just created a testing branch work-probe-20170609 (git fetch ; git
checkout work-probe-20170609 ) with some demo code for probing.  Add a
[probe] section to the config (see the updated example.cfg file) and
then one should be able to issue a new PROBE g-code command.  The new
PROBE command will move the head down (from its current XY location)
until the configured probe pin triggers.  One can then issue an M114
command to read out where the head is.

It's demo code, but hopefully shows how one can probe the height.

-Kevin

outlookhazy, 2017-06-09:
**Awesome!**  I'll get started testing this tonight

I've actually been getting myself set up for python development to see about helping out instead of whining. :-)

KevinOConnor, 2017-06-10:
On Fri, Jun 09, 2017 at 01:13:46PM -0700, Charlie wrote:
> **Awesome!**  I'll get started testing this tonight

Upon further review, it looks like the Klipper delta code was not
taking into account the position_endstop on each stepper.  There's a
good chance this explains why your previously calculated angle
corrections did not apply properly to Klipper.  I've rebase'd and
committed a fix to the work-probe-20170609 branch.

Please fetch and checkout the work-probe-20170609 branch again and see
if it fixes the issue.

Thanks,
-Kevin

outlookhazy, 2017-06-10:
I _really_ wish I had seen this sooner...  
On the bright side, out of frustration I wrote another app, (WAY more hack-y than the others) to calibrate klipper over SSH.
I'm creating a terminal session to monitor the remote machine virtual port (cat /tmp/printer)
Then using that same session to send commands to the printer via via echo pipe.
So I can issue a series of moves/probes/m114s and monitor results. Then I run the calibration routine based on the data returned, apply calculated settings changes to a template config file, then transfer that file to the remote host via SCP and send a restart

So far the probe functionality seems to be working (besides some frustrations with the serial behavior)

Current RMS height deviation over a 200mm diameter build area is now down to **.017mm** - huzzah!

closed 2017-06-10 by outlookhazy