Klipper3d/klipper #12: Add secondary serial support instead of USB (enhancement)

TunaLatte, 2017-06-04:
Hello,

As the title suggests, specifically for platforms that emulate usb like at90usb1286.
Trying to expose the serial pins of the printrboard on this firmware for esp8266 wifi instead of USB.

Thanks and great work.

KevinOConnor, 2017-06-05:
On Sun, Jun 04, 2017 at 04:48:44AM -0700, TunaLatte wrote:
> As the title suggests, specifically for platforms that emulate usb like at90usb1286.

I put a patch up on a new work-at90-serial-20170605 branch (git fetch
; get checkout work-at90-serial-20170605).  Let me know if it works
for you.

-Kevin

TunaLatte, 2017-06-06:
Thanks that was very quick, will try ASAP and report.

TunaLatte, 2017-06-08:
So I have set the baud rate to 115200, im getting garbled data "?". Tried using default 25000 baud rate, same garbled data.
Thank you.

KevinOConnor, 2017-06-09:
On Thu, Jun 08, 2017 at 02:48:31PM -0700, TunaLatte wrote:
> So I have set the baud rate to 115200, im getting garbled data "?". Tried using default 25000 baud rate, same garbled data.

Alas, there isn't much assistance I can provide sice I don't have
access to the board in question.  Do note, however, that the protocol
between the micro-controller and the host is a binary protcol (as
described in docs/Protocol.md) so what looks like "garbled data" may
actually be okay.  The klippy/console.py script (as in
```~/klippy-env/bin/python ~/klipper/klippy/console.py /dev/ttyACM0
250000```) can be used to issue and read low-level commands if
desired.

-Kevin

TunaLatte, 2017-06-09:
Noted.
For info: What I get in /tmp/klippy.log
`Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
....
`

TunaLatte, 2017-06-09:
#15 Problem solved after closing this issue. Horray

(Just printed a wall cube, with transparent pla, im very impressed by the lack of vibrations being visible on each layer, making it more glass. Printing a circle now, more clear/transparent, and no sharp outer layer feel. The difference in noise between this and Marlin is significant, also this is the first time the printer is able to do Z travel speed at >30mm/s, even when all axis are moving at the same time like what curaengine slicer does at the beginning of a print, using modified bed printrbot jr here +using old router with internal serial connected to the mcu, 800mhz mips, Ill attempt to run klippy on it later (hopefully its not too slow), for now its just a wireless serial mode, esp8266 is kinda unstable.)
Many thanks and great work, I'll try to help as long as I'm using it. (no reason I wont anytime soon tho :))

closed 2017-06-09 by TunaLatte

KevinOConnor, 2017-06-09:
Thanks - I committed this change to the master branch ( b8094de129de17280f58d1042e6dcdd48f9eaef3 ).