Klipper3d/klipper #13: AT90USB error

TunaLatte, 2017-06-08:
Hello,

It seems that I have skipped the log before trying the serial option in the other issue ticket.
I'm getting the following error:

```
=======================
Starting serial connect
Stats 15614.4: gcodein=0 print_time=0.000 buffer_time=0.000 print_stall=0  mcu_task_avg=0.000000 mcu_task_stddev=0.000000
Loaded 51 commands (?-20170608_234759-tmash-nuc)
MCU config: ADC_MAX=1023 PWM_MAX=255 CLOCK_FREQ=16000000 SOFT_PWM_MAX=256 MCU=at90usb1286 STATS_SUMSQ_BASE=256
Stats 15615.4: gcodein=0 print_time=0.000 buffer_time=0.000 print_stall=0 bytes_write=345 bytes_read=1961 bytes_retransmit=9 bytes_invalid=0 send_seq=46 receive_seq=46 retransmit_seq=2 srtt=0.001 r
ttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 est_clock=0.000 last_ack_time=15615.132 last_ack_clock=116493520 mcu_task_avg=0.000000 mcu_task_stddev=0.000000
Config error
Traceback (most recent call last):
  File "/home/tmash/Downloads/klipper-work-at90-serial-20170605/klippy/klippy.py", line 202, in connect
    self.mcu.connect()
  File "/home/tmash/Downloads/klipper-work-at90-serial-20170605/klippy/mcu.py", line 465, in connect
    self._build_config()
  File "/home/tmash/Downloads/klipper-work-at90-serial-20170605/klippy/mcu.py", line 553, in _build_config
    cmd,))
Error: Unable to translate pin name: config_soft_pwm_out oid=0 pin=ar21 cycle_ticks=TICKS(0.100000) default_value=0 max_duration=TICKS(5.000000)
```
I have tried changing the pin number, removing 'ar', and playing with other settings (enabling fan hardware pwm etc), but the error is persistent. N.b: Took from the serial branch but compiled with USB instead of serial to test first. I don't see how that would be an issue though.

Thanks!

KevinOConnor, 2017-06-08:
On Thu, Jun 08, 2017 at 02:16:14PM -0700, TunaLatte wrote:
> I have tried changing the pin number, removing 'ar', and playing with other settings (enabling fan hardware pwm etc), but the error is persistent. N.b: Took from the serial branch but compiled with USB instead of serial to test first. I don't see how that would be an issue though.

I'm not aware of an Arduino board (or similar) for the AT90USB.  So,
there is no Arduino pin name to hardware pin name mapping.  You can
use the hardware pin names (eg, PA0 - PF7) directly.

-Kevin

TunaLatte, 2017-06-09:
Current config:
[printer.txt](https://github.com/KevinOConnor/klipper/files/1064741/printer.txt)

Error:
```
=======================
Starting serial connect
Stats 21573.2: gcodein=0 print_time=0.000 buffer_time=0.000 print_stall=0  mcu_task_avg=0.000000 mcu_task_stddev=0.000000
Loaded 51 commands (?-20170609_143057-tmash-nuc)
MCU config: ADC_MAX=1023 PWM_MAX=255 CLOCK_FREQ=16000000 SOFT_PWM_MAX=256 MCU=at90usb1286 STATS_SUMSQ_BASE=256
Stats 21574.2: gcodein=0 print_time=0.000 buffer_time=0.000 print_stall=0 bytes_write=350 bytes_read=1968 bytes_retransmit=9 bytes_invalid=0 send_seq=46 receive_seq=46 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 est_clock=0.000 last_ack_time=21573.918 last_ack_clock=655284894 mcu_task_avg=0.000000 mcu_task_stddev=0.000000
Config error
Traceback (most recent call last):
  File "/home/tmash/Downloads/klipper-work-at90-serial-20170605/klippy/klippy.py", line 202, in connect
    self.mcu.connect()
  File "/home/tmash/Downloads/klipper-work-at90-serial-20170605/klippy/mcu.py", line 465, in connect
    self._build_config()
  File "/home/tmash/Downloads/klipper-work-at90-serial-20170605/klippy/mcu.py", line 553, in _build_config
    cmd,))
Error: Unable to translate pin name: config_analog_in oid=1 pin=PF0

```
Tried modifying pins.py, specifically with port_pin, though that affected other pins (earlier like stepper pins).
PF# range is not working.

KevinOConnor, 2017-06-09:
On Fri, Jun 09, 2017 at 10:13:14AM -0700, TunaLatte wrote:
> Current config:
> [printer.txt](https://github.com/KevinOConnor/klipper/files/1064741/printer.txt)
>
> Error:
> ```
> =======================
> Starting serial connect
> Stats 21573.2: gcodein=0 print_time=0.000 [...]

Looks like a typo in pins.py - there should be 6 pins defined for
AT90usb1286:

--- a/klippy/pins.py
+++ b/klippy/pins.py
@@ -28,7 +28,7 @@ def beaglebone_pins():
 
 MCU_PINS = {
     "atmega168": port_pins(4), "atmega328": port_pins(4),
-    "atmega644p": port_pins(4), "at90usb1286": port_pins(5),
+    "atmega644p": port_pins(4), "at90usb1286": port_pins(6),
     "atmega1280": port_pins(12), "atmega2560": port_pins(12),
     "sam3x8e": port_pins(4, 32),
     "pru": beaglebone_pins(),

TunaLatte, 2017-06-09:
Issue resolved, thanks a lot for the feedback!

closed 2017-06-09 by TunaLatte

KevinOConnor, 2017-06-09:
Thanks - committed in ( d0e6a0928b483f29a3bcfd86d16192bef5652042 ).