Klipper3d/klipper #15: [AT90USB1286] Printer extremely slow

TunaLatte, 2017-06-09:
Hello,

As the title suggests, the stepper motors move extremely slow, when homing, endstop trigger sequence switching from axis x to y is also slow, takes about 5 seconds to switch, sending m105 temperature report is slow aswell (3 seconds).
I'm guessing this has to do with clock/tick frequency?
```
===== Config file =====
[stepper_x]
step_pin = PA0
dir_pin = !PA1
enable_pin = !PE7
step_distance = 0.0015782
endstop_pin = ^PE3
position_min = -0.25
position_endstop = 0
position_max = 100

[stepper_y]
step_pin = PA2
dir_pin = PA3
enable_pin = !PE6
step_distance = 0.0015782
endstop_pin = ^PB0
position_min = -0.25
position_endstop = 0
position_max = 100

[stepper_z]
step_pin = PA4
dir_pin = !PA5
enable_pin = !PC7
step_distance = 0.0000440001408005
endstop_pin = ^PE4
position_min = 0.1
position_endstop = 0.5
position_max = 200

[extruder]
step_pin = PA6
dir_pin = PA7
enable_pin = !PC3
step_distance = 0.000203388451604
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = PC5
filament_diameter = 1.750                                                                                                                   [0/1935]
heater_pin = PC5
max_power = 1.0
sensor_type = EPCOS 100K B57560G104F
sensor_pin = PF0
control = pid
pid_kp = 22.2
pid_ki = 1.08
pid_kd = 114
min_temp = 0
max_temp = 110

[fan]
pin = PC6

[mcu]
serial = /dev/ttyACM0
pin_map = None
custom =

[printer]
kinematics = cartesian
max_velocity = 500
max_accel = 3000
max_z_velocity = 25
max_z_accel = 30

=======================
```
I may record a video if necessary.

KevinOConnor, 2017-06-09:
On Fri, Jun 09, 2017 at 12:50:16PM -0700, TunaLatte wrote:
> Hello,
> 
> As the title suggests, the stepper motors move extremely slow, when homing, endstop trigger sequence switching from axis x to y is also slow, takes about 5 seconds to switch, sending m105 temperature report is slow aswell (3 seconds).
> I'm guessing this has to do with clock/tick frequency?

Is this on the AT90USB?  My AT90USB chip shipped with the prescaler
set to 8 and I couldn't change it via the AVR fuses because the
factory bootloader required that speed at bootup.  If you have a
similar setup, recompile the firmware and be sure to set the
CONFIG_CLEAR_PRESCALER option to true.

-Kevin

TunaLatte, 2017-06-09:
Printer is no longer a sloth :smile: 
CONFIG_CLEAR_PRESCALER set to true on printrboard with stock dfu bootloader works flawlessly.
again thanks.

closed 2017-06-09 by TunaLatte

KevinOConnor, 2017-06-09:
Thanks.  Since this seems to be a common issue with the at90usb1286 chip, I've committed a change to make that option enabled by default (46846d4297dcacf5b3fe70dd161fc18c754f8469).

KevinOConnor, 2017-10-12:
Is the config above a good starting point for printrboards?  I'd like to add a config/generic-printrboard.cfg file to help other users with the hardware.

TunaLatte, 2017-10-18:
Yes, though it should be noted that the heater pin can be either PC5 or PC3 depending on revision. (heatbed and extruders are swapped) In my case the heatbed is PC3.

KevinOConnor, 2017-10-18:
On Wed, Oct 18, 2017 at 06:54:09AM +0000, TunaLatte wrote:
> Yes, though it should be noted that the heater pin can be either PC5
> or PC3 depending on revision. (heatbed and extruders are swapped) In
> my case the heatbed is PC3.

Thanks.  Do you flash with `make flash FLASH_DEVICE=/dev/ttyACM0` or
do you need special flashing commands?

-Kevin

TunaLatte, 2017-10-19:
> Thanks.  Do you flash with `make flash FLASH_DEVICE=/dev/ttyACM0` or
> do you need special flashing commands?

By default, Printrboard comes with DFU bootloader, I used dfu-programmer.
On rev D, BOOT0 has to be shorted, followed by a powercycle, then running the following command after klipper make, (w/udev perm) `dfu-programmer at90usb1286 erase && dfu-programmer at90usb1286 flash pathtofirmware.bin`
Older revisions need to remove the jumper (jumper installed by default) according to RepRapWiki.
CDC bootloader is available for direct flash with avrdude, though I haven't tested it (limits flash space to 60k), I believe that would work with make flash?