Klipper3d/klipper #16: Virtual serial issues

outlookhazy, 2017-06-10:
I've noticed problems with some command sequences not registering when sent in succession,
Octoprint, for example, sends every jog move like so:
G91
G0 <axis move>
G90
I.E. relative

However, klipper often doesn't always register the switch back to absolute so a subsequent move fails.
Along the same lines, sending a sequence like this:
g0 x0 y0 z5
m400
probe
m400
m114

will (most of the time) fail. The location request might fire before the probe (though as probe isn't a standard command, I don't know if that's a special case)
Maybe i'm misunderstanding the protocol, but I would assume some sort of command queue would be handling this

KevinOConnor, 2017-06-11:
On Sat, Jun 10, 2017 at 02:44:45PM -0700, Charlie wrote:
> I've noticed problems with some command sequences not registering when sent in succession,
> Octoprint, for example, sends every jog move like so:
> G91
> G0 <axis move>
> G90
> I.E. relative
>
> However, [...]

Thanks.  It looks like a recent commit (d0932009) introduced a
regression.  It should be fixed as of commit 95950949.

-Kevin

outlookhazy, 2017-06-13:
This seems to have mostly cleared up that issue, but there's still some quirky behavior going on. I haven't yet figured out exactly what it is, unfortunately. On several occasions I've started a print, the system heats up, and then as soon as it starts running the GCode, it throws an error and the print cancels. I know this isn't helpful information yet, but that's why I haven't closed this issue. In one instance it claimed that it was due to an over-extrude request, another it seemed to think it hadn't been homed or the move was out of bounds (which leads me back to the possibility that this is another absolute vs relative position thing).

Like I said, this is at the beginning of a print - doing a restart of the firmware, re-home, and restart the same gcode file fixed it eventually (sometimes requiring multiple attempts) in each case - but I haven't tracked down the common link as i'm not printing lots of small jobs currently. What I can say is that once a print has successfully started, it's mostly fine (except in one instance where it started trying to print at ~z20, that didn't repeat)

KevinOConnor, 2017-06-13:
On Tue, Jun 13, 2017 at 07:51:55AM -0700, Charlie wrote:
> This seems to have mostly cleared up that issue, but there's still some quirky behavior going on. I haven't yet figured out exactly what it is, unfortunately. On several occasions I've started a [...]

If you see any strange behavior, issue an M112 command - it will cause
the Klipper host to shutdown the firmware and log a bunch of
information to its log.  Then attach the full /tmp/klippy.log file.

-Kevin

outlookhazy, 2017-06-15:
I still don't have a log for you yet (sorry), but the issue appears to be max_extrude_cross_section triggering intermittently (very early in the print, just after heating up). Running the exact same GCode can both trigger it and run successfully. I've not yet narrowed down whether it's something that I'm doing via Octoprint or not. I also don't have a definitive answer for what i'm doing to fix it. I confirmed it was repeatable, then restarted everything to snap you a clean log file, then it worked ... seemingly because I restarted everything... derp. :-\

KevinOConnor, 2017-06-15:
On Thu, Jun 15, 2017 at 02:27:08PM -0700, Charlie wrote:
> I still don't have a log for you yet (sorry), but the issue appears to be max_extrude_cross_section triggering intermittently (very early in the print, just after heating up). Running the exact same [...]

That sounds like the gcode may not be clearing the "origin" correctly.
I'd verify that the gcode is explicitly setting G90/G91, G92, and/or
M82/M83 prior to any moves or extrusions.

-Kevin

KevinOConnor, 2017-07-01:
Are you still seeing this issue?

outlookhazy, 2017-07-01:
Printer has been out of commission for awhile as I've been busy with other projects. I got it fixed last night and have started running a couple prints. I'll let you know, so far it seems mostly ok - there was one issue with a print cancelling itself before it's begun due to "must home first" even though G28 is the second line of my gcode header. I haven't narrowed down whether that's only on fresh boot.  

for the record, this is the start header i'm using:  

> G90 ; absolute coordinates  
> G28 ; Home
> M190 S{material_bed_temperature}; Heat Bed
> G0 X-69.168 Y71.147 Z240 F12000 ; Move to drip position
> M109 S{material_print_temperature} ; heat extruder
> G0 Z5 F12000 ; Drop
> M400 ; Wait for Move
> G0 X0 Y100 Z.1 F12000; Wipe any junk
> G92 E0 ; Zero Extruder Coords

KevinOConnor, 2017-07-01:
On Sat, Jul 01, 2017 at 02:04:48PM +0000, Charlie wrote:
> Printer has been out of commission for awhile as I've been busy with other projects. I got it fixed last night and have started running a couple prints. I'll let you know, so far it seems mostly ok [...]

If you observe any unusual behavior, issue an M112 command, and then
forward the /tmp/klippy.log file.  The log file has debugging
information which can help diagnose these types of issues.

-Kevin

outlookhazy, 2017-07-13:
I haven't seen this in awhile, it was probably something I did. Closing until I can replicate

closed 2017-07-13 by outlookhazy

KevinOConnor, 2017-09-02:
FYI, commit d093200 was not a complete fix.  I've confirmed that it was possible for Klippy to process commands out-of-order in some cases if Octoprint sent them without waiting for an "ok" response.  A further fix (0d138342) was just committed to address this.

outlookhazy, 2017-09-02:
Thanks for the heads up, I haven't done much printing in awhile but I just pulled this into the branch I'm using (just a merge of master and probe).