Klipper3d/klipper #17: High CPU usage

TunaLatte, 2017-06-14:
Hello,

I have noticed that whenever klippy is idle disconnected or connected and not printing or doing anything (e.g if I set the extruder temp cpu goes down to 1%, when temperature is reach, spikes 100 again and alters between 100 and 1 in slow frequency, if i turn it off, 100% all the time), cpu spikes to 100% on one cpu thread:
[image]

Also CPU spikes immediatly after repeating pattern of the 'no serial found' message:
[image]

KevinOConnor, 2017-06-15:
On Wed, Jun 14, 2017 at 02:04:35PM -0700, TunaLatte wrote:
> I have noticed that whenever klippy is idle disconnected or connected and not printing or doing anything (e.g if I set the extruder temp cpu goes down to 1%, when temperature is reach, spikes 100 [...]

Thanks.  I think that should be fixed now (as of commit c03b4921).

FYI, separately, I'd recommend running the Klipper host software as a
non-root user.

-Kevin

TunaLatte, 2017-06-15:
Thanks! CPU average is down to 1% average.
And yeah both octoprint and klippy were running as root as I was diagnosing something with socat permissions for wireless printing.
Issue resolved.

closed 2017-06-15 by TunaLatte