Klipper3d/klipper #18: [Error code clarification] Timer too close

TunaLatte, 2017-06-17:
Hello,

I had frequent print crash lately, I'm not sure whether this had to do with wireless communication or klippy/baudrate config.
After crash I get the following message:
`Recv: // Firmware shutdown: Timer too close
`
Thank you.

KevinOConnor, 2017-06-18:
Please attach the full /tmp/klippy.log file containing the event and I'll take a look.

TunaLatte, 2017-06-18:
Alright!, I'll dry run the same 2 hour print, hopefully it is not random.

KevinOConnor, 2017-06-18:
On Sun, Jun 18, 2017 at 03:59:17AM -0700, TunaLatte wrote:
> Alright!, I'll dry run the same 2 hour print, hopefully it is not random.

It's fine if you wish to recreate the error, but you should also be
able to find the last five days of klippy logs in /tmp/ .  So, feel
free to post the one containing the error, or even to post all of
them.

-Kevin

TunaLatte, 2017-06-19:
Wish I didn't reboot! That cleared the /tmp directory.
Anyways I just noticed that my ttl voltage converter (serial) is not working properly. Instead of disclosing this issue as user error only, hopefully the error code might be a hint for others for communication failure perhaps?.
Thanks.

TunaLatte, 2017-06-20:
Hello,
After trying with USB, direct connection to my PC, printer shutdown after 5 hours print :sob: . (Move out of queue)
Here is my klippy log: 
[klippy.txt](https://github.com/KevinOConnor/klipper/files/1089331/klippy.txt)
(note: you might find in the log that the printer did shutdown before due to sensor error adc, diy heatbed here, tho firmware/soft restarted and no issues since then. Guess this is completely unrelated)
I hope its something like bad usb cable/software (no dmesg errors for usb), else reliability might be an issue for me.
Thanks.

KevinOConnor, 2017-06-20:
On Tue, Jun 20, 2017 at 11:51:19AM -0700, TunaLatte wrote:
> Hello,
> After trying with USB, direct connection to my PC, printer shutdown after 5 hours print :sob: . (Move out of queue)
> Here is my klippy log: [...]

A "Move queue empty" micro-controller error is unrelated to a "Timer
too close" micro-controller error.  I don't see anything indicating a
communication fault in the log (no retransmits, no significant
variation in round-trip-time) and the load on the mcu and host look
normal.

It looks like you have a custom setup for the host.  The build was
unable to extract the software version for either the micro-controller
or the host.  I need to know the exact version of the software you are
using (git describe --long --tags --dirty).

Also, can you attach (or email me) the gcode file being printed during
this error?

-Kevin

TunaLatte, 2017-06-20:
I simply downloaded the latest master commit as zip 2 days ago on a separate PC, and thus no git.
Here is the gcode:
[mod.txt](https://github.com/KevinOConnor/klipper/files/1089736/mod.txt)

KevinOConnor, 2017-06-21:
Upon further inspection of the log, the drift between the host and mcu clocks sometimes changes rapidly.  I suspect this is what led to the above error.  I need to further investigate to determine if this is due to some attribute of the AT90USB chip, a code error, or something else.

TunaLatte, 2017-06-22:
I'll try replicating the issue on other computer dry run. Could it be host side stuttering due to kernel/hardware issue?

KevinOConnor, 2017-06-22:
> I'll try replicating the issue on other computer dry run. Could it be host side stuttering due to kernel/hardware issue?

Yes, that's certainly possible.  If you have an RPi2 or RPi3 that would make an excellent test.

FYI, I've run some tests on my AT90USB1286 chip and I can confirm that the odd timing behavior is not inherent to that micro-controller.  I've also identified a few areas in the Klipper clock synchronization code that could be improved.  It will likely be sometime next week before that's ready for testing though.

KevinOConnor, 2017-06-28:
Please try the testing branch work-clocksync-20170627 (git fetch ; git checkout work-clocksync-20170627).  I've reworked the clock synchronization code - I think it should be more stable now.  It would be great if you could run some prints with the new code and post the klippy.log file (I'd like to see the log even if there is no failures).

Thanks,
-Kevin

TunaLatte, 2017-06-28:
Will do with the same setup and gcode uploaded above.

TunaLatte, 2017-06-28:
Hello,
I accidentally ran the same klippy branch, same gcode as above, here is the log:
[klippy.txt](https://github.com/KevinOConnor/klipper/files/1109358/klippy.txt)
Printer did not crash. Ill try the same with the latest commit and post the log.

TunaLatte, 2017-06-28:
work-clocksync-20170627 branch: 
[klippy2.txt](https://github.com/KevinOConnor/klipper/files/1110067/klippy2.txt)
No crash aswell. I'll stay on this branch as primary firmware.

KevinOConnor, 2017-07-01:
Thanks.  The log indicates the new code has detected the clock drift.  The results should be more stable now.  I've committed the changes on the test branch to the master branch.

TunaLatte, 2017-07-01:
Thanks, I noticed that its quite repetitive, donno if this has to do with system latency, clock source set to TSC, I hear that its affected with cpu scaling, Ill try switching to HPET, anyways this is host specific now, closing.
Issue resolved.

closed 2017-07-01 by TunaLatte