Klipper3d/klipper #19: Delta Autocalibration

JohannesOtto, 2017-06-22:
Hi,
I didn't find anything for G33 Delta Autocalibration. Is there a way to implement it?

Best Regards
Johannes

KevinOConnor, 2017-06-22:
The code still needs to be written.

There is a test branch with demo code available at work-probe-20170609 (git fetch ; git checkout work-probe-20170609 ). Add a [probe] section to the config (see the updated example.cfg file) and then one should be able to issue a new PROBE g-code command. The new PROBE command will move the head down (from its current XY location) until the configured probe pin triggers. One can then issue an M114 command to read out where the head is. It's demo code, but hopefully shows an outline of what the code would look like should someone wish to tackle it.

johannes-otto, 2017-06-26:
Thanks,
I ll have a look into that

KevinOConnor, 2017-12-27:
FYI, I've done some work on this recently.  It's in the work-probe-20170609 branch.  The code is in a very raw state, but I've been able to do some basic delta auto calibration with it.  The code isn't ready for regular users, but developers may be interested in taking a look.

The branch allows a "probe" section to be defined in the config, and then one can run a DELTA_AUTOCALIBRATE RADIUS=123.4 command to probe a handful of points around the given bed radius.  At the conclusion of the test, the code should report (via the klippy.log file) new values for endstop positions and tower angles.  It's also possible to run DELTA_MANUAL RADIUS=123.4 to do a manual "paper test" style calibration - it will move the head to various XY points and at each point the user can jog the head to a Z position where a paper between the head and bed moves with a small amount of friction and then issue a NEXT command.  It too will reports updated delta params (via the klippy.log file) at the conclusion of the test.

GruffyPuffy, 2017-12-30:
@KevinOConnor: 
Thanks for the hard work with this project! I downloaded and tested on my small delta (kossel mini clone). I do miss the calibration from Marlin and I saw your branch on this. I switched to that and started to use that as a base for learning this a little. Hence I took the liberty to improve/change it a little so that I could test it. I currently keep it in a branch (based on your) here: [https://github.com/GruffyPuffy/klipper/tree/autocalibration_feature
](https://github.com/GruffyPuffy/klipper/tree/autocalibration_feature)
One thought I had. Do you see the workflow as to do a "DELTA_CALIBRATION", wait for results, update in printer.cfg and restart to take effect? This is at least how I tested my branch. I do think it would be nice to have a way to trigger a calibration, have the results saved and then you can start using it. Maybe you see this, or do you think it must be edited into the printer.cfg manually?`

Best of luck with the code! I will follow it closely!

KevinOConnor, 2017-12-30:
On Sat, Dec 30, 2017 at 02:06:12PM +0000, Stefan Grufman wrote:
> @KevinOConnor:
> Thanks for the hard work with this project! I downloaded and tested on my small delta (kossel mini clone). I do miss the calibration from Marlin and I saw your branch on this. I [...]

Thanks!

One interesting thing I noticed on your branch is that you seem to do
the calibration points at the sensor XY location instead of the nozzle
location.  I'm not sure what the best approach is.  For bed leveling,
the sensor location seems best, but for the delta arm calibration I'm
not sure if that is accurate.

> One thought I had. Do you see the workflow as to do a "DELTA_CALIBRATION", wait for results, update in printer.cfg and restart to take effect? This is at least how I tested my branch. I do think it [...]

For now, my plan is to require a config file edit followed by a
RESTART.  It's not technically difficult to directly implement the
updated values.  But, I like having a config file that accurately
describes the machine so that I can do backups and restores, and I
like being able to review all the printer settings in one location.

This is certainly an area that will need to be revisited.

-Kevin

megablue, 2017-12-31:
> One interesting thing I noticed on your branch is that you seem to do
the calibration points at the sensor XY location instead of the nozzle
location.  I'm not sure what the best approach is.  

For a delta XY/nozzle location is the best way... given that the user is also using some nozzle triggered probe, I believe there is no way to reliably calculates the probe XY offset during a calibration.

GruffyPuffy, 2017-12-31:
@KevinOConnor @megablue 
You might be right...the more I learn about this, the more I understand that I do not know that much. :) I need to dig into the maths again and think this through. I did do a calibration&test-print with what I have and it seemed pretty good. I did however not benchmark against not using the offsets of my probe. 
Well, see my branch of this as my experiments only. Just wanted to inform and thank you for the hard work on this project!
When I was thinking of this, I wanted to get the Z-height for where the probe actually hits as the error function only uses the Z-height for error. But, as this is actually offset from the nozzle, it might not be correct when minimizing the error. (Sometimes one tend to think too much...) I will do some tests where I remove the XY-offsets from my printer.cfg and see what happens...one good thing is that it is easy to test... :)
Happy new year!

KevinOConnor, 2018-01-20:
On Fri, Jan 19, 2018 at 01:47:05AM +0000, Matt Peterson wrote:
> @KevinOConnor Has something changed in master that breaks the probe branch? I rebased the probe branch onto master. When I do DELTA_CALIBRATE, it probes all the points and then right after the last point it prints `MCU 'mcu' shutdown: Command request` and never prints any results from calibration.

To diagnose something like this, I'd need to see the klippy.log file
as described at:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

That said, it's too painful to diagnose a merged branch, so please try
to reproduce on the original master or the probe branch.

-Kevin

KevinOConnor, 2018-01-29:
I've merged the work-probe-20170609 into the master branch.

closed 2018-01-29 by KevinOConnor