Klipper3d/klipper #20: Compiling fails on Arch-Linux

johannes-otto, 2017-06-26:
Hello,
I can't compile on my Arch Linux host. When I ssh into the PI, I can compile there. Here is my Terminal output:

>[johannes@Johannes-Desktop klipper]$ gcc --version
gcc (GCC) 7.1.1 20170621
Copyright (C) 2017 Free Software Foundation, Inc.
Dies ist freie Software; die Kopierbedingungen stehen in den Quellen. Es
gibt KEINE Garantie; auch nicht für MARKTGÄNGIGKEIT oder FÜR SPEZIELLE ZWECKE.

>[johannes@Johannes-Desktop klipper]$ make
  Building out/compile_time_request.o
Traceback (most recent call last):
  File "./scripts/buildcommands.py", line 374, in <module>
    main()
  File "./scripts/buildcommands.py", line 291, in main
    for req in data.split('\0'):
TypeError: a bytes-like object is required, not 'str'
make: *** [Makefile:83: out/compile_time_request.o] Fehler 1
[johannes@Johannes-Desktop klipper]$

KevinOConnor, 2017-06-26:
On Mon, Jun 26, 2017 at 04:34:23AM -0700, Johannes Otto wrote:
> Hello,
> I can't compile on my Arch Linux host. When I ssh into the PI, I can compile there. Here is my Terminal output:
>
> gcc (GCC) 7.1.1 20170621
> Copyright (C) 2017 Free Software Foundation, Inc. [...]

Thanks.  It looks like the first line of scripts/buildcommands.py
should use "python2" instead of "python".  I'll update it.

-Kevin

KevinOConnor, 2017-06-28:
I think this should be fixed as of d2fafbdd.

johannes-otto, 2017-06-28:
looks good. But in the Makefile PYTHON=python2 should also be set. With that it compiles.

KevinOConnor, 2017-06-28:
On Wed, Jun 28, 2017 at 07:38:28AM -0700, Johannes Otto wrote:
> looks good. But in the Makefile PYTHON=python2 should also be set. With that it compiles.

Thanks.  I've made that change as well (commit b7b368a1).

-Kevin

johannes-otto, 2017-06-28:
builds like a charm

closed 2017-06-28 by johannes-otto