Klipper3d/klipper #22: reconnect after printer hard reset

johannes-otto, 2017-06-27:
When I do a hard reset on the printer (power off & on), the port of the printer changes from /dev/ttyUSB0 to /dev/ttyUSB1.
Of course I then get a timeout while connecting. So I have to edit the config file and do a restart. Could it be a good idea  to implement some kind of probing for another port on a timeout?

KevinOConnor, 2017-06-27:
On Tue, Jun 27, 2017 at 10:48:45AM -0700, Johannes Otto wrote:
> When I do a hard reset on the printer (power off & on), the port of the printer changes from /dev/ttyUSB0 to /dev/ttyUSB1.

To avoid that problem, change the configured device from
"/dev/ttyUSB0" to it's corresponding static device in
/dev/serial/by-id/

-Kevin

johannes-otto, 2017-06-27:
Good idea, thanks

closed 2017-06-27 by johannes-otto