Klipper3d/klipper #24: Setup issues

smellzmore, 2017-07-02:
#

TunaLatte, 2017-07-02:
Hey

Post your Marlin steps per millimeter.
The simple formula I used is 1/yourmarlinstepspermillimter, then round it to the nearest 7th digit.

smellzmore, 2017-07-03:
Thank you! My steps per millimeter in Marlin are X88.88888888889, Y88.88888888889, Z1511.811023622047, and E136.611842. I probably can round things off but those settings are pretty accurate on prints.

TunaLatte, 2017-07-03:
I don't think stepper motors can achieve that level of accuracy, in my case printed objects dimensional accuracy was on par if not better than Marlin (due to less vibration on outer perimeters aswell).
Guess it should be: X0.01125 Y0.01125 Z0.0006615 E0.0073200 (x & y weren't rounded).

smellzmore, 2017-07-04:
Thank you for your help again! I will play with the configuration once I get a chance.

closed 2017-07-04 by smellzmore