Klipper3d/klipper #25: homing order configuration option

mhv-shared, 2017-07-09:
On my corexy, Y must be homed before X, otherwise it won't hit the stop. 

Obviously I can use `G28 Y G28 X G28 Z`, but it would be nice if `G28` or `G28 XYZ` would function safely as well, in case a misconfigured slicer was used.

It looks like this will be a simple change, if you agree it's worthwhile I'll open a PR.

KevinOConnor, 2017-07-17:
I have no objection to supporting it.  Is there a similar feature in Marlin?  If you have code available, I'd like to see it.

Thanks,
-Kevin

mhv-shared, 2017-07-17:
I think this is the relevant code in Marlin: https://github.com/MarlinFirmware/Marlin/pull/1957

I opened a PR #27.

KevinOConnor, 2017-07-28:
Thanks.  That PR looks like it changes the default homing order, but doesn't prevent an invalid homing operation.  If your printer doesn't home Y before X, does the head crash into the frame?  I'd be more concerned about verifying that a crash wont happen then making the G28 more user friendly.

Also, I think some users have printer that can only home the Z with a bed probe, and thus they have to move their X and Y to a set coordinate before homing Z.  I wonder if a similar mechanism could be used to support that setup.

-Kevin

KevinOConnor, 2018-01-17:
I've created a new test branch ( work-homing-override-20180116 ) that will allow one to define a series of g-code commands to run in place of G28 commands found in the normal input stream. It's activated by a new [homing_override] config section - see config/example-extras.cfg on the new branch for details.

I'd like to get feedback on this code.

git fetch ; git checkout origin/work-homing-override-20180116 ; sudo service klipper restart

KevinOConnor, 2018-01-29:
I've merged the work-probe-20170609 branch (which includes the work-homing-override-20180116 changes) into the master branch.

closed 2018-01-29 by KevinOConnor

StefanMeGit, 2018-05-12:
Sorry i still have a question about this... 

How is to use this? I tried a lot combinations like:
```
[homing_override]
gcode: G28Y \nG28X
```
or 
```
[homing_override]
gcode: G28Y G28X
```

But nothing gives me the Y before X homing. What did i wrong here?

mmone, 2018-05-12:
Had the same issue. Put each GCODE command in a separate line. And for some reason a whitespace is needed in front of every new line.

Like that: 

```
[homing_override]
gcode:
 G28 Y0
 G28 X0
 G28 Z0
```

StefanMeGit, 2018-05-12:
ahhhh i tried this also, but without space in front. Thanks! I ll try it.