Klipper3d/klipper #26: Homing after print

outlookhazy, 2017-07-14:
I actually have a log file this time!  
I've seen this before, but this is the first time i've caught it in the act...

**Sometimes** after a print, only the X/Y (front towers) home on the G28 that ends the G-Code. The rear tower stays still, typically resulting in a print head hanging out in front of the printer.
I have a feeling it has something to do with the relative movement stuff again, but I haven't been able to spot the common factor.

[klippy.zip](https://github.com/KevinOConnor/klipper/files/1149812/klippy.zip)

KevinOConnor, 2017-07-17:
Nothing in the log looks suspicious.  Generally when one axis does not home, it is the result of an endstop switch error.  Did you get a "Failed to home stepper c: Timeout during endstop homing" or "Endstop c still triggered after retract" error in the g-code terminal window during the event?  If you run QUERY_ENDSTOPS after an event, does it show the correct results?

outlookhazy, 2017-07-17:
I would assume the same, and I don't recall seeing a message like that. Immediately after this happened and I grabbed the log, I restarted klipper and did another home and all 3 worked as expected (I didn't touch anything physical). The only time I've ever seen this (and it has happened a few times) is directly following a print. No manually-issued home has ever behaved this way.  Based on that, I have a nagging suspicion that sometimes the switch to relative/absolute position being issued too close to a move command results in a race condition - but that is 100% speculation based on absolutely nothing.

KevinOConnor, 2017-07-18:
On Mon, Jul 17, 2017 at 04:52:21PM +0000, Charlie wrote:
> I would assume the same, and I don't recall seeing a message like
> that. Immediately after this happened and I grabbed the log, I
> restarted klipper and did another home and all 3 worked as expected [...]

That sounds like it could be interference on the switch wire.  Klipper
could be enhanced to sample the switch multiple times to ensure it
truly has triggered.

Is it always the C stepper that fails?

>Based on
> that, I have a nagging suspicion that sometimes the switch to
> relative/absolute position being issued too close to a move command
> results in a race condition - but that is 100% speculation based on
> absolutely nothing.

Delta homing is implemented in klippy/delta.py:home() and it always
issues movement commands for all three steppers.  So, I don't see how
anything on the g-code level could cause the issue you are reporting.

The latest version of the code will now log g-code error responses.
I'd upgrade so that any further logs will include info on whether or
not the homing operation itself reported an error to the g-code
terminal.

-Kevin

outlookhazy, 2017-07-18:
Yes, always tower C - but with seemingly no repeat-ability. I'll take another look at how i'm doing the signaling - I thought it was a pull-up on the MCU going to a NO grounding on the switch - which would require some pretty heavy interference... But I may be remembering wrong, i'll have to double check later - maybe it needs a cap

KevinOConnor, 2017-08-15:
Are you still seeing this issue?

outlookhazy, 2017-08-18:
Yes, but still intermittently - and most of the time it's after a long print so I can't snap a log right after. I've been busy with other stuff and it's not really harmful so I haven't dug deeper

krepa098, 2017-10-01:
I'm having the same issue. Although in my case it's the A tower getting stuck.

It starts homing normally but then stops for no reason. It does however retract and rehome after both the Y and Z tower hit their endstop. There is nothing suspicious in the log apart from "High clock drift".

KevinOConnor, 2017-10-01:
On Sun, Oct 01, 2017 at 09:44:53AM -0700, krepa098 wrote:
> I'm having the same issue. Although in my case it's the A tower
> getting stuck.
> 
> It starts homing normally but then stops for no reason. It does
> however retract and rehome after both the Y and Z tower hit their
> endstop. There is nothing suspicious in the log apart from "High
> clock drift".

That sounds like electrical noise on the endstop pin.

I implementing some basic noise filtering in software (commit
ece1f71c).

Can you pull the latest code, reflash the micro-controller, and see if
the problem goes away?

-Kevin

krepa098, 2017-10-02:
EM noise makes sense. My A tower endstop wires are pretty close the power wires of the extruder motor. The retraction before homing may cause the issue.

Today, I completed two 1h prints without any issues. As such, your changes definitely look promising.  Thank you.

KevinOConnor, 2017-10-06:
Okay, thanks.  I'm going to close this item on the understanding that electrical noise was the problem and that the new support addresses it.  If the problem persists, please re-open.

closed 2017-10-06 by KevinOConnor