Klipper3d/klipper #28: Use an alternative UART on the RAMPS stack?

MortenGuldager, 2017-08-09:
Is it possible to configure the klipper firmware touse a different UART on the RAMPS stack? 

I would very much prefer not to use the sloppy USB interfaces but rather go directly uart-to-uart between the raspi and atmega.

menuconfig didn't reveal it as a configurable parameter, should I look elsewhere? 

If it is not implemented, this might be seen  as a feature request/suggestion.

MortenGuldager, 2017-08-09:
Very easy. Hacked the serial.c as you suggested, made a section pointing to uart2 and bingo, it works. Seems a bit hackish to redefine the registers, but hey..

I might try to clean up the code and make you pull request, still not quite sure I fully understand whats going on there.

And yes, level shifters are indeed needed between RPi and ramps stack.

KevinOConnor, 2017-08-09:
On Wed, Aug 09, 2017 at 08:32:06PM +0000, Morten Guldager wrote:
> Very easy. Hacked the serial.c as you suggested, made a section
> pointing to uart2 and bingo, it works. Seems a bit hackish to
> redefine the registers, but hey..

Ideally there would be a Kconfig option (with appropriate selections
for each type of chip) and the code would not redefine the register
names, but just map them to new names (eg, SERIAL_UDR instead of
UDR0).  Only the four functions immediately after the define block use
those definitions.

-Kevin

cfelicio, 2017-08-21:
That sounds pretty cool. Does it provide better speed or just better stability? 

@MortenGuldager If you can get a tutorial on how to do this, that would be highly appreciated! :)

Thanks!

dragonnn, 2017-11-10:
@MortenGuldager can you post you hacked serial.c? I am trying to do the same but I got errors:
`INFO:root:Starting serial connect
WARNING:root:got {'count': 40, '#receive_time': 1203.450843292, '#name': 'identify', '#sent_time': 0.0, 'offset': 0}
WARNING:root:Timeout on serial connect
`

MortenGuldager, 2017-11-11:
For those of you asking for my crude hack, here is what my git diff tells me about it:

```
pi@octopi014:~/klipper $ git diff src/avr/serial.c
diff --git a/src/avr/serial.c b/src/avr/serial.c
index 10b9599..432dd50 100644
--- a/src/avr/serial.c
+++ b/src/avr/serial.c
@@ -24,9 +24,44 @@ static uint8_t transmit_pos, transmit_max;
  * Serial hardware
  ****************************************************************/
 
+
+#undef UCSR0A
+#undef UCSR0B
+#undef UCSR0C
+#undef UBRR0
+#undef UDR0
+#undef UCSZ01
+#undef UCSZ00
+#undef U2X0
+#undef RXEN0
+#undef TXEN0
+#undef RXCIE0
+#undef UDRIE0
+#undef USART0_RX_vect
+#undef USART0_UDRE_vect
+
+#define UCSR0A UCSR2A
+#define UCSR0B UCSR2B
+#define UCSR0C UCSR2C
+#define UBRR0 UBRR2
+#define UDR0 UDR2
+
+#define UCSZ01 UCSZ21
+#define UCSZ00 UCSZ20
+#define U2X0 U2X2
+#define RXEN0 RXEN2
+#define TXEN0 TXEN2
+
+#define RXCIE0 RXCIE2
+#define UDRIE0 UDRIE2
+#define USART0_RX_vect USART2_RX_vect
+#define USART0_UDRE_vect USART2_UDRE_vect
+
+
 DECL_CONSTANT(SERIAL_BAUD, CONFIG_SERIAL_BAUD);
 
 // Define serial port registers on AT90USB1286
+/*
 #if !defined(UCSR0A) && defined(UCSR1A)
 #define UCSR0A UCSR1A
 #define UCSR0B UCSR1B
@@ -49,6 +84,7 @@ DECL_CONSTANT(SERIAL_BAUD, CONFIG_SERIAL_BAUD);
 #define USART0_RX_vect USART_RX_vect
 #define USART0_UDRE_vect USART_UDRE_vect
 #endif
+*/
 
 void
 serial_init(void)
```

I would appreciate if someone a bit more structured than me cleaned it up so it could enter the official klipper codebase.

dragonnn, 2017-11-11:
Thanks it works but I find out that you can use direct UART0 from AUX0, so for my it isn't necessary.

mikedahlgren, 2018-02-19:
Does anyone know if anyone is building the hardware required for this?  Best I can tell this would be ideal but is not being produced.  https://github.com/j-laird/Printrboard-Pi-Link

TunaLatte, 2018-02-20:
@mikedahlgren checkout RAMBO Einsy 1.1 (used in prusa mk3) I think it has rpi zero link underneath.

KevinOConnor, 2018-02-25:
I'm going to close this issue as it doesn't appear anyone is actively working on implementing it.

closed 2018-02-25 by KevinOConnor

ETE-Design, 2019-11-19:
@KevinOConnor Will this give any performance, or just free up an USB?

KevinOConnor, 2019-11-20:
Selecting the UART on the AVR platform was added long ago.  It's not at all related to performance.

-Kevin