Klipper3d/klipper #29: "restart_method: command" - deadlocks

MortenGuldager, 2017-08-09:
Configured `restart_method: command` in the `[mcu]` section in `printer.cfg`

But when I do a `FIRMWARE_RESTART` the atmega just enters a crazy deadlock where it blinks forever. 

Even hitting the reset button doesn't help, it just continue blinking when I release the button again.
Flashing new firmware is impossible too.

Only thing to clear the mishap is a power cycle.

KevinOConnor, 2017-08-09:
On Wed, Aug 09, 2017 at 01:43:39PM -0700, Morten Guldager wrote:
> Configured `restart_method: command` in the `[mcu]` section in `printer.cfg`
>
> But when I do a `FIRMWARE_RESTART` the atmega just enters a crazy deadlock where it blinks forever.
>
> Even hitting the [...]

Alas, that's not a Klipper bug - it's a bug in some old versions of
the atmega bootloader.  I had the same thing on my atmega2560 before
updating it to a newer version.  I reprogrammed mine with the software
at:

https://github.com/arduino/Arduino-stk500v2-bootloader

-Kevin

MortenGuldager, 2017-08-12:
Confirmed, the problem was board related, tried with a newer board and the "command" reset worked just fine.

closed 2017-08-12 by MortenGuldager