Klipper3d/klipper #30: Single Nozzle / Multi Extruder

cfelicio, 2017-08-21:
Hi There,

I see that there is a thread for multiple extruders being tested, I wonder if it would be possible to have multiple extruders with a single nozzle? Similar to the e3d cyclops.

Thanks!

KevinOConnor, 2017-08-29:
Using a single nozzle with multiple extruders was kept in mind when designing the multiple extruder code.  Let us know if you have the hardware and would like to help test new software revisions.  Lets use the existing issue #6 to track this.

closed 2017-08-29 by KevinOConnor

FeuerVogel666, 2018-12-31:
I have build my Ender 3 up to a single Nozzle and two Extruders.

Hardware:
Arduino 2650 Mega
Ramps converted to 24 Volts

So I would love to test with you.
But I have not very much knowledge in Coding.

manu7irl, 2019-01-03:
I got an geeetech a10m, with dual extruder shared heater and single nozzle.
So this this config in marlin can create mixing or partial mixing of 2 filaments.
can a function like this be used?
like extruder1 gives 50% extrusion and extruder2 gives also 50% or any combinations possible of percentage?

KevinOConnor, 2019-01-03:
@manu7irl - See issue #950.  It would require software development to support it.

joonadec, 2019-02-21:
@KevinOConnor am thinking of giving this a go, by trying to implement Marlin's M165 "Set Mix" gcode...would it be easier to modify gcode.py or write a module?

Am thinking this would just need a handler for the M165 command, storing the ratios, then either modifying the extruder move lengths via ToolHead.move() or modifying extrude_factor.

Thoughts?

hevilp, 2019-11-02:
@joonadec Hi, did you reach anything?

gcurtis79, 2020-02-02:
I'm trying to configure this, extruder and extruder1 both use analog13 (single nozzle) for the thermistor, but the firmware is complaining:
```
Recv: // pin analog13 used multiple times in config
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Printer is halted
Recv: !! pin analog13 used multiple times in config
```
Is there a way to make klipper accept both extruders on the same analog pin?

jalanjarosz, 2020-02-02:
So, you are trying to use like a Y-splitter or a 2in-1out non-mixing nozzle?

Klipper has the ability to have multiple extruders per single hot-end.  There is more info in the: https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg

Here is some basic info on my benchtop Re-ARM with a prototoype RAMPS-32 board:

[extruder]
step_pin: P2.8 #ar36
dir_pin: P2.13 #ar34
enable_pin: !P4.29 #!ar30
step_distance: .0025
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: P2.5 #ar10
sensor_type: beta4267
sensor_pin: P0.23  #(A13)
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 275

[extruder1]
step_pin: P2.0 #ar26
dir_pin: P0.5 #ar28
enable_pin: !P0.4 #!ar24
step_distance: .0025
nozzle_diameter: 0.400
filament_diameter: 1.750
shared_heater: extruder

jalanjarosz, 2020-02-02:
Based on the design of the extruder you have chosen, I think you just need to define a single extruder an a servo.
All your tool changing commands would be done in your slicer with G-Code.

gcurtis79, 2020-02-02:
But I can't figure out how to make the stepper operate in reverse when switching to [extruder1].

When the servo switches over, it needs to run the stepper in the opposite direction. That's all I really need.

Maybe I can set a dummy pin for the direction, and manually set the direction pin during the tool change scripts. I'll look into that.

gcurtis79, 2020-02-03:
I got it.

In my [extruder] section, I set dir_pin: ar22 (an unused pin on the RAMPS 1.4)
```
[extruder]
dir_pin: ar22
```
Then set up a  custom output_pin:
```
[output_pin extruder_dir]
pin: ar28
```
Then set up my tool change script to swap the servo and direction pin value:
```
[gcode_macro T0]
gcode:
  SET_SERVO SERVO=extruder_servo angle=150
  SET_PIN PIN=extruder_dir value=0

[gcode_macro T1]
gcode:
  SET_SERVO SERVO=extruder_servo angle=30
  SET_PIN PIN=extruder_dir value=1
```
Now to set up the rest of the tool change scripting for the full filament swap.