Klipper3d/klipper #31: Can't configure needed step size for my printer

Megamannen, 2017-08-27:
Found this project today and wanted to try it out immedietly. I have configured X, Y, Extruder, Heating, Bed
 
But it seems like your implementation can't handle my step_distance for Z

My step_distance for Z should be somewhere around: .000155
When I set I get this immidietly when trying any move on Z:
// Firmware shutdown: Timer too close

If I set it to 0.00155 (10x higher) it works correctly, except that a raise of 100mm gives me 10mm :P

KevinOConnor, 2017-08-27:
Another user recently reported a similar issue (which is a confirmed bug due to an obscure issue with step compression).  In order for me to confirm this is the same issue, please attach the full contents of the /tmp/klippy.log file that contains the error.

Megamannen, 2017-08-27:
You can confirm it also by setting your step_distance, but sure:

[klippy.txt](https://github.com/KevinOConnor/klipper/files/1254733/klippy.txt)

KevinOConnor, 2017-08-27:
That step_distance works for me on my local tests.  The log you provided is showing the error from a previous session.  Please run FIRMWARE_RESTART from the octoprint console, issue a G28 Z0, and then attach the /tmp/klippy.log so that the log contains the error during the home itself.

Megamannen, 2017-08-27:
I stopped the service
removed the klippy.log
started the service
triggered the error by homing Z
How can it be from another session?

Now I only did a firmware_restart per your instruction, attached new log.

Got another error now (might have been to quick to home):
Row 572: Failed to home stepper z: MCU is shutdown

But same error as always on:
Row 588: Firmware shutdown: Timer too close
[klippy2.txt](https://github.com/KevinOConnor/klipper/files/1254740/klippy2.txt)

Megamannen, 2017-08-27:
Ok, made a less chatty log now that failed before Z-move even, but still related since it doesn't fail when I change step_distance to an higher value

[klippy.no-z-move.less-chatty.txt](https://github.com/KevinOConnor/klipper/files/1254744/klippy.no-z-move.less-chatty.txt)

KevinOConnor, 2017-08-28:
On Sun, Aug 27, 2017 at 09:53:37PM +0000, Megamannen wrote:
> I stopped the service
> removed the klippy.log
> started the service
> triggered the error by homing Z
> How can it be from another session?

There is software running on both the raspberry-pi and the
micro-controller.  The steps above only clear the error from the
software running on the raspberry-pi.  Use the FIRMWARE_RESTART
command to clear the error state in both locations.

> Now I only did a firmware_restart per your instruction, attached new log.
> 
> Got another error now (might have been to quick to home):
> Row 572: Failed to home stepper z: MCU is shutdown

Thanks - this log shows the error.  It is a bug in timing due to the
large number of steps (~2 million) issued during the homing of your Z
axis.  I need to think about how to fix this, but in the interim, I
think you should be able to work around it by adding
"buffer_time_start: 0.600" to the "[printer]" section of the config
file.  Let me know if that does work.

-Kevin

Megamannen, 2017-08-28:
buffer_time_start did the trick, though the motors didn't sound happy when stepping towards Z-home (no problem I'm sure)

Megamannen, 2017-08-28:
False alarm, worked twice before I turned on the heaters, now it doesn't comply. Klippy log attached 
[klippy.with-buffer.txt](https://github.com/KevinOConnor/klipper/files/1257767/klippy.with-buffer.txt)

KevinOConnor, 2017-08-29:
Alas, you've hit the recently reported step size bug I mentioned earlier (the obscure step compression problem).  Until we can get that fixed, you can workaround it by returning to a step size of .000155 (instead of using .000156).

Megamannen, 2017-08-29:
Hehe, ok, thanks for the quick responses. Looks like i might get the first print out today then :)

KevinOConnor, 2017-08-29:
Could you pull down the latest code and retest?  I believe both issues should now be fixed.  (You should remove the buffer_time_start from the config, and you should be able to use .000156.)

Thanks,
-Kevin

Megamannen, 2017-08-30:
Doesn't seem to crash now! (tried with the odd step size and no buffer_time_start)

Still have a problem though... Is there any special logic on how it steps when homing, or is it only the speed difference? Z-motors really sounding retarded and gives up before getting to home. It's like they are missing "strength" or something.
After it has managed to home it steps beautifully, any idea?

KevinOConnor, 2017-08-31:
On Wed, Aug 30, 2017 at 05:53:03PM +0000, Megamannen wrote:
> Doesn't seem to crash now! (tried with the odd step size and no buffer_time_start)

Thanks!

> Still have a problem though... Is there any special logic on how it steps when homing, or is it only the speed difference? Z-motors really sounding retarded and gives up before getting to home. It's [...]

My first guess is that your Z motor does not have sufficient torque to
drive the axis at 5mm/s.  Keep in mind that with your Z step
precision, Marlin would be unable to drive the Z faster than 1-2mm/s.
With .000155 mm per step, 5mm/s is ~32000 steps per second - Klipper
can do that easily, but it's way past the ability of Marlin.

So, you might want to try reducing your maximum Z speed.

If you think that's not it or if anything else unusual occurs that
Klipper does not raise an error for, then after the event you should
issue an M112 command in the Octoprint terminal.  This causes Klipper
to implement a shutdown, and it will cause the host software to log
all recent commands that it received and generated.  Then post the
full log.

-Kevin

KevinOConnor, 2017-08-31:
On Wed, Aug 30, 2017 at 05:53:03PM +0000, Megamannen wrote:
>Is there any special logic on how it steps when homing, or is it only the speed difference?

To answer your specific question - no, the same step generation code
is used for both regular moves and homing moves: gcode.py:cmd_G1() ->
toolhead.move() vs homing.py:home() -> toolhead.move()

-Kevin

Megamannen, 2017-08-31:
Since speed is the only difference your theory matches mine... not enough juice.

I have also investigated my controller now and I have decreased the number of microsteps to 8 from 32 on the Z axis. Also realized that the Z-motor was connected to the lead screw at 1:1 ratio and I measured the pitch to 1mm, so my so my perverse value of 0.000156 was clearly wrong.

closed 2017-08-31 by Megamannen