Klipper3d/klipper #32: Error: Move exceeds maximum extrusion cross section

cfelicio, 2017-09-02:
[dbot-std-corexy.txt](https://github.com/KevinOConnor/klipper/files/1271912/dbot-std-corexy.txt)
So, I installed Klipper on my D Bot, with Ramps 1.4 and Raspi 3.

I'm attaching my configuration file. 

I tried to print a 3d benchy, and I noticed while mostly ok, the printer does make some weird moves (maybe I can make a video to give an idea), and after printing a few layers, I got this error:

```
Send: N4314 G1 X137.543 Y156.776 E0.0181*107
Recv: ok
Send: N4315 G1 X125.022 Y144.255 E0.5300*98
Recv: ok
Send: N4316 G1 X125.156 Y144.137 E0.0054*99
Recv: ok
Send: N4317 G1 X125.235 Y144.089 E0.0028*107
Recv: ok
Send: N4318 G1 X125.345 Y144.043 E0.0036*107
Recv: ok
Send: N4319 G1 X125.369 Y144.036 E0.0007*100
Recv: ok
Send: N4320 G1 X138.147 Y156.815 E0.5409*105
Recv: ok
Send: N4321 M105*19
Recv: ok T0:205.0 /205.0 B:69.1 /70.0
Send: N4322 G1 X138.752 Y156.854 E0.0181*108
Recv: ok
Send: N4323 G1 X125.872 Y143.974 E0.5452*101
Recv: ok
Send: N4324 G1 X126.403 Y143.940 E0.0159*103
Recv: ok
Send: N4325 G1 X139.356 Y156.892 E0.5483*98
Recv: ok
Send: N4326 G1 X139.961 Y156.931 E0.0181*101
Recv: !! Move exceeds maximum extrusion cross section: 139.356 156.892 0.680 [138.590]
Changing monitoring state from 'Printing' to 'Error:  Move exceeds maximum extrusion cross section: 139.356 156.892 0.680 [138.590]\x0a'
```

Also here is the klippy.log

```
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.4.0-157-g026b9c3-dirty'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.9 (default, Mar  8 2015, 00:52:26) \n[GCC 4.9.2]'
Start printer at Sat Sep  2 02:17:05 2017 (1504318625.9 10.6)

Starting serial connect
Stats 4905.4: gcodein=0 print_time=0.000 buffer_time=0.000 print_stall=0  mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000
Loaded 51 commands (v0.4.0-157-g026b9c3-20170902_024250-octopi)
MCU config: ADC_MAX=1023 PWM_MAX=255 CLOCK_FREQ=16000000 SOFT_PWM_MAX=256 SERIAL_BAUD=250000 MCU=atmega2560 STATS_SUMSQ_BASE=256
Sending printer configuration...
Configured (640 moves)
Stats 4906.4: gcodein=0 print_time=0.000 buffer_time=0.000 print_stall=0 bytes_write=638 bytes_read=2244 bytes_retransmit=0 bytes_invalid=0 send_seq=80 receive_seq=80 retransmit_seq=0 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stal$
High clock drift! Now 15926943:16001600 was 15998400:16001600
Stats 4907.4: gcodein=0 print_time=0.000 buffer_time=0.000 print_stall=0 bytes_write=644 bytes_read=2312 bytes_retransmit=0 bytes_invalid=0 send_seq=81 receive_seq=81 retransmit_seq=0 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stal$
High clock drift! Now 15926191:16001600 was 15926943:16001600
Stats 4908.4: gcodein=41 print_time=0.000 buffer_time=0.000 print_stall=0 bytes_write=650 bytes_read=2406 bytes_retransmit=0 bytes_invalid=0 send_seq=82 receive_seq=82 retransmit_seq=0 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 sta$
Stats 4909.4: gcodein=46 print_time=0.000 buffer_time=0.000 print_stall=0 bytes_write=656 bytes_read=2526 bytes_retransmit=0 bytes_invalid=0 send_seq=83 receive_seq=83 retransmit_seq=0 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 sta$
High clock drift! Now 15871886:16001600 was 15926191:16001600
```

KevinOConnor, 2017-09-02:
On Sat, Sep 02, 2017 at 04:58:33AM +0000, cfelicio wrote:
[...]
> Also here is the klippy.log
[...]

Please attach the full klippy.log file.

-Kevin

KevinOConnor, 2017-10-06:
If the problem persists, please reopen this issue and attach the full klippy.log file.

closed 2017-10-06 by KevinOConnor