Klipper3d/klipper #35: Add support for Melzi boards

saullipton, 2017-10-05:
Hi, 
I am super excited about your solution, and have exactly the hardware and requirements for it - with one exception. I have a Melzi with an ATMEGA1284P chip. 
I tried to build and load the solution with the ATMEGA1280 settings, but this didn't work. I am having troubles getting the RPi to communicate with the board after the firmware is loaded and OctoPi is communicating with the klipper service. Basically the serial communication times out. Klippy.log:

Timeout on serial connect

I am at least marginally able to work with Python, and am actually a CNC engineer, so I'd be happy to work on getting the board running myself if you could point me in the right direction. Once that happens, I'd be interested in seeing what I could contribute to improving kinematics, etc.

Thanks!

KevinOConnor, 2017-10-05:
On Thu, Oct 05, 2017 at 03:35:25PM +0000, saullipton wrote:
> Hi,
> I am super excited about your solution, and have exactly the hardware and requirements for it - with one exception. I have a Melzi with an ATMEGA1284P chip.
> I tried to build and load the [...]

I added the 1284p definitions - can you grab the latest code and
retry?  (Be sure to recompile and flash the firmware with the new
atmega1284p option.)

> Timeout on serial connect
>
> I am at least marginally able to work with Python, and am actually a CNC engineer, so I'd be happy to work on getting the board running myself if you could point me in the [...]

Great!

-Kevin

saullipton, 2017-10-05:
Thanks, that was quick! I have recompiled the firmware, however now I am unable to load the firmware onto the Arduino. I was never able to do it with your "make flash FLASH_DEVICE=/dev/ttyUSB0" script, but instead using the bash command:

avrdude -p m1284p -c stk500v1 -P /dev/ttyUSB0 -b57600 -U flash:w:klipper.elf.hex

Now, I get the error:

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x8c

I expect the make flash command uses the wrong (stk500v2) programmer for this board. 

I will try a few more methods to flash the firmware into the board and let you know how it goes.
Thanks!

KevinOConnor, 2017-10-05:
On Thu, Oct 05, 2017 at 06:57:42PM +0000, saullipton wrote:
> Thanks, that was quick! I have recompiled the firmware, however now I am unable to load the firmware onto the Arduino. I was never able to do it with your "make flash FLASH_DEVICE=/dev/ttyUSB0" [...]

I'm not familiar with that error.  Be sure to stop klipper (sudo
service klipper stop) prior to attempting to flash the device.

Also, in my earlier commit of 1284p support I neglected to add the
host pin mappings for the 1284p.  The latest code now has the hardware
pin mappings in place.  I did not add Arduino aliases - I'm not sure
if the Arduino aliases make sense or what they would be.

-Kevin

Linusoderlind, 2017-10-10:
Could you add your configuration as an example file for Melzi as for Ramps and Rumba?

saullipton, 2017-10-10:
Linus,
I will do that for sure. However, I am still fighting my Atmel firmware upload issues. I will use another Arduino as ISP tomorrow and see where I get. 
Thanks,
Saul

Linusoderlind, 2017-10-11:
Right, I could not help myself and wrote my own config.

I to had problems with flasing the Melzi with 
`make flash FLASH_DEVICE=/dev/ttyUSB0`
but
`avrdude -p atmega1284p -c avrisp -P /dev/ttyUSB0 -b 19200 -U flash:w:klipper.elf.hex`
flashed fine.

I have yet to print anything but all axis homes and the extruder + hotend works. I have a TronXY X1 so I do not have a heated bed.

KevinOConnor, 2017-10-11:
On Wed, Oct 11, 2017 at 08:58:22PM +0000, Linusoderlind wrote:
> Right, I could not help myself and wrote my own config.

Thanks.  Could you attach your config?  It would be good to add a
melzi example to the config files.

> I to had problems with flasing the Melzi with 
> `make flash FLASH_DEVICE=/dev/ttyUSB0`
> but
> `avrdude -p atmega1284p -c avrisp -P /dev/ttyUSB0 -b 19200 -U flash:w:klipper.elf.hex`
> flashed fine.

Do you need the "-b 19200"?  (The default is normally fine.)

-Kevin

Linusoderlind, 2017-10-12:
I do not need the -b 19200. 

As I do not have a heated bed I have not tried that part of the config.
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/1379009/printer.cfg.txt)

saullipton, 2017-10-12:
OK, I have made quite a bit of progress and am ready to get down to the nitty gritty. 
Here are my hardware specifics:
 I have an Alunar M505 board, which uses an Anet3D motherboard. This is a Melzi clone, from what I can gather. The chip is an Atmel 1284P. 

To get to this point:
- I had to burn a new bootloader onto the chip to get reliable firmware loading working. The optiboot bootloader works great. It uses a 115200 baud rate for firmware downloads. 
- My firmware download command is this now:
avrdude -c arduino -p m1284p -P /dev/ttyUSB0 -b 115200 -U flash:w:klipper.elf.hex:i
- I have used Linus' printer.cfg and also tried my own version. 

I have gotten the latest klipper code, and compiled the firmware for the 1284p chip, however I am still unable to have the RPi host connect to the ATMega chip. I have attached my klippy.log. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/1379833/klippy.log)

Do you have any suggestions for getting more verbose diagnostics on the host <> printer communications? 
Thanks,
Saul

PS. Here is the source I have used to configure my pinouts.  [http://reprap.org/wiki/Melzi](http://reprap.org/wiki/Melzi) Would this be for the PA- or ar- style pin assignments in printer.cfg? I suspect ar-.

> Melzi Arduino Pin Numbers
> #define X_STEP_PIN         15
> #define X_DIR_PIN          21
> #define X_MIN_PIN          18
> 
> #define Y_STEP_PIN         22
> #define Y_DIR_PIN          23
> #define Y_MIN_PIN          19
> 
> #define Z_STEP_PIN         3
> #define Z_DIR_PIN          2
> #define Z_MIN_PIN          20
> 
> #define E0_STEP_PIN         1
> #define E0_DIR_PIN          0
> 
> #define LED_PIN            27
> 
> #define FAN_PIN            4 
> 
> #define HEATER_0_PIN       13 // extruder
> 
> #define HEATER_BED_PIN     10 // bed (change to 12 for breakout pin on header)
> #define X_ENABLE_PIN       14
> #define Y_ENABLE_PIN       14
> #define Z_ENABLE_PIN       26
> #define E0_ENABLE_PIN      14
> 
> #define TEMP_0_PIN          7   // Analogue pin
> #define TEMP_BED_PIN        6   // Analogue pin
> #define SDSS               31
> 
> #define SLAVE_CLOCK	   16

KevinOConnor, 2017-10-12:
> I have gotten the latest klipper code, and compiled the firmware for the 1284p chip, however I am still unable to have the RPi host connect to the ATMega chip. I have attached my klippy.log.

You have an extra zero in the baud field of your printer.cfg.

-Kevin

saullipton, 2017-10-12:
Womp womp. Haha thanks.

KevinOConnor, 2017-10-12:
On Thu, Oct 12, 2017 at 03:52:49AM -0700, Linusoderlind wrote:
> I do not need the -b 19200. 
> 
> As I do not have a heated bed I have not tried that part of the config.
> [printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/1379009/printer.cfg.txt)

Thanks.  I added a sample "config/generic-melzi.cfg" config file based
on the pin mappings from your config (commit 3033b037).

-Kevin

saullipton, 2017-10-12:
Update: Looks like everything is working swimmingly. I am printing out a test print now, looks great. Here is my config: 
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/1380665/printer.cfg.txt)
This is for a standard Alunar M505 / Anet A8, except with changed X step direction and extruder settings (for an E3Dv6 upgrade). The added heated bed is the only major change from Linus' cfg.
I think we can consider this request closed, and work on further items elsewhere.

THANKS!
Saul

closed 2017-10-12 by saullipton

rumlyne, 2020-04-22:
For my old Zonestar P802Q it worked with:
sudo avrdude -c arduino -p m1284p -P /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A104LXET-if00-port0 -U flash:w:out/klipper.elf.hex:i -v -b 57600