Klipper3d/klipper #36: "ADC module not enabled" on BeagleBone Black with CRAMPs and a couple other errors

prcdslnc13, 2017-10-18:
I just setup my BBB with clipper and when I got to the check area after setting up OctoPrint I get the following three errors.

After boot: 
"MCU 'mcu' shutdown: ADC module not enabled"

After doing recommended "FIRMWARE_RESTART" I still get that in the log but the outputted error in OctoPrint is:
ecv: // MCU 'mcu' shutdown: Rescheduled timer in the past
Recv: // This generally occurs when the micro-controller has been
Recv: // requested to step at a rate higher than it is capable of
Recv: // obtaining.

Not sure where to go from those two errors. 

Here is my log:
[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/1395221/klippy.log.gz)

KevinOConnor, 2017-10-18:
On Wed, Oct 18, 2017 at 08:36:28AM -0700, prcdslnc13 wrote:
> I just setup my BBB with clipper and when I got to the check area after setting up OctoPrint I get the following three errors.
> 
> After boot: 
> "MCU 'mcu' shutdown: ADC module not enabled"

Normally the ADC module is loaded by the "cape manager".  There's no
harm in loading it manually though, so I updated the start scripts to
do that.  Pull the latest code, rerun the install-beaglebone.sh
script, and then reboot.  It should be fixed.

> After doing recommended "FIRMWARE_RESTART" I still get that in the log but the outputted error in OctoPrint is:
> ecv: // MCU 'mcu' shutdown: Rescheduled timer in the past
> Recv: // This generally [...]

This error was due to the ADC module not being loaded.  I've fixed it
as well (though you should no longer see the ADC error).

There hasn't been a lot of testers with a CRAMPS board - let me know
your results.

-Kevin

prcdslnc13, 2017-10-20:
Alright that fixed the issue. I can get past the status messages but have other issues to report that Ill make a new issue for. Thanks!

closed 2017-10-20 by prcdslnc13