Klipper3d/klipper #37: Temp readings off about 50% and no movement or switch signals

prcdslnc13, 2017-10-20:
Hotend and bed thermistors are reading about 7deg C when they should be reading about 22. 
The Hotend: ATC Semitec 104GT-2
The Bed: EPCOS 100K B57560G104F

[image]

When I trigger M119 the XY list untriggered and Z lists triggered. Whether switches are triggered or even plugged in. 

With movement, it send the movement signal, but I am not getting any movement or enabling of the motors. 

I'm using the CRAMPS 2.1 and the schematic matches up to the pinout listed in the config.

KevinOConnor, 2017-10-21:
I think what's happening here is that the pin configuration is not
being setup properly.  This hasn't been an issue for other users (in
particular the replicape users) because the cape manager loads the pin
mappings automatically for them.

On Thu, Oct 19, 2017 at 09:28:26PM -0700, prcdslnc13 wrote:
> Hotend and bed thermistors are reading about 7deg C when they should be reading about 22. 
> The Hotend: ATC Semitec 104GT-2
> The Bed: EPCOS 100K B57560G104F

Not sure why this would be.  Can you try compiling the code for a
"linux process" and flash it - as described for the replicape in the
beaglebone.md file and then add the following to the printer.cfg file:
```
[mcu host]
serial: /tmp/klipper_host_mcu
```
and change "sensor_pin: P9_36" to "sensor_pin: host:analog5" and
"sensor_pin: P9_33" to "sensor_pin: host:analog4".  Finally, run "sudo
reboot" and see if the temperatures look okay on the next boot.

> When I trigger M119 the XY list untriggered and Z lists triggered. Whether switches are triggered or even plugged in. 

If you run the following from the command line, do the switches start
to work?

```
sudo config-pin overlay cape-universala
sudo config-pin P8_8 in+
sudo config-pin P8_10 in+
sudo config-pin P9_13 in+
```

> With movement, it send the movement signal, but I am not getting any movement or enabling of the motors. 
> 
> I'm using the CRAMPS 2.1 and the schematic matches up to the pinout listed in the config. 

It looks like the CRAMPS board has (oddly) two enable lines that must
be set to operate the printer.  If you add the following to your
printer.cfg does it allow the steppers and heaters to work?
```
[static_digital_output force_machine_enable_on]
pins: P9_23
```

-Kevin

prcdslnc13, 2017-10-23:
Followed the steps with no change in behavior. But may have found the issue. When I run the config-pin commands I get this for each of them:

`P9_13 pinmux file not found!
WARNING: GPIO pin not exported, cannot set direction or value!
bash: /sys/devices/platform/ocp/ocp*P9_13_pinmux/state: No such file or directory
Cannot write pinmux file: /sys/devices/platform/ocp/ocp*P9_13_pinmux/state`

Ill look more at it tomorrow.

prcdslnc13, 2017-10-31:
Hey Kevin,
I did a fresh git pull last night and have poked around some but I haven't made much progress. Do you have any ideas on this one? I can find a pinmux file for Machinekit but I'm not sure if thats what I need.

KevinOConnor, 2017-10-31:
On Tue, Oct 31, 2017 at 07:56:52AM -0700, prcdslnc13 wrote:
> I did a fresh git pull last night and have poked around some but I
> haven't made much progress. Do you have any ideas on this one?

What Beaglebone board are you using?  What OS did you install?

>I can
> find a pinmux file for Machinekit but I'm not sure if thats what I
> need.

No - the error refers to a Linux kernel system control file - the file
is part of a mechanism to alter the board pins, it's not a real data
file.

Did you run "sudo config-pin overlay cape-universala" first?  I get a
similar error, but it resolves itself when I run the above.  If
curious, read through the description at:
https://github.com/cdsteinkuehler/beaglebone-universal-io

What does "ls -l /sys/devices/platform/ocp/" report?

-Kevin

KevinOConnor, 2017-11-17:
Were you able to find any further information on this issue?

prcdslnc13, 2017-11-17:
I didn't see the last reply. Sorry. Ill check this later tonight and report back. I dont have it accessible to the outside networks right now. Thanks for following back!

prcdslnc13, 2017-11-20:
`~$ ls -l /sys/devices/platform/ocp/
total 0
drwxr-xr-x 3 root root    0 Nov 18 04:49 40300000.ocmcram
drwxr-xr-x 3 root root    0 Nov 18 04:49 40302000.ocmcram_nocache
drwxrwxr-x 4 root gpio    0 Nov 18 04:49 44e07000.gpio
drwxr-xr-x 4 root root    0 Nov 18 04:49 44e09000.serial
drwxr-xr-x 4 root root    0 Nov 18 04:49 44e0b000.i2c
drwxr-xr-x 4 root root    0 Nov 18 04:49 44e0d000.tscadc
drwxr-xr-x 5 root root    0 Nov 18 04:49 44e35000.wdt
drwxr-xr-x 4 root root    0 Nov 18 04:49 44e3e000.rtc
drwxr-xr-x 9 root root    0 Nov 18 04:49 47400000.usb
drwxr-xr-x 3 root root    0 Nov 18 04:49 48038000.mcasp
drwxr-xr-x 3 root root    0 Nov 18 04:49 48042000.timer
drwxr-xr-x 3 root root    0 Nov 18 04:49 48044000.timer
drwxr-xr-x 3 root root    0 Nov 18 04:49 48046000.timer
drwxr-xr-x 3 root root    0 Nov 18 04:49 48048000.timer
drwxr-xr-x 3 root root    0 Nov 18 04:49 4804a000.timer
drwxrwxr-x 4 root gpio    0 Nov 18 04:49 4804c000.gpio
drwxr-xr-x 4 root root    0 Nov 18 04:49 48060000.mmc
drwxr-xr-x 4 root root    0 Nov 18 04:49 480c8000.mailbox
drwxr-xr-x 3 root root    0 Nov 18 04:49 480ca000.spinlock
drwxr-xr-x 4 root root    0 Nov 18 04:49 4819c000.i2c
drwxrwxr-x 4 root gpio    0 Nov 18 04:49 481ac000.gpio
drwxrwxr-x 4 root gpio    0 Nov 18 04:49 481ae000.gpio
drwxr-xr-x 4 root root    0 Nov 18 04:49 481d8000.mmc
drwxr-xr-x 3 root root    0 Nov 18 04:49 48200000.interrupt-controller
drwxr-xr-x 5 root root    0 Nov 18 04:49 4830e000.lcdc
drwxr-xr-x 3 root root    0 Nov 18 04:49 48310000.rng
drwxr-xr-x 4 root root    0 Nov 18 04:49 49000000.edma
drwxr-xr-x 3 root root    0 Nov 18 04:49 49800000.tptc
drwxr-xr-x 3 root root    0 Nov 18 04:49 49900000.tptc
drwxr-xr-x 3 root root    0 Nov 18 04:49 49a00000.tptc
drwxr-xr-x 7 root root    0 Nov 18 04:49 4a100000.ethernet
drwxr-xr-x 6 root root    0 Nov 18 04:49 4a300000.pruss
drwxr-xr-x 3 root root    0 Nov 18 04:49 4c000000.emif
drwxr-xr-x 3 root root    0 Nov 18 04:49 53100000.sham
drwxr-xr-x 3 root root    0 Nov 18 04:49 53500000.aes
drwxr-xr-x 3 root root    0 Nov 18 04:49 56000000.sgx
-rw-r--r-- 1 root root 4096 Nov 20 15:56 driver_override
-r--r--r-- 1 root root 4096 Nov 20 15:56 modalias
drwxr-xr-x 6 root root    0 Nov 18 04:49 ocp:l4_wkup@44c00000
lrwxrwxrwx 1 root root    0 Nov 20 15:56 of_node -> ../../../firmware/devicetree/base/ocp
drwxr-xr-x 2 root root    0 Nov 20 15:56 power
lrwxrwxrwx 1 root root    0 Nov 18 04:49 subsystem -> ../../../bus/platform
-rw-r--r-- 1 root root 4096 Nov 18 04:49 uevent`

outlookhazy, 2017-11-20:
..sorry, couldn't help myself
*fixed

~$ ls -l /sys/devices/platform/ocp/ 
total 0 
drwxr-xr-x 3 root root 0 Nov 18 04:49 40300000.ocmcram 
drwxr-xr-x 3 root root 0 Nov 18 04:49 40302000.ocmcram_nocache 
drwxrwxr-x 4 root gpio 0 Nov 18 04:49 44e07000.gpio 
drwxr-xr-x 4 root root 0 Nov 18 04:49 44e09000.serial 
drwxr-xr-x 4 root root 0 Nov 18 04:49 44e0b000.i2c 
drwxr-xr-x 4 root root 0 Nov 18 04:49 44e0d000.tscadc 
drwxr-xr-x 5 root root 0 Nov 18 04:49 44e35000.wdt 
drwxr-xr-x 4 root root 0 Nov 18 04:49 44e3e000.rtc 
drwxr-xr-x 9 root root 0 Nov 18 04:49 47400000.usb 
drwxr-xr-x 3 root root 0 Nov 18 04:49 48038000.mcasp 
drwxr-xr-x 3 root root 0 Nov 18 04:49 48042000.timer 
drwxr-xr-x 3 root root 0 Nov 18 04:49 48044000.timer 
drwxr-xr-x 3 root root 0 Nov 18 04:49 48046000.timer 
drwxr-xr-x 3 root root 0 Nov 18 04:49 48048000.timer 
drwxr-xr-x 3 root root 0 Nov 18 04:49 4804a000.timer 
drwxrwxr-x 4 root gpio 0 Nov 18 04:49 4804c000.gpio 
drwxr-xr-x 4 root root 0 Nov 18 04:49 48060000.mmc 
drwxr-xr-x 4 root root 0 Nov 18 04:49 480c8000.mailbox 
drwxr-xr-x 3 root root 0 Nov 18 04:49 480ca000.spinlock 
drwxr-xr-x 4 root root 0 Nov 18 04:49 4819c000.i2c 
drwxrwxr-x 4 root gpio 0 Nov 18 04:49 481ac000.gpio 
drwxrwxr-x 4 root gpio 0 Nov 18 04:49 481ae000.gpio 
drwxr-xr-x 4 root root 0 Nov 18 04:49 481d8000.mmc 
drwxr-xr-x 3 root root 0 Nov 18 04:49 48200000.interrupt-controller 
drwxr-xr-x 5 root root 0 Nov 18 04:49 4830e000.lcdc 
drwxr-xr-x 3 root root 0 Nov 18 04:49 48310000.rng 
drwxr-xr-x 4 root root 0 Nov 18 04:49 49000000.edma 
drwxr-xr-x 3 root root 0 Nov 18 04:49 49800000.tptc 
drwxr-xr-x 3 root root 0 Nov 18 04:49 49900000.tptc 
drwxr-xr-x 3 root root 0 Nov 18 04:49 49a00000.tptc 
drwxr-xr-x 7 root root 0 Nov 18 04:49 4a100000.ethernet 
drwxr-xr-x 6 root root 0 Nov 18 04:49 4a300000.pruss 
drwxr-xr-x 3 root root 0 Nov 18 04:49 4c000000.emif 
drwxr-xr-x 3 root root 0 Nov 18 04:49 53100000.sham 
drwxr-xr-x 3 root root 0 Nov 18 04:49 53500000.aes 
drwxr-xr-x 3 root root 0 Nov 18 04:49 56000000.sgx 
-rw-r--r-- 1 root root 4096 Nov 20 15:56 driver_override 
-r--r--r-- 1 root root 4096 Nov 20 15:56 modalias 
drwxr-xr-x 6 root root 0 Nov 18 04:49 ocp:l4_wkup@44c00000 
lrwxrwxrwx 1 root root 0 Nov 20 15:56 of_node -> ../../../firmware/devicetree/base/ocp 
drwxr-xr-x 2 root root 0 Nov 20 15:56 power 
lrwxrwxrwx 1 root root 0 Nov 18 04:49 subsystem -> ../../../bus/platform 
-rw-r--r-- 1 root root 4096 Nov 18 04:49 uevent

prcdslnc13, 2017-11-20:
So dont use the code brackets. Got it.

KevinOConnor, 2017-11-20:
Was that "ls" output after running "sudo config-pin overlay cape-universala"?  What output do you receive when running "sudo config-pin overlay cape-universala"?

prcdslnc13, 2017-11-20:
Yes that is the output after "sudo config-pin overlay cape-universala"

prcdslnc13, 2017-11-20:
$ sudo config-pin overlay cape-universala
[sudo] password for debian:
debian@beaglebone:~$ ls -l /sys/devices/platform/ocp/
total 0
drwxr-xr-x 3 root root    0 Nov 20 15:56 40300000.ocmcram
drwxr-xr-x 3 root root    0 Nov 20 15:56 40302000.ocmcram_nocache
drwxrwxr-x 4 root gpio    0 Nov 18 04:49 44e07000.gpio
drwxr-xr-x 4 root root    0 Nov 20 15:56 44e09000.serial
drwxr-xr-x 4 root root    0 Nov 20 15:56 44e0b000.i2c
drwxr-xr-x 4 root root    0 Nov 20 15:56 44e0d000.tscadc
drwxr-xr-x 5 root root    0 Nov 20 15:56 44e35000.wdt
drwxr-xr-x 4 root root    0 Nov 20 15:56 44e3e000.rtc
drwxr-xr-x 9 root root    0 Nov 20 15:56 47400000.usb
drwxr-xr-x 3 root root    0 Nov 20 15:56 48038000.mcasp
drwxr-xr-x 3 root root    0 Nov 20 15:56 48042000.timer
drwxr-xr-x 3 root root    0 Nov 20 15:56 48044000.timer
drwxr-xr-x 3 root root    0 Nov 20 15:56 48046000.timer
drwxr-xr-x 3 root root    0 Nov 20 15:56 48048000.timer
drwxr-xr-x 3 root root    0 Nov 20 15:56 4804a000.timer
drwxrwxr-x 4 root gpio    0 Nov 18 04:49 4804c000.gpio
drwxr-xr-x 4 root root    0 Nov 20 15:56 48060000.mmc
drwxr-xr-x 4 root root    0 Nov 20 15:56 480c8000.mailbox
drwxr-xr-x 3 root root    0 Nov 20 15:56 480ca000.spinlock
drwxr-xr-x 4 root root    0 Nov 20 15:56 4819c000.i2c
drwxrwxr-x 4 root gpio    0 Nov 18 04:49 481ac000.gpio
drwxrwxr-x 4 root gpio    0 Nov 18 04:49 481ae000.gpio
drwxr-xr-x 4 root root    0 Nov 20 15:56 481d8000.mmc
drwxr-xr-x 3 root root    0 Nov 20 15:56 48200000.interrupt-controller
drwxr-xr-x 5 root root    0 Nov 20 15:56 4830e000.lcdc
drwxr-xr-x 3 root root    0 Nov 20 15:56 48310000.rng
drwxr-xr-x 4 root root    0 Nov 20 15:56 49000000.edma
drwxr-xr-x 3 root root    0 Nov 20 15:56 49800000.tptc
drwxr-xr-x 3 root root    0 Nov 20 15:56 49900000.tptc
drwxr-xr-x 3 root root    0 Nov 20 15:56 49a00000.tptc
drwxr-xr-x 7 root root    0 Nov 20 15:56 4a100000.ethernet
drwxr-xr-x 6 root root    0 Nov 20 15:56 4a300000.pruss
drwxr-xr-x 3 root root    0 Nov 20 15:56 4c000000.emif
drwxr-xr-x 3 root root    0 Nov 20 15:56 53100000.sham
drwxr-xr-x 3 root root    0 Nov 20 15:56 53500000.aes
drwxr-xr-x 3 root root    0 Nov 20 15:56 56000000.sgx
-rw-r--r-- 1 root root 4096 Nov 20 15:56 driver_override
-r--r--r-- 1 root root 4096 Nov 20 15:56 modalias
drwxr-xr-x 6 root root    0 Nov 20 15:56 ocp:l4_wkup@44c00000
lrwxrwxrwx 1 root root    0 Nov 20 15:56 of_node ->
../../../firmware/devicetree/base/ocp
drwxr-xr-x 2 root root    0 Nov 20 15:56 power
lrwxrwxrwx 1 root root    0 Nov 20 15:56 subsystem -> ../../../bus/platform
-rw-r--r-- 1 root root 4096 Nov 20 15:56 uevent

KevinOConnor, 2017-11-21:
On Mon, Nov 20, 2017 at 05:48:05PM +0000, prcdslnc13 wrote:
> $ sudo config-pin overlay cape-universala
> [sudo] password for debian:
> debian@beaglebone:~$ ls -l /sys/devices/platform/ocp/
[...]

I just tried booting bone-debian-9.2-iot-armhf-2017-10-10-4gb and I
don't have the problems you are experiencing.  I'm not sure what's
different about your setup.

Can you reboot, ssh into the beaglebone, and run "dmesg | gzip >
dmesg.gz" and then scp dmesg.gz to your desktop and attach it to this
issue?  I'll take a look through and see if I spot something
different.

Also, what does "sudo cat /sys/devices/platform/bone_capemgr/slots"
report for you?

-Kevin

KevinOConnor, 2017-11-21:
One other item - can you attach your /boot/uEnv.txt file as well?

prcdslnc13, 2017-11-21:
"sudo cat /sys/devices/platform/bone_capemgr/slots" Output:
0: PF----  -1
1: PF----  -1
2: PF----  -1
3: PF----  -1
4: P-O-L-   0 Override Board Name,00A0,Override Manuf,BB-ADC

[dmesg.gz](https://github.com/KevinOConnor/klipper/files/1493256/dmesg.gz)
[uEnv.txt](https://github.com/KevinOConnor/klipper/files/1493257/uEnv.txt)

KevinOConnor, 2017-11-21:
On Tue, Nov 21, 2017 at 10:43:01PM +0000, prcdslnc13 wrote:
> "sudo cat /sys/devices/platform/bone_capemgr/slots" Output:
> 0: PF----  -1
> 1: PF----  -1
> 2: PF----  -1
> 3: PF----  -1
> 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,BB-ADC
> 

Not sure what's going on - I get different results.  What does the
following report: sudo /opt/scripts/tools/version.sh

As near as I can tell, the "cape manager" is doing something odd.  As
I understand it, because you have "enable_uboot_cape_universal=1" in
your uEnv.txt file then config-pin should just work...  You could read
through this and see if you can track it down:
https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays

-Kevin

KevinOConnor, 2017-11-22:
On Wed, Nov 22, 2017 at 01:00:13AM +0000, prcdslnc13 wrote:
> debian@beaglebone:~$  sudo /opt/scripts/tools/version.sh
> [sudo] password for debian:
> git:/opt/scripts/:[d36fe9a7be9ebfc872b10a470e904ab4c61c4516]
> eeprom:[A335BNLT00C03214BBBK8443]
> dogtag:[BeagleBoard.org Debian Image 2017-10-10]
> bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-00002-g0f3f1c7907]
> bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2015.01-00001-gb2412df]

If you're booting from eMMC, then I think updating the U-Boot version
may help - but that's a guess.  The previous link I sent has the
details for upgrading U-Boot.

> kernel:[4.4.91-ti-r133]
> nodejs:[v6.12.0]
> uboot_overlay_options:[enable_uboot_overlays=1]
> uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
> uboot_overlay_options:[enable_uboot_cape_universal=1]
> pkg:[bb-cape-overlays]:[4.4.20171107.0-0rcnee1~stretch+20171107]
> pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
> pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
> 
> Heres the output from above. 
> 
> Are you getting the same with a CRAMPS cape attached or a replicape? I may just wipe and restart from scratch. 

I have neither.  My testing is with a beaglebone green and flashing
leds.  I've had a couple of users report success with the replicape.
You are the first to report anything on the CRAMPS.

-Kevin

KevinOConnor, 2018-01-04:
Any further updates on this?

3DSTB, 2018-03-12:
I have the same problem: Real temperature is much higher then read values.

debian@beaglebone:~$ config-pin -q P9.33
Pin is not modifyable: P9_33 AIN4

debian@beaglebone:~$ config-pin -q P9.36
Pin is not modifyable: P9_36 AIN5

Switches: this works for me (2 Pin switch wirth + ans S)
config-pin P8.8 lo

3DSTB, 2018-03-12:
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
[sudo] password for debian:
git:/opt/scripts/:[e307a944e0be0610ff5296e0abe4ad31a6e70daa]
eeprom:[A335BNLT0A6A0414BBBK1349]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-03-06]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-ge9ff418fb8]:[location: dd MBR]
kernel:[4.4.113-ti-r147]
nodejs:[v4.8.7]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180305.0-0rcnee0~jessie+20180305]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~jessie+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~jessie+20170830]
groups:[debian : debian adm tty kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev i2c bluetooth cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait uboot_detected_capes=cape-universal, coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.185262] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.186334] gpio-of-helper ocp:cape-universal: Allocated GPIO id=0
[    1.186524] gpio-of-helper ocp:cape-universal: Allocated GPIO id=1
[    1.186670] gpio-of-helper ocp:cape-universal: Allocated GPIO id=2
[    1.186819] gpio-of-helper ocp:cape-universal: Allocated GPIO id=3
[    1.187236] gpio-of-helper ocp:cape-universal: Allocated GPIO id=4
[    1.187399] gpio-of-helper ocp:cape-universal: Allocated GPIO id=5
[    1.187545] gpio-of-helper ocp:cape-universal: Allocated GPIO id=6
[    1.187692] gpio-of-helper ocp:cape-universal: Allocated GPIO id=7
[    1.187840] gpio-of-helper ocp:cape-universal: Allocated GPIO id=8
[    1.187984] gpio-of-helper ocp:cape-universal: Allocated GPIO id=9
[    1.188126] gpio-of-helper ocp:cape-universal: Allocated GPIO id=10
[    1.188278] gpio-of-helper ocp:cape-universal: Allocated GPIO id=11
[    1.188477] gpio-of-helper ocp:cape-universal: Allocated GPIO id=12
[    1.188634] gpio-of-helper ocp:cape-universal: Allocated GPIO id=13
[    1.188779] gpio-of-helper ocp:cape-universal: Allocated GPIO id=14
[    1.188922] gpio-of-helper ocp:cape-universal: Allocated GPIO id=15
[    1.189089] gpio-of-helper ocp:cape-universal: Allocated GPIO id=16
[    1.189236] gpio-of-helper ocp:cape-universal: Allocated GPIO id=17
[    1.189368] gpio-of-helper ocp:cape-universal: Allocated GPIO id=18
[    1.189507] gpio-of-helper ocp:cape-universal: Allocated GPIO id=19
[    1.189643] gpio-of-helper ocp:cape-universal: Allocated GPIO id=20
[    1.189788] gpio-of-helper ocp:cape-universal: Allocated GPIO id=21
[    1.189920] gpio-of-helper ocp:cape-universal: Allocated GPIO id=22
[    1.190052] gpio-of-helper ocp:cape-universal: Allocated GPIO id=23
[    1.190190] gpio-of-helper ocp:cape-universal: Allocated GPIO id=24
[    1.190368] gpio-of-helper ocp:cape-universal: Allocated GPIO id=25
[    1.190503] gpio-of-helper ocp:cape-universal: Allocated GPIO id=26
[    1.190645] gpio-of-helper ocp:cape-universal: Allocated GPIO id=27
[    1.190788] gpio-of-helper ocp:cape-universal: Allocated GPIO id=28
[    1.191293] gpio-of-helper ocp:cape-universal: Allocated GPIO id=29
[    1.191453] gpio-of-helper ocp:cape-universal: Allocated GPIO id=30
[    1.191592] gpio-of-helper ocp:cape-universal: Allocated GPIO id=31
[    1.191734] gpio-of-helper ocp:cape-universal: Allocated GPIO id=32
[    1.191873] gpio-of-helper ocp:cape-universal: Allocated GPIO id=33
[    1.192007] gpio-of-helper ocp:cape-universal: Allocated GPIO id=34
[    1.192149] gpio-of-helper ocp:cape-universal: Allocated GPIO id=35
[    1.192290] gpio-of-helper ocp:cape-universal: Allocated GPIO id=36
[    1.192433] gpio-of-helper ocp:cape-universal: Allocated GPIO id=37
[    1.192570] gpio-of-helper ocp:cape-universal: Allocated GPIO id=38
[    1.192582] gpio-of-helper ocp:cape-universal: ready
END

3DSTB, 2018-03-12:
Same here: to low temperature values with "linux process"

KevinOConnor, 2018-03-12:
Closing this again.  As discussed in issue #232 - the root cause of this issue is that the cramps board effectively uses a 2K pullup resistor.

closed 2018-03-12 by KevinOConnor