Klipper3d/klipper #38: Printer crashing

Spartus, 2017-10-29:
CoreXY + RPI3 + Arduino Mega 16Mhz

Printing at 80mm/s extrude and 120mm/s travel. All steppers are at 1/64 microstep. By my calculation these are both below 100 kHz total stepping.

Log attached. There seems to be some quantity of retransmit happening. I don'k know if the quantity is significant or if non-zero is normal?

[klippy.zip](https://github.com/KevinOConnor/klipper/files/1425501/klippy.zip)

MCU 'mcu' shutdown: Timer too close
clocksync state: mcu_freq=16000000 last_clock=2156399423790 clock_est=(137464.852 2154315356768 16007119.785) min_half_rtt=0.000338 min_rtt_time=136511.884 time_avg=137464.852(17432.689) clock_avg=2154315356768.808(279047141457.125) pred_variance=38847647.776

Dumping serial stats: bytes_write=9379689 bytes_read=15863752 bytes_retransmit=118402 bytes_invalid=0 send_seq=281926 receive_seq=281920 retransmit_seq=281924 srtt=0.007 rttvar=0.005 rto=0.874 ready_bytes=1397 stalled_bytes=3929
Dumping send queue 100 messages

travel speed / tooth pitch / teeth per rotation / steps per rotation / microsteps
(80mm/s)/(2mm*16/200/64) = 32 kHz * 2 for corexy = 64 kHz

I can travel at 120mm/s safely which is 96 kHz, just under the 100 kHz you quote.

extuder:

flow rate = 0.4mm*0.2mm*80mm/s = 6.4 mm^3/s
Extruder frequency = flow rate / filament area / step size
6.4 mm^3/s / (PI*(1.75mm)^2/4)) / 0.0015985mm = 1.66 kHz

so roughly I estimate 66 kHz stepping required for 80mm/s extrustion, yet I get crashing. Is it a calculation issue, communication issue, or a misunderstanding on my part? My goal would be to be able to print at 100 mm/s given this configuration.

Spartus, 2017-10-29:
I forgot to mention I am on the 0.5.0 release that just came out. 

also this graph feature isn't working:

graphstats.py /tmp/klippy.log2 loadgraph.png
Traceback (most recent call last):
  File "/home/pi/klipper/klipper/scripts/graphstats.py", line 120, in <module>
    main()
  File "/home/pi/klipper/klipper/scripts/graphstats.py", line 114, in main
    data = parse_log(logname)
  File "/home/pi/klipper/klipper/scripts/graphstats.py", line 24, in parse_log
    keyparts = dict(p.split('=', 1) for p in parts[2:])
ValueError: dictionary update sequence element #4 has length 1; 2 is required

Spartus, 2017-10-29:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/1425605/klippy.zip)

Another crash of the same print running at slower speeds. I changed usb cable but still have retransmits. I tried lowering to 115.2kbps but then it doesn't seem to connect so I flashed it back.

1. make menuconfig
2. 115200 bps
3. make
4. service stop
5. make flash FLASH_DEVICE=/dev/ttyACM0
6. service start
7. change to 115200 in octoprint
8. connect
9. fail to connect...

Also, how do I move the print head without homing? I tried G92 to set position but it still says needs to home.  When a print fails I get buried in a glob of plastic because of sudden stop with no retraction. I want to lift the nozzle before any other moves.

KevinOConnor, 2017-10-30:
On Sun, Oct 29, 2017 at 02:45:56PM -0700, Spartus wrote:
> I forgot to mention I am on the 0.5.0 release that just came out. 
> 
> also this graph feature isn't working:
[...]

Thanks.  I fixed the graphstats.py script.

On Sun, Oct 29, 2017 at 02:10:30PM -0700, Spartus wrote:
> CoreXY + RPI3 + Arduino Mega 16Mhz
>
> Printing at 80mm/s extrude and 120mm/s travel. All steppers are at 1/64 microstep. By my calculation these are both below 100 kHz total stepping.
>
> Log attached. [...]
[...]
> 
> travel speed / tooth pitch / teeth per rotation / steps per rotation / microsteps
> (80mm/s)/(2mm*16/200/64) = 32 kHz * 2 for corexy = 64 kHz
> 
> I can travel at 120mm/s safely which is 96 kHz, just under the 100 kHz you quote.

Indeed, but the benchmark numbers are with the mcu doing nothing but
stepping.  The mcu needs additional resources to parse and queue
incoming commands, perform the thermistor readings, etc.  I'd half
your current max travel rate - verify everything is stable and then
try increasing the rate.

The reason you are seeing the retransmits and eventual "Timer too
close" error is that the mcu is dutifully stepping at the rate
requested, but it ends up falling behind on the parsing of incoming
messages.  This is leading to the retransmits and eventually the comms
slow to the point that an error is raised.

If you run the (now fixed) graphstats.py script, you'll see how high
the mcu load is.  (Don't take the absolute usage numbers too seriously
- it's just a general indicator - but it is extremely high on both the
logs you've sent.)

On Sun, Oct 29, 2017 at 04:47:44PM -0700, Spartus wrote:
> [klippy.zip](https://github.com/KevinOConnor/klipper/files/1425605/klippy.zip)
>
> Another crash of the same print running at slower speeds. I changed usb cable but still have retransmits. I tried [...]

To work at a slower baud, don't change the value in octoprint (it's
not relevant) - instead, be sure to change the baud value in the mcu
section of the printer.cfg file.

However, I doubt changing the baud will improve your results.

> Also, how do I move the print head without homing? I tried G92 to set position but it still says needs to home.  When a print fails I get buried in a glob of plastic because of sudden stop with no retraction. I want to lift the nozzle before any other moves.

There is no current way to move the motor without homing.  It is
listed as a possible feature in the docs/Todo.md file.

-Kevin

Spartus, 2017-10-30:
You have absolutely nailed every question I had there. Thank you so much for the awesome support.

I thought I was having signal quality / interference issues, now I know its just overworked. Unfortunate, but I can deal with that. 

I revisited my settings and I now realize that 8mm pitch lead screws @ 64microsteps @ 40mm/s is 64 kHz, much higher than I realized. The setting I had recently changed was z-hop, so I'll have to revisit these settings (less microsteps).

Some followup questions. To understand the speed limits a little better, can you provide some insight on the effort of extruding complex curves/infill  vs high speed travel. As you saw, by simple calculations I needed 66 kHz to extrude at 80mm/s and 96 kHz to travel at 120. When I retried that print I cut my speeds to 70 & 100 mm/s, but still failed. Since the travel is needing higher stepping is that what I should focus on lowering (z-hop aside), or is there more behind the scenes effort on extruding curves. Any tips on an appropriate ratio (I realize this may depend on some other factors, but are they roughly equal effort). I thinking it is probably 1:1 or very close to it. I would be very satisfied if simply dropping travel to 80 max fixed things if I could keep 80 extrude.

Any tips on interpreting MCU load graph? You say the absolute value is not representative of anything, but can you put into perspective what is "high"? I ask because the only way I can crash it is waiting until I'm a good chunk of the way into a big print. Perhaps you have some test input that is a heavy load test? Should I just do some dry print runs (with heater)?  Is heater necessary for stress test (PID control) or is most of the effort in the ADC?

Would a concept like dual-step mode or quad-step mode be a good feature to consider to keep microsteps high, motor quiet, but cpu loads lower? I realize this firmware is basically already doing the harder math on the RPI and only timing the steps on the MCU, so maybe there is little to gain.

Can you recommend a guide for Arduino overclocking? Does the firmware support higher than 20 MHz? I see 25 MHz is usually quite achievable.

Thanks for the awesome project and support!

P.S.  I would like to buy you lunch. Donation instructions? Do you take crypto coins?

Spartus, 2017-10-30:
While watching my printer do a dry print with new settings (its going well), it came to me that I should run my tests with simulavr

does it run simulations faster than real time? Will it to some degree replicate the serial re-transmission issues I was seeing as a symptom?

KevinOConnor, 2017-10-30:
On Sun, Oct 29, 2017 at 07:54:09PM -0700, Spartus wrote:
> Some followup questions. To understand the speed limits a little
> better, can you provide some insight on the effort of extruding
> complex curves/infill vs high speed travel. As you saw, by simple [...]

What stepper motor drivers are you using?  Can you decrease the
micro-stepping rate to 32 or 16?  (Due to the internal PWM in most
stepper drivers, higher microstepping often doesn't result in improved
accuracy.)

If you're calculating 66K steps/s during extrude moves, then I think
that should be okay.  There is some additional work in the mcu when a
stepper changes speed, but most of that work is done on the RPi.

On further thought, the errors you are seeing are compounded by the
retransmits.  As a test, you could try reducing baud to 115200 - the
host pacing heuristics should be better at the lower speed, which
should reduce retransmits.

> Any tips on interpreting MCU load graph? You say the absolute value
> is not representative of anything, but can you put into perspective
> what is "high"?

An "MCU load" of 100% roughly means that some background tasks (eg,
parsing incoming messages) were delayed up to 2.5ms.  (Or, more
accurately, the 99 percentile of background task duration was 2.5ms.)
That's roughly the time it takes to transmit a "message block" at
250000 baud.  An "MCU load" of 200% means some background tasks were
likely delayed up to 5ms, which makes dropped messages at 250000 baud
likely.  I'd call "high" anything over 75% load.

That said - if you don't encounter errors, it's not a big deal to have
a higher load.

>I ask because the only way I can crash it is waiting
> until I'm a good chunk of the way into a big print. Perhaps you have
> some test input that is a heavy load test? Should I just do some dry
> print runs (with heater)?

The easiest way is probably to do a dry run - you should be able to
comment out the extruder enable_pin line and set the min_extrude_temp
to zero in printer.cfg.  (You can also edit homing.py:__init__() and
force "self.verify_retract = False", comment out all the enable_pin
lines, and invert your endstop polarities - then none of the steppers
have to actually move to test stepping rates.)

>Is heater necessary for stress test (PID
> control) or is most of the effort in the ADC?

The PID logic is almost exclusively done on the RPi.

> Would a concept like dual-step mode or quad-step mode be a good
> feature to consider to keep microsteps high, motor quiet, but cpu
> loads lower? I realize this firmware is basically already doing the
> harder math on the RPI and only timing the steps on the MCU, so
> maybe there is little to gain.

I don't think that would help.  I have some ideas to slightly improve
stepper performance, but nothing that would be ready in the short
term.

> Can you recommend a guide for Arduino overclocking? Does the
> firmware support higher than 20 MHz? I see 25 MHz is usually quite
> achievable.

Alas, I have no knowledge of that.  The only restriction Klipper has
is in the Kconfig files (which allows users to easily choose common
values) - all of the C and python code will take a different speed
without issue.

> Thanks for the awesome project and support!
> 
> 
> P.S.  I would like to buy you lunch. Donation instructions? Do you take crypto coins?
> 

I don't have anything setup for that right now, but I appreciate the
offer!

-Kevin

KevinOConnor, 2017-10-30:
On Sun, Oct 29, 2017 at 08:47:38PM -0700, Spartus wrote:
> While watching my printer do a dry print with new settings (its going well), it came to me that I should run my tests with simulavr
> 
> does it run simulations faster than real time? Will it to some degree replicate the serial re-transmission issues I was seeing as a symptom?

I find simulavr runs dramatically slower than a real AVR chip.  (The
actual achievable speed depends on the desktop processor available.)

It will reproduce the serial retransmits.  Unfortunately, due to
difficulties in doing clock synchronization with simulavr, it's not
great at long running tests (when run over long durations, the high
probability of a spurious error due to the unstable simulavr clock
makes it hard to test for real errors).

-Kevin

Spartus, 2017-10-30:
Well no matter, if its slower than real time I may as well do dry runs, are you tips are helpful so in the future I wont have to unload filament.

I am doing a test where all 4 axis are 32 kHz maximum step rate in klippy, and print settings are designed to max them out. Of course corexy means double for travel. The print is progressing but there are serial issues and the graph looks bad:

[image]

I notice that the worst of it is doing the outer perimeter, which is gently curved and thus able to run at full speed through the curve. The solid infill is comparatively easy because it is at 45 degrees and thus basically only 1 stepper on corexy. I'm not sure why the outside curves are even harder than the triangular infill. At this point it seems I will be switching to 32 microstep, at least until I get time for some overclock mods.

I think I have thoroughly racked your brain, thanks so much.

KevinOConnor, 2017-11-07:
Were you able to resolve this with a 32 microstep setting?

Spartus, 2017-11-07:
Yup, naturally it can keep up at 80mm/s and faster (120+) at 32 microstep. Got some experience with different settings and I think I can get the print speeds I want at 64 microstep with a 20Mhz AVR, but thats a battle for latter. Thanks again for the awesome support.

KevinOConnor, 2017-11-08:
Okay, thanks.  I'm going to close this issue then - reopen it if you encounter further issues.

closed 2017-11-08 by KevinOConnor