Klipper3d/klipper #39: Does klipper support dual Z ?

rafaljot, 2017-11-03:
Hi, 
I'm just understood this approach to firmware is the future. I have three printers, Prusa , K8400, HyperCube Evo. I want to try klipper in one. So, I have some questions before decide which one.

1. Does klipper support dual Z? I mean dual Z motor with two min end stops, each motor move separately, like Repetier does.
2. Does it support min, max end stops? 
4. Does it support tmc2130 and SPI ?

sorry for dummy questions. I think it is last time ;)

KevinOConnor, 2017-11-03:
On Fri, Nov 03, 2017 at 03:45:01PM +0000, rafaljot wrote:
> I'm just understood this approach to firmware is the future. I have three printers, Prusa , K8400, HyperCube Evo. I want to try klipper in one. So, I have some questions before decide which one.
>
> 1. [...]

If by "dual Z" you mean two separate stepper motor drivers with two
separate sets of step and direction pins, then it is not currently
supported.  Support would require some additional host python code.

> 2. Does it support min, max end stops? 

The current code can either home to a minimum position or home to a
maximum position on each axis.  There is no current support for
runtime endstop detection - support is on the todo list.

> 4. Does it support tmc2130 and SPI ?

There is no explicit support for the TMC2130.  If the SPI interface is
only used to configure the driver at startup (and regular step and
direction pins are used during runtime), then it may be possible to
configure it with send_spi_message commands.  Though, adding host
python code to do this would probably be easier (see the AD5206 code
in klippy/chipmisc.py as an example).

-Kevin

Papsnoek, 2017-11-03:
Currently the Main "missing feature" that is stopping me trying out Klipper is the Dual Z capabilities. I have a HyperCube Evolution and without this function, it would be a nightmare to keep the bed level before each print.

Could you point me in the right direction where to start looking at the code and I will see if I can contribute something?

rafaljot, 2017-11-03:
thank you. I'll start with Prusa. 

> If by "dual Z" you mean two separate stepper motor drivers with two
separate sets of step and direction pins, then it is not currently
supported.  Support would require some additional host python code.

Exactly. Marlin 2.0 extended this functionality to X and Y as well . In the extreme case 6 independent drivers only for  XYZ. Fortunately your firmware has greater potential to manage it ;)

rafaljot, 2017-11-03:
@Papsnoek
this is my remix ;)
[https://www.thingiverse.com/thing:2501604](https://www.thingiverse.com/thing:2501604)
It is more then standard dual Z in HyperCube . Do you have it or similar?

KevinOConnor, 2017-11-03:
On Fri, Nov 03, 2017 at 04:50:21PM +0000, Bradley Muller wrote:
> Currently the Main "missing feature" that is stopping me trying out Klipper is the Dual Z capabilities. I have a HyperCube Evolution and without this function, it would be a nightmare to keep the [...]

I'd start by looking at klippy/corexy.py - you'll see the move()
method converts each high-level move request from the "trapezoid
generator" into movement on each stepper.  For dual-Z, you could
duplicate the Z movement calls on both Z steppers.  You'll also need
to make sure that the home() method passes both Z steppers to the
homing code when homing the Z axis.

It should be possible to get basic support working this way.  However,
general dual stepper support is a bit more complicated as it would
likely require some code refactoring so that dual-z also works with
cartesian printers and so that dual-x / dual-y support is also
provided.

-Kevin

zen, 2017-11-06:
@Papsnoek people are running dual Z motors on Evo just using splitter from single driver (like most Prusa do). You could also do a trick with custom G-code: move head away from over bed (should be possible with most Evos), home Z, then move Z to -5mm. It will make motors to loose steps and synchronise. It will require some voltage tweaking not to damage motors.

Papsnoek, 2017-11-07:
@zen Thanks for the suggestions. I have been spoilt by using The Dual Z as bed leveling is so simple and easy, so loosing this feature would not be the best for me.

@KevinOConnor Thanks for the tips... I did a fork and started digging around and got over whelmed :) But with your suggestions I will take another look and combine this with how Marlin has implemented this :)

@rafaljot I did create my Own Z Optical Endstop, but never shared it cause it was difficult to adjust. I see yours makes it easy to adjust so I think Im going to give it a try soon :) Thanks

KevinOConnor, 2017-11-07:
On Tue, Nov 07, 2017 at 11:40:25AM +0000, Bradley Muller wrote:
> @KevinOConnor Thanks for the tips... I did a fork and started digging around and got over whelmed :) But with your suggestions I will take another look and combine this with how Marlin has implemented this :)

Ah, Ok.  I put together a prototype.  It's on the
work-multistepper-20171107 branch (git fetch ; git checkout
origin/work-multistepper-20171107).

I don't have a good way to test this, but if anyone wants to give it a
try then let me know how it goes.  In order to use the support, one
will need to add a "stepper_z1" section (in addition to the existing
"stepper_z" section) to the printer.cfg file as described in the
updated config/example-extras.cfg file.

-Kevin

Papsnoek, 2017-11-08:
Has anyone mentioned that this guy IS A LEGEND!!

I will definitely give this a test now as it means I can now easily switch between Klipper and Marlin without have to fiddle with any cables now.

@KevinOConnor I have been checking the code and I don't see any change to "Homing"? Would the Homing home both Steppers individually? Or reading the Code, is it instead of sending 1 Z Command to 1 Stepper, its now just Sending 1 Z Command to 2 Steppers? It looks this way and I cannot believe how Elegant your solution is :)

KevinOConnor, 2017-11-08:
On Wed, Nov 08, 2017 at 02:39:59PM +0000, Bradley Muller wrote:
> Has anyone mentioned that this guy IS A LEGEND!!
> 
> I will definitely give this a test now as it means I can now easily switch between Klipper and Marlin without have to fiddle with any cables now.
> 

Great!  Thanks.

> @KevinOConnor I have been checking the code and I don't see any change to "Homing"? Would the Homing home both Steppers individually? Or reading the Code, is it instead of sending 1 Z Command to 1 [...]

Movement (even during homing) is implemented by the kinematics class'
move() method.  The prototype code sends moves to both steppers (see
stepper.py:PrinterMultiStepper:step_multi_const() - so this will be
done for both regular moves and homing moves.  The change to get
multiple endstops for an axis working was mostly done in commit
970524fd where the homing.py code was changed to get a list of
endstops from each axis being homed.

-Kevin

lenne0815, 2017-11-09:
Thank you so much Kevin ! Ill be back home on monday, conversing my big printer is the first thing im gonna do !

Papsnoek, 2017-11-10:
Im not sure if I said this, but this guys is a LEGEND :)

I tested the Dual Z last night it worked Perfect! Printed my first Test Cube on Klipper just before bed and it came out perfect!

Just on the Config, in my [stepper_z1] it kept saying the option "position_endstop" was not valid, so I removed "position_endstop" and "position_max" from [stepper_z1] and it worked... Im just not 100% sure if these are needed and it was reported in error, if if they are not needed?
Also, the Homing, it worked nicely with dual endstops, and it home each separately, but the second home doesn't feel right, its almost like on the First home, the left side clicks, and the  continues to home the right side, then bed moves down again, and the second home stops when only 1 side clicks, not both.... I would have to do more testing on this as it might be they both "clicked" at exactly the same time, but I was too tired to pick this up :)

I picked up 1 or 2 other issues after my print experience, but I will create new issues around these... Thanks for the amazing product so far!

Papsnoek, 2017-11-12:
@KevinOConnor Not sure if you have Facebook, but I created a Post in the Hypercube Evolution Group about my results with your Software. Might want to check it out, and also, it might cause a lot of extra interest soon :)
https://www.facebook.com/groups/Hypercubeevolution/permalink/937403903079557/

lenne0815, 2017-11-13:
Hey Kevin, im about to converse my big printer, the process for dual Z is the same for any axis ? ( in my specific case i need to mirror z and x )

KevinOConnor, 2017-11-13:
On Fri, Nov 10, 2017 at 08:48:02AM +0000, Bradley Muller wrote:
> Just on the Config, in my [stepper_z1] it kept saying the option "position_endstop" was not valid, so I removed "position_endstop" and "position_max" from [stepper_z1] and it worked... Im just not 100% sure if these are needed and it was reported in error, if if they are not needed?

The current code does not support any position arguments.  The
position of secondary endstops must match the position of the primary
endstop.

On Sun, Nov 12, 2017 at 10:11:02AM +0000, Bradley Muller wrote:
> @KevinOConnor Not sure if you have Facebook, but I created a Post in the Hypercube Evolution Group about my results with your Software. Might want to check it out, and also, it might cause a lot of extra interest soon :)
> https://www.facebook.com/groups/Hypercubeevolution/permalink/937403903079557/

Thanks!

-Kevin

KevinOConnor, 2017-11-13:
On Mon, Nov 13, 2017 at 11:08:06AM +0000, lenne0815 wrote:
> Hey Kevin, im about to converse my big printer, the process for dual Z is the same for any axis ? ( in my specific case i need to mirror z and x )

It should work (add stepper_x1 and stepper_z1 config sections).

Note that for a dual-x setup the micro-controller needs to step twice
for each X move and thus the maximum XY speed obtainable might be
lower than a setup with just one X stepper.

-Kevin

lenne0815, 2017-11-13:
I got it halfway working already, got some homing issues with z right now ( dual x and single y work already ) but i think i can get around them myself, will report back how it plays out !

lenne0815, 2017-11-14:
Ok, sadly i wasnt able to get around the problem, even though dual works on x without problems, z seems to cause an error: as soon as i issue G28 Z - MCU 'mcu' shutdown: Timer too close happens. Heres my corresponding [Klippy.log](MCU 'mcu' shutdown: Timer too close)

KevinOConnor, 2017-11-14:
On Tue, Nov 14, 2017 at 01:37:18AM +0000, lenne0815 wrote:
> Ok, sadly i wasnt able to get around the problem, even though dual works on x without problems, z seems to cause an error: as soon as i issue G28 Z - MCU 'mcu' shutdown: Timer too close happens. Heres my corresponding [Klippy.log](MCU 'mcu' shutdown: Timer too close)

As a test, can you try adding "buffer_time_start = .750" to the
"[printer]" section of the config and report the results?

-Kevin

lenne0815, 2017-11-14:
That worked ! Thank you so much ! 

Maybe its interesting what Hardware Klipper is running on; its a rumba board with the "power from usb" jumper disabled maybe that has something to do with the error.

lenne0815, 2017-11-14:
Unfortunately, it doesnt work all the time now :/ sometimes i still get the error, ill try to tweak the buffer_time_start = .750 value until it works now.

lenne0815, 2017-11-14:
Aaaaaand its fixed ! it was indeed the "power over usb" jumber on the rumba, set it to open ( arduino gets usb power ) and it works now ! Ill try to go full circle and remove buffer_time_start from the config again.

KevinOConnor, 2017-11-14:
On Tue, Nov 14, 2017 at 10:10:38AM +0000, lenne0815 wrote:
> Unfortunately, it doesnt work all the time now :/ sometimes i still get the error, ill try to tweak the buffer_time_start = .750 value until it works now.

Can you send the log with the error when buffer_time_start=.750 (even
if you're able to fix the issue with other tweaks).

-Kevin

lenne0815, 2017-11-14:
Thank you for investigating further Kevin, heres the latest [log](https://drive.google.com/file/d/1Ol0PSCbXHs5o1yFk8gb-vssqjArU_stp/view?usp=sharing) I went away for a bit so can only start trying more approaches now, its really hard to pin down because it doesnt happen all the time. What is always the same is that it happens right after spinning up the z motors up for the first homing travel. When it succeeds homing its all fine afterwards and works as expected.

KevinOConnor, 2017-11-14:
On Tue, Nov 14, 2017 at 04:44:47PM +0000, lenne0815 wrote:
> Thank you for investigating further Kevin, heres the latest [log](https://drive.google.com/file/d/1Ol0PSCbXHs5o1yFk8gb-vssqjArU_stp/view?usp=sharing) I went away for a bit so can only start trying [...]

Okay - I think I know what is going on.  Because you have a very large
Z axis and because you have two steppers, the host software is
generating the timing for 8.4 million steps during the Z home
operation.  The processing time it is taking to do that is leading to
the error.  I need to think about how to fix this.

In the mean time, can you try removing the "buffer_time_start" line
from the config and change the HOMING_DELAY line at the top of
klippy/homing.py from 0.250 to 0.750.

-Kevin

lenne0815, 2017-11-14:
Ok, ill do that straight away !

lenne0815, 2017-11-14:
No, unfortunately the first G28 from a actual gcode File failed again [Log](https://drive.google.com/file/d/192-a5aG5MTT3fdMYDnVqm1uXFhtXYULP/view?usp=sharing)

lenne0815, 2017-11-14:
Ok, i reduced my z height in the config hugely ( to 200mm ) and that seems to do the trick, just as you predicted. Ill use that for dialing in the printer and hope your able to work around / fix it without to much work. ( Maybe theres a way to split homing movement into individual blocks of f.e. 100mm ? as i understand it klipper is preparing the "worst case" for homing and creating all the pulses needed as if it would have to go 700mm in one go ? )

lenne0815, 2017-11-15:
I thought about the problem and found a much better approach to fix the issue; microstepping on z without mesh bed leveling or running 3D code in general is pretty much useless anyway, so i simply reduced microstepping to 1/4 from 1/16 on z and now klipper can calculate the pulses in the given timeframe with ease.

KevinOConnor, 2017-11-18:
Given the positive feedback on the multistepper branch, I've merged that branch into the master branch (commit 8c2fa2e2).  I've also removed the work-multistepper-20171107 branch.

@lenne0815 - I haven't forgotten your issue.  However, since you have a workaround for now, I'm going to take some additional time thinking of how best to solve the issue.

TunaLatte, 2017-11-21:
Hello, I can also confirm the same issue on my new Pegasus 12" with RAMPS shield. I often get timer too close just after homing when printing.

KevinOConnor, 2017-11-21:
On Tue, Nov 21, 2017 at 08:26:43PM +0000, TunaLatte wrote:
> Hello, I can also confirm the same issue on my new Pegasus 12" with RAMPS shield. I often get timer too close just after homing when printing.

Okay - thanks.  Can you attach the log (I'd like to see your config so
I can try to reproduce with that as well).

-Kevin

TunaLatte, 2017-11-21:
Hello sorry forgot to attach! : https://gist.github.com/TunaLatte/881b7fa1118eab9de307bffe6913ff02

KevinOConnor, 2017-11-21:
On Tue, Nov 21, 2017 at 09:33:02PM +0000, TunaLatte wrote:
> Hello sorry forgot to attach! : https://gist.github.com/TunaLatte/881b7fa1118eab9de307bffe6913ff02

A "Timeout with MCU" error is not the same as a "Timer too close"
error.  Your config also doesn't have the same pattern (large number
of steps needed to home), so I don't think your issue is related to
this issue.

A "Timeout with MCU" error is generally due to a communications outage
between host and micro-controller (or perhaps a complete failure of
the micro-controller).  If you're unable to track down the problem,
then please open a separate issue and attach the complete log of the
session containing the error.  It must be the complete log as
described in docs/Contact.md - snippets from the log are of no value.

-Kevin

lenne0815, 2017-11-23:
Just wanted to give a quick heads up on the issue again, ive printed lots and lots now on that printer and the problem didnt resurface at all since reducing microstepping. Im off to converse my third printer now, planning a small graveyard for the 12860 displays in the garden.

KevinOConnor, 2017-11-30:
On Wed, Nov 15, 2017 at 07:29:53PM +0000, lenne0815 wrote:
> I thought about the problem and found a much better approach to fix the issue; microstepping on z without mesh bed leveling or running 3D code in general is pretty much useless anyway, so i simply [...]

I was able to track down the original "timer too close" issue.  There
were three separate problems:

1 - For really big axes, we need to give a little extra time for
generating the steps - this is now implemented as of commit 6930a7de.

2 - A recent commit broke the proactive flushing done on moves that
generated a ton of steps.  This should now be fixed (commit 6d663882).

3 - Your mcu/printer isn't capable of reliably homing the Z at 10mm/s.
On your printer 10mm/s on the Z would be 80K steps per second.
Klipper can nominally handle that step rate on a 16Mhz avr, but not
while homing.  The extra time needed to check the endstops during
homing reduces the available step rate.  I suspect you'll be okay with
Z homing at 5mm/s (if you go back to 16 microsteps).

On Thu, Nov 23, 2017 at 05:28:51PM +0000, lenne0815 wrote:
> Just wanted to give a quick heads up on the issue again, ive printed lots and lots now on that printer and the problem didnt resurface at all since reducing microstepping. Im off to converse my third printer now, planning a small graveyard for the 12860 displays in the garden.

Out of curiosity - did you notice any print/quality differences on the
larger printer?

-Kevin

lenne0815, 2017-11-30:
Hey Kevin,

thanks for the fixes !

I think ill just leave it at 1/4 stepping, this gives me a theoretical resolution of 0,001 which is already way to precise for my mechanical build ;) I solved noisy chopping by just adding two stepper dampeners.

In operation i couldnt spot any differences to my smaller printer, pressure advance works just as smooth and movement seems all around flawless, seems like when staying in the 8bit envelope klipper doesnt discriminate ;)

I havent fully tuned the big one though, running conservative 1.8k accels and 70mms median print speeds, again with 0.01 junction deviation as 0.02 jerked the steppers on specific gcodes ( observed that behaviour on the small printer, happened for finely subdivided "circles" )

Later on im planning to switch the big one over to 9mm gates belts and thereby enabling something in the 3k accel / 100mms range so maybe that gives us more reliable answers if machine size plays a role.

KevinOConnor, 2017-12-03:
Dual stepper support is now has been merged and tested - closing this issue.

closed 2017-12-03 by KevinOConnor