Klipper3d/klipper #42: Possible E1 pins backwards in generic-rambo.cfg

geodave810, 2017-11-08:
Before trying this firmware on my Makerfarm 10" Prusa i3v with Rambo Electronics, titan extruder & e3d Lite6, I was looking at the definitions of these pins.

I came across this page, http://reprap.org/wiki/Rambo_development that described these pins & the pins for the E1 extruder seem like they may be backwards since the other pairs of pins go in order of Microstep1 & then Microstep2.  This is line 106 of the file generic-rambo.cfg.  I am not using a 2nd extruder, so it is not a problem for me.

52  PG1 ( RD )			Digital pin 40  X Microstep1 
51  PG0 ( WR )  		Digital pin 41  X Microstep2
		
82  PK7 ( ADC15/PCINT23 )	Analog pin 15  Y Microstep1
70  PG2 ( ALE )			Digital pin 39  Y Microstep2

83  PK6 ( ADC14/PCINT22 )	Analog pin 14  Z Microstep1
84  PK5 ( ADC13/PCINT21 )	Analog pin 13  Z Microstep2

86  PK3 ( ADC11/PCINT19 )	Analog pin 11  E0 Microstep1
85  PK4 ( ADC12/PCINT20 )  	Analog pin 12  E0 Microstep2

87  PK2 ( ADC10/PCINT18 )  	Analog pin 10  E1 Microstep2
88  PK1 ( ADC9/PCINT17 )	Analog pin 9  E1 Microstep1

26  PB7 ( OC0A/OC1C/PCINT7 )  Digital pin 13 (PWM)  LED, PWM-Ext 3

KevinOConnor, 2017-11-08:
On Wed, Nov 08, 2017 at 02:46:30PM +0000, David wrote:
> Before trying this firmware on my Makerfarm 10" Prusa i3v with Rambo Electronics, titan extruder & e3d Lite6, I was looking at the definitions of these pins.
>
> I came across this page, [...]

Thanks - updated in commit b58a897b.  The order doesn't actually
impact the results, but it is nice to define them in-order for
documentation purposes.

-Kevin

geodave810, 2017-11-08:
Thanks for the reply.  I have it mostly set up now.  The hotend temperature works, the X & Y work using G28 X & G28 Y, but Z motors  just makes a noise when I issue a command G28 Z.  I can clear that noise by holding my finger on the z endstop.  I was originally using .54 for the digipots, but changed that to the original 1.1 for Z since there are 2 motors.  That did not work either, but think the noise  was louder.  I attached a zip file of klippy.log

In the Marlin firmware I had 100 for all the motors.

 

Thanks for your help,

 

-David

KevinOConnor, 2017-11-09:
On Wed, Nov 08, 2017 at 09:37:15PM +0000, David wrote:
> Thanks for the reply.  I have it mostly set up now.  The hotend temperature works, the X & Y work using G28 X & G28 Y, but Z motors  just makes a noise when I issue a command G28 Z.  I can clear [...]

The github issue does not show the klippy.log file.  Make sure to
attach the file via the web interface.

I'd also double check that your pin assignments are correct.  You have
two Z motors attached to a single stepper motor driver?  The setup
works with Marlin?  If so, can you also attach the Marlin config?

-Kevin

geodave810, 2017-11-09:
I have 2 Z motors & they work with Marlin.  I have attached a klippy.zip, the 2 configurations files for Marlin in marlin.zip & an image showing the wiring on my board using the web interface this time.  Thanks for your help.

[image]
[Marlin.zip](https://github.com/KevinOConnor/klipper/files/1457350/Marlin.zip)
[klippy.zip](https://github.com/KevinOConnor/klipper/files/1457351/klippy.zip)

Papsnoek, 2017-11-09:
My First Guess is the Z axis is not moving cause the Endstop is triggered, so the Printer thinks it should not be moving. And Looking at the Klippy Log File, I see an error:
Endstop z still triggered after retract

Try Reversing the Endstop pin as below:
[stepper_z]
step_pin = PC2
dir_pin = PL2
enable_pin = !PA5
step_distance = 0.00025
endstop_pin = ^!PB4
position_endstop = 0.5
position_max = 250

geodave810, 2017-11-09:
That did not do it, but the sound did not last as long.  I am wondering if the message you saw in the previous log file is related to me manually triggering that switch so I could make the motors stop the noise without me turning it off.  Here is the log file from the latest attempt.
[klippy_2.zip](https://github.com/KevinOConnor/klipper/files/1457762/klippy_2.zip)

KevinOConnor, 2017-11-09:
On Thu, Nov 09, 2017 at 11:06:04AM +0000, David wrote:
> I have 2 Z motors & they work with Marlin.  I have attached a klippy.zip, the 2 configurations files for Marlin in marlin.zip & an image showing the wiring on my board using the web interface this time.  Thanks for your help.

Make sure to run QUERY_ENDSTOPS (type the command into the octoprint
terminal window), verify that it reports the z endstop open when it is
not triggerred, and verify that it reports triggerred while you
manually trigger the endstop.  If it doesn't, update the endstop pin
settings until the above does happen.

If you change the printer.cfg file to use a max_z_velocity=2 and
max_z_accel=5 (as is in your marlin config) does it improve your
results?

-Kevin

geodave810, 2017-11-09:
Setting the max_z_velocity=2 and max_z_accel=5 fixed the problem.  Glad you spotted that in my Marlin config file. I also had to reverse the direction of pin for the Z motor with dir_pin: !PL2

After you pointed this out, I see in my Marlin config, the max_velocity = 250 & the max_accel = 1000.  Should I change those also?

geodave810, 2017-11-09:
On my 1st test print, I am getting blobbing along all sides as shown in the attached photo.  I checked for correct extrusion amount by setting a mark 60mm on the filament & extruding 50mm & I got 50mm.  I am using Slic3r Prusa Edition 1.37.2 and have First layer extrusion width set to 200% which has worked well before.  I thought that might be the problem, so changed it to 120% with same results.  I double checked the bed level & it seemed fine to me.  Also I noticed when I cancel the print in Octoprint, I have to rehome the Z in order to go up a few mm's above the part.  With Marlin, I did not have to do that.  I like to move the hot tip above the part before returning x,y to Home.  I saw in your thread http://forums.reprap.org/read.php?147,667655,page=2 you mentioned tuning pressure_advance for blobs on corners, but I am getting them all around.  The filament I am using is eSun 1.75mm PLA+ Silver extruding at 205c which is the temperature that has worked best for me previously.  I also attached my latest klippy.log file.
[image]

[klippy_3.zip](https://github.com/KevinOConnor/klipper/files/1458819/klippy_3.zip)

KevinOConnor, 2017-11-09:
On Thu, Nov 09, 2017 at 03:01:07PM +0000, David wrote:
> Setting the max_z_velocity=2 and max_z_accel=5 fixed the problem.  Glad you spotted that in my Marlin config file. I also had to reverse the direction of pin for the Z motor with dir_pin: !PL2 [...]

Your Marlin config uses an accel of 500mm/s^2.  (Marlin's
configuration is terribly confusing - it uses the minimum of
DEFAULT_MAX_ACCELERATION and DEFAULT_ACCELERATION.)  That value seems
a bit low to me, but I don't enough about your hardware to say whether
it is correct or not.

The maximum velocity limits the speed requested in the g-code - as
long as you don't ask the printer to go faster than 250mm/s it wont
really matter what that setting is.

Also, FYI, your extruder.step_distance should be 0.00252 according to
your Marlin config.

-Kevin

geodave810, 2017-11-09:
I had noticed extruder_step_distance being off a little & think you will see that change in the latest log file.  I did not know how many decimal places to take that number out, so I went with .00251889. If I had got my calculator out, I would have realized 0.00252 would be close enough.  I do not know the finer points of most of these firmware settings.

KevinOConnor, 2017-11-09:
On Thu, Nov 09, 2017 at 06:54:22PM +0000, David wrote:
> On my 1st test print, I am getting blobbing along all sides as shown in the attached photo.  I checked for correct extrusion amount by setting a mark 60mm on the filament & extruding 50mm & I got [...]

It's hard to say.  Does the blobbing occur on further levels as well
(that is, could it be a z height issue)?

Otherwise, it seems you have klipper at a higher accel (1000 vs 500 -
though that seems an unlikely explanation) and you're using a lower
amp setting on your steppers (100 in Marlin would be 0.8125 in Klipper
- 2.08 / 256 * 100 == 0.8125).

-Kevin

geodave810, 2017-11-09:
I was confused as to how to calculate the amps from the way Marlin does, so I changed it to the 0.8125 you calculated.  I left the max_acc set to 1000 as that seems like what it was in my Marlin config.  I see the generic-rambo.cfg is set to 3000.  I ran the print again with the same problem, but it looks like it is going lower than it should to start with.  Is there some z offset somewhere or maybe I should relevel the bed at a higher clearance?  I can not really get past the 1st level to see how the other layers are as the hotend rubs too much on the already printed surface.  I am using a .1mm feeler gauge to set the level. I attached the gcode file I am printing in case you see something in there I need to change.
[square_P137_2.zip](https://github.com/KevinOConnor/klipper/files/1459014/square_P137_2.zip)

KevinOConnor, 2017-11-09:
On Thu, Nov 09, 2017 at 08:02:50PM +0000, David wrote:
> I was confused as to how to calculate the amps from the way Marlin does, so I changed it to the 0.8125 you calculated.  I left the max_acc set to 1000 as that seems like what it was in my Marlin [...]

If the head is too close to the bed on the first layer, then decrease
stepper_z.position_endstop.  Your current config is informing klipper
that the head (when hot) is .5mm away from the bed when the endstop
triggers.

If you've set the endstop to trigger when the nozzle is .1mm away
(when hot) then set the field to .1mm.  (If you've done the
measurements when the hotend is cold then try setting the field to
zero, as it's not unusual for the hotend to expand by .1mm during
heating.)

-Kevin

geodave810, 2017-11-09:
That was it.  I sure did not look at that close enough.  I see that most of those example files have that 0.5 set.  Is that normal to not have 0.0 for that?  Test print came well.  I have a little ripple on the edge where the Y bed moves & noticed one eccentric spacer needed adjustment.  I am printing a tree frog now to compare it to one printed using Marlin.  Thanks for finding all my errors.

KevinOConnor, 2017-11-09:
On Thu, Nov 09, 2017 at 09:42:56PM +0000, David wrote:
> That was it.  I sure did not look at that close enough.  I see that most of those example files have that 0.5 set.  Is that normal to not have 0.0 for that?  Test print came well.  I have a little [...]

In general, I recommend adjusting the Z endstop so that it triggers
before the nozzle hits the bed (ie, at a positive position) as
otherwise one risks head collisions during homing.  Indeed, I
recommend homing to Z max if possible.

Cheers,
-Kevin

closed 2017-11-17 by KevinOConnor