Klipper3d/klipper #43: float division by zero when adc reads 1.0

dragonnn, 2017-11-11:
When no thermistor is connected:
`Traceback (most recent call last):
  File "/opt/klipper/klippy/serialhdl.py", line 51, in _bg_thread
    hdl(params)
  File "/opt/klipper/klippy/mcu.py", line 404, in _handle_analog_in_state
    self._callback(last_read_time, last_value)
  File "/opt/klipper/klippy/heater.py", line 157, in adc_callback
    temp = self.sensor.calc_temp(read_value)
  File "/opt/klipper/klippy/heater.py", line 39, in calc_temp
    r = self.pullup * adc / (1.0 - adc)
ZeroDivisionError: float division by zero`

I know no thermistor connected is an "hardware error" but this schould produce an error "float division by zero". Maybe returning a negative value?

KevinOConnor, 2017-11-13:
Thanks.  Should be fixed as of commit fda98888.

dragonnn, 2017-11-13:
Thank you, works great.

closed 2017-11-13 by dragonnn