Klipper3d/klipper #44: M42 - Set Pin State support?

dragonnn, 2017-11-11:
Marlin have support for M42 command http://marlinfw.org/docs/gcode/M042.html I used it to control some pins to enable ramps cooling and others. I only need digital control.

lenne0815, 2017-11-11:
This would be great, i planned getting around that somehow by trying to copy the fan code somehow and failing miserably on doing it.

dragonnn, 2017-11-11:
Hmmm what is static_digital_output :D?

dragonnn, 2017-11-11:
Ok, I added:
`[static_digital_output my_output_pins]
pins: ar63`
And now my pin I set to high when turning on. The question is can we control those pins via g-code?

KevinOConnor, 2017-11-11:
On Sat, Nov 11, 2017 at 09:19:06PM +0000, Mateusz wrote:
> Marlin have support for M42 command http://marlinfw.org/docs/gcode/M042.html I used it to control some pins to enable ramps cooling and others. I only need digital control. 

Can you explain how you use the M42 command?  (What you want it to do,
how and when you run it?)

The typical setup on Klipper is to use a "static_digital_out" (or
"static_pwm_out") config section if the pin is to be set at a static
rate the entire time the micro-controller is on.  Alternatively, for
fans, the "heater_fan" section is used to control fans that should be
enabled when a heater is on.  The settings for all of the above are
documented in the config/example-extras.cfg file.

-Kevin

dragonnn, 2017-11-11:
In slicer start g-code I pull with M42 my pin 63 to HIGH, in end-gcode I have a command which pulls it down. So after the print finish my printer is complete silent. I think allowing to control the static_digital_out would be a good idea (or create a new section called for examplechangeable_digital_out).

KevinOConnor, 2017-11-11:
On Sat, Nov 11, 2017 at 11:00:27PM +0000, Mateusz wrote:
> In slicer start g-code I pull with M42 my pin 63 to HIGH, in end-gcode I have a command which pulls it down. So after the print finish my printer is complete silent. I think allowing to control the [...]

You should be able to get the same effect by declaring a heater_fan.

-Kevin

dragonnn, 2017-11-11:
Hmmm yes that would be true but in my printer this pin shutdowns the entire power to the X carriage and I have put my thermocouple board on the X carriage. So I need to pull the pin high to get a temperature reading form the hotend. Specific situation but I think this is still a good idea to allow to control other pins, could be useful in other scenarios (controlling an laser module, controlling some leds etc.).

lenne0815, 2017-11-12:
In my case i want to do M42 P6 Sxxx to set my case lights on off / brightness, thats why i wanted to use the fan section because it can interpret "Sxxx"

dragonnn, 2017-11-12:
I tried doing it myself https://github.com/dragonnn/klipper/commit/403acb39e60a391e1f1d8b6ddfd11dcdbdc0ca03 I think it is maybe mostly done. Only two problems - the pin is set and then goes back and second after doing the command twice it gives an error "Missed scheduling of next pin event". Hmm i need to add check if the set value is the same as the old value.

dragonnn, 2017-11-12:
Ok I think I fixed it :D. I really starting to like how klipper is written :D. Creating a pull request

KevinOConnor, 2017-12-19:
I've added code to be able to set pins at run-time (commit 3c564921).  This takes into account the recent changes I've made to how g-code commands are registered.  It also supports both pwm and digital output pins.  See the updated config/example-extras.cfg file for details on how to setup and use.

closed 2017-12-27 by KevinOConnor

dragonnn, 2017-12-28:
Big thank you :D. You are amazing!

lenne0815, 2017-12-28:
This totally went unnoticed by me until you closed it, huge thanks !