Klipper3d/klipper #47: Unable to parse move

lenne0815, 2017-11-19:
Happens on longer g-code files with the multistepper branch, tried two different gcode base files. [Log](https://drive.google.com/file/d/1jZcg0M3ZU-w2Tgvkd8scKUCTb5x0Rh5e/view?usp=sharing)

KevinOConnor, 2017-11-19:
The log shows two instances of the error:
```Unable to parse move 'N220620 G1 X280.589 Y230.266 E-0.5000 F0*21'```
and
```Unable to parse move 'N215275 G1 X274.951 Y199.425 E-0.5000 F0*20'```
Both are invalid gcode commands - it's not valid to request a speed of zero (F0).

See if these commands are present in your gcode file - if so, then it seems like a real bug in the producer of the gcode.  If they are not present, then it would seem something went wrong in transmission - could you attach the gcode file?  Also, some additional information would be available by reproducing, issuing an M112 in the octoprint terminal window after the error, and then sending the klippy.log file.

Also, going forward it's easier for us if you gzip the klippy.log file and attach it to the github issue (not linking to google drive) as described in the docs/Contact.md document.

lenne0815, 2017-11-19:
Ok, didnt know its possible to attach something here ;)

I did a quick search of the gcode but couldnt find the offending F0

[panzer_4_track_left.zip](https://github.com/KevinOConnor/klipper/files/1485734/panzer_4_track_left.zip)

lenne0815, 2017-11-19:
Heres the log after M112 ( console was still in after error state )

[klippy.zip](https://github.com/KevinOConnor/klipper/files/1485737/klippy.zip)

KevinOConnor, 2017-11-19:
Well, that's a bit of a mystery - not only does the gcode file not contain an F0, but it doesn't have any move with an X274, X280, Y230, or Y199.  Are you sure it was that gcode file?

The klippy.log file after m112 will be necessary to further diagnose (alas, it has to be right after the error report - the m112 causes klipper to dump its recent gcode input - the log you have no longer shows the interesting gcode).

lenne0815, 2017-11-19:
Now this is really weird, you are right it wasnt the right g-code file, i redownloaded it again from octoprint and this time the file is a lot bigger and does in fact contain the F0 error !

Sorry if i pinned that problem wrongfully to klipper, never had that specific issue with s3d before, ill investigate on the slicer side of things. ( And i already found it, seems to be a bug with wiping :/ ) 

[panzer_4_track_left (1).zip](https://github.com/KevinOConnor/klipper/files/1485742/panzer_4_track_left.1.zip)

closed 2017-11-20 by lenne0815