Klipper3d/klipper #48: Klipper hangs on print

TunaLatte, 2017-11-22:
Hello, I will describe the symptom in order:

1. Klipper connects to mcu
2. Octoprint connects to Kliper without issues
3.Manual control, temperature control works perfectly.
4.Load a gcode, printer heatbed heatsup aswell as hotend, axis home
5.Printer does not respond on the first G0 gcode command, does not move etc.
6.After couple of minutes i get the following: 
> Recv: // Lost communication with MCU 'mcu'
> Recv: // Once the underlying issue is corrected, use the "RESTART"
> Recv: // command to reload the config and restart the host software.
> Recv: !! Printer is halted
7.Printer keeps stepper on, and heatcontrol is still working till I halt by power.

I tried changing USB cables, I might try external usb serial adapter if it is any related (once its shipped)

Hardware specs:
Arduino Mega 2560 (taurino clone, reprapdiscount)
RAMPS1.4

Thanks alot, looking forward to use klipper as a daily driver on all of my printers.

[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/1495798/klippy.log.gz)
[FilamentGuideABS.gcode.gz](https://github.com/KevinOConnor/klipper/files/1495817/FilamentGuideABS.gcode.gz)

KevinOConnor, 2017-11-22:
On Wed, Nov 22, 2017 at 11:54:11AM +0000, TunaLatte wrote:
> Hello, I will describe the symptom in order:
>
> 1. Klipper connects to mcu
> 2. Octoprint connects to Kliper without issues
> 3.Manual control, temperature control works perfectly.
> 4.Load a gcode, printer [...]

The nozzle_diameter in your config is incorrect and that is causing
the hang you are perceiving - it's actually moving the extruder at a
ridiculously slow rate.

The log you provided doesn't include the "Lost communication with MCU"
error.  The above error shouldn't cascade to a comms error, but it's
hard for me to further diagnose.  I suggest you correct the first
error and see if the problem persists.  If it does persist, please
attach a new log containing the error.

-Kevin

TunaLatte, 2017-11-22:
Please forgive me, my bad, it did indeed fix the issue, got no idea how and why I changed this, sorry!
I don't mind deleting this issue due to clutter at your request.
Thanks!

closed 2017-11-22 by TunaLatte