Klipper3d/klipper #49: Improvement: Material Dependence of Pressure Advance

laudiblellc, 2017-11-28:
Hi Kevin,
I have been running your firmware for several months and am very happy. One thing I think we could easily improve is to acknowledge and mitigate the fact that the pressure advance constant needs to change depending upon what type of plastic you are using in the machine. 
The solution consists of two parts:
1 - Create a table or guide to allow a quick change of advance constant based on material. 
I see this table as having a list of average advance constants. You can use the relative difference between materials on this table to offset your own constant. 
To achieve this table, I recommend that people conduct tests on their own printers and report the constants that they use. We can then aggregate this value and come up with a guide. 
2- Allow the constant to be changed via g-code or other online command. 
This is only a nice-to-have. Presently, you must edit the printer.cfg and restart Klipper. It would be ideal if you could simple put a custom G-code in your program that would set the advance that you desire. This would allow the printer settings to remain the same while still allowing the user to quickly print with different materials. This would also be extremely useful in the case of multiple extruders. 
Let me know what you think. I can supply you with my constants for around 10 different materials. 

Thanks,
Saul

KevinOConnor, 2017-11-29:
On Tue, Nov 28, 2017 at 08:27:12PM +0000, laudiblellc wrote:
> Hi Kevin,
> I have been running your firmware for several months and am very happy. One thing I think we could easily improve is to acknowledge and mitigate the fact that the pressure advance constant [...]

I'd definitely be interested in seeing your results with pressure
advance!

I agree that being able to update the pressure advance setting at
runtime would be a nice feature.  It's not difficult to implement, but
I'm not sure having users remember the magic g-code to do that is
better than having them update the config and issue RESTART.  Ideally
there would be some way to group together overall print options (like
bed temp, bed offset, nozzle temp, pressure advance, etc.) so that
users could update everything with a single command.

I'm certainly interested in seeing your results with pressure advance,
but I'm not sure it will translate directly to other printers.  In
addition to material, I suspect the extruder design and nozzle
temperature will also significantly impact the pressure advance
coefficient.  (And differences in thermistor calibration and placement
make it difficult to accurately compare temperatures across printers.)
As you suggest, gathering lots of data to compare results makes a lot
of sense.

Thanks,
-Kevin

dragonnn, 2017-11-29:
I don't think we need a material table in config. Just the option to set pressure advance via g-code. And btw. option to set other settings via g-code will be fine to like acceleration, junction and others.

lenne0815, 2017-11-29:
A table wouldnt make much sense imo either, a "gcode" command would be great though, with octoprints custom buttons one could quickly wip up a button field for all the different materials one uses. Im not 100% certain if "gcode" is really necessary either, i like the full text command approach klipper started using and with custom start end console commands it should work with any given slicer aswell. Problem with gcodes is that its prone to confusion, f.e. acceleration values in marlin vs repetier is just messy.

dragonnn, 2017-11-29:
Or you can add to you slicers start g-code if you have separate profiles (S3D allows have separate start g-code for profiles). And yes I agree it doesn't need to be a true g-code command, I like the "full text command approach" too.

BlackStump, 2017-11-29:
I personally like the idea that even the config is done with gcode.
I have one printer the uses the Duet board and DC42's Reprap firmware which is gcode everwhere.
It would be nice to have similar if not the same gcode so any gcode added to slicer would be the same for klipper and duet ...just a personal preference tho. 
Klipper imo is awesome firmware as it is but gcode everywhere would make it even better.
full text alias gcode best of both worlds

laudiblellc, 2017-11-29:
I feel that I need to explain the material table. I agree that everyone's mileage will vary, and their actual settings will differ. However, their **relative** settings between materials should be similar - for example, Nylon will generally need 50% higher advance than PLA. _That_ knowledge is the thing that I think will help the community. The alternative is for every user to repeat each test for themselves. This is wasted effort, in my opinion, and will be a stumbling block to a wider adoption of the pressure advance. 

As for the gcode / online modification question, I agree that there are lots of possibilities and opinions on what would be great. I for one would love an Octoprint plugin GUI to modify the config.

In the meantime, though, I will work on the ratios. 
Thanks,
Saul

lenne0815, 2017-11-30:
Hey Saul,

now i understand it better, but i think even for relative settings a guideline document would be enough once we can change the pressure advance modifier from the octoprint console.

For now with a windows box i found Winscp to be a great, quick way to change the config, edit, upload / overwrite, issue restart, done.

KevinOConnor, 2017-12-19:
Hi Saul,

Any updates on your tests with pressure_advance and various materials?

-Kevin

laudiblellc, 2017-12-19:
Hi Kevin,
No real updates to report.
I'm working on getting printer reliability so I haven't gotten the
documentation going yet. I will get some more data points soon.

I'm not sure if you want to keep this issue open  or not, just for keeping
tabs on things.

Saul

KevinOConnor, 2018-01-04:
I'm going to close this for now.  Feel free to reopen when ready.

closed 2018-01-04 by KevinOConnor