Klipper3d/klipper #50: Add support M220 M221 command.

khseal, 2017-12-02:
It is not possible to change speed and extrusion during printing.

hg42, 2017-12-19:
I have implemented M220 (speed factor), which was easy...and will create a PR once it is tested enough.
I could implement it at gcode level, so all is included in gcode.py

M221 (flow factor) is another beast.

I recently had one firmware, that did it wrong. Changing M221 mid-print caused excessive E movement.
I think it was Marlin2, the only other firmware I have used lately was Smoothieware, but this worked for years (I think, or I did never change the factor late in the print).

If we had only relative movement it would be easy.
With absolute movement mode it becomes more complex.
I'll try that later.

khseal, 2017-12-19:
Nice. Thank you for your work. 
Change of speed sometimes it is necessary.

hg42, 2017-12-20:
are you able to test the PR?

KevinOConnor, 2017-12-27:
FYI, there is code for M220/M221 available on the work-m221-20171222 branch.

KevinOConnor, 2018-01-03:
FYI, I committed this to the master branch (commit 3ab9a8d2) and I have removed the work-m221-20171222 branch.

closed 2018-01-04 by KevinOConnor