Klipper3d/klipper #52: Trying to start Klipper in Ubuntu 12.04

khseal, 2017-12-04:
I want to run a Klipper on an old laptop.
Octoprint works without problems.
Klipper hangs at startup and does not write an error.
Here is the log
Starting Klippy...
Args: ['/home/khseal/klipper/klippy/klippy.py', '/home/khseal/printer.cfg', '-l', '/tmp/klippy.log']
Git version: ''
CPU: 1 core Intel(R) Pentium(R) M processor 1.10GHz
Python: '2.7.3 (default, Oct 26 2016, 21:04:23) \n[GCC 4.6.3]'
Starting Klippy...
Args: ['/home/khseal/klipper/klippy/klippy.py', '/home/khseal/printer.cfg', '-l', '/tmp/klippy.log']
Git version: ''
CPU: 1 core Intel(R) Pentium(R) M processor 1.10GHz
Python: '2.7.3 (default, Oct 26 2016, 21:04:23) \n[GCC 4.6.3]'
What could be the problem?

KevinOConnor, 2017-12-04:
On Mon, Dec 04, 2017 at 09:33:38AM +0000, Sergey wrote:
> I want to run a Klipper on an old laptop.
> Octoprint works without problems.
> Klipper hangs at startup and does not write an error.

It's hard to say without further debug information.  Did you run the
scripts/install-octopi.sh script (or manually execute all of the steps
found there)?  Was the program launched from the startup script or
manually?

As a next step, I suggest making sure all instances of klippy are no
longer running (sudo service klipper stop ; sudo killall
~/klippy-env/bin/python ) and then try running the klippy service in
the foreground with:

~/klippy-env/bin/python ~/klipper/klippy/klippy.py ~/printer.cfg -v

and then cut-and-paste the full terminal output to this issue.

-Kevin

khseal, 2017-12-04:
The installation script started according to the instructions, it worked without problems.
Here log after command ~/klippy-env/bin/python ~/klipper/klippy/klippy.py ~/printer.cfg -v:
khseal@C110:~$ ~/klippy-env/bin/python ~/klipper/klippy/klippy.py ~/printer.cfg -v
INFO:root:Starting Klippy...
Traceback (most recent call last):
  File "/home/khseal/klipper/klippy/klippy.py", line 341, in <module>
    main()
  File "/home/khseal/klipper/klippy/klippy.py", line 329, in main
    printer = Printer(input_fd, bglogger, start_args)
  File "/home/khseal/klipper/klippy/klippy.py", line 132, in __init__
    self.reactor = reactor.Reactor()
  File "/home/khseal/klipper/klippy/reactor.py", line 130, in __init__
    SelectReactor.__init__(self)
  File "/home/khseal/klipper/klippy/reactor.py", line 37, in __init__
    self.monotonic = chelper.get_ffi()[1].get_monotonic
  File "/home/khseal/klipper/klippy/chelper.py", line 113, in get_ffi
    FFI_lib = FFI_main.dlopen(os.path.join(srcdir, DEST_LIB))
  File "/home/khseal/klippy-env/local/lib/python2.7/site-packages/cffi/api.py", line 139, in dlopen
    lib, function_cache = _make_ffi_library(self, name, flags)
  File "/home/khseal/klippy-env/local/lib/python2.7/site-packages/cffi/api.py", line 782, in _make_ffi_library
    backendlib = _load_backend_lib(backend, libname, flags)
  File "/home/khseal/klippy-env/local/lib/python2.7/site-packages/cffi/api.py", line 771, in _load_backend_lib
    return backend.load_library(name, flags)
OSError: cannot load library /home/khseal/klipper/klippy/c_helper.so: /home/khseal/klipper/klippy/c_helper.so: undefined symbol: clock_gettime

khseal, 2017-12-04:
I think I fixed the problem myself. Googled updated the library and the Klipper began to ask for the config.
Here instruction for solve problem https://forums.bistudio.com/forums/topic/186086-howto-installing-latest-stable-glibc-for-extdb-on-ubuntu-1204/

closed 2017-12-06 by khseal

sashalex007, 2019-07-20:
Ran into this issue, updating libraries was not an option. Added -lrt flag in c_helper.so compile commands. Order of flags seems to matter, this is what worked for me

`" -o %s %s -lrt")`