Klipper3d/klipper #53: Klipper timeout after AVR restart.

khseal, 2017-12-06:
AVR controller has its own power supply.
If turn off and turn on AVR power then Klipper is no longer connected to AVR.
Restart service solves the problem.
Example in log
Timeout with MCU 'mcu' (eventtime=3040.583039)
Timeout with MCU 'mcu' (eventtime=3041.584679)

KevinOConnor, 2017-12-06:
On Wed, Dec 06, 2017 at 05:49:13PM +0000, Sergey wrote:
> AVR controller has its own power supply.
> If turn off and turn on AVR power then Klipper is no longer connected to AVR.
> Restart service solves the problem.
> Example in log
> Timeout with MCU 'mcu' (eventtime=3040.583039)
> Timeout with MCU 'mcu' (eventtime=3041.584679)

That's by design.  If the host detects an error (in this case the mcu
resetting) then it enters an error state.  The error state is cleared
by issuing a FIRMWARE_RESTART command in the Octoprint terminal
window.

-Kevin

khseal, 2017-12-06:
Your command is working.  I need to figure out how to automate this. Thanks for the info.

closed 2017-12-07 by khseal