Klipper3d/klipper #54: Atmega328 PINs

pamalofeev, 2017-12-09:
Hello!
Firstly, I could not use arduino pins names, for example ar2, analog1 .. But it's not scary, atmega names like PC1, PC0 work fine, except for two. These are the ADC6 and ADC7 pins, they do not have gpio, they are only connected to the ADC. I use them for connecting thermistors, changing the board will be difficult. I watched klipper / klippy / pins.py and saw
```
  Arduino_analog_standard = [
     "PC0", "PC1", "PC2", "PC3", "PC4", "PC5", "PE0", "PE1",
]
```
I tried PE0 and PE1, but it also did not work, "Unable to translate pin name".
  For Teacup Firmware, these pins worked well.

pamalofeev, 2017-12-09:
OK, I fixed it.
In pins.py
`MCU_PINS = {
    "atmega168": port_pins(4), "atmega328": port_pins(5),     //replaced port_pins(4) by port_pins(5)
    "atmega644p": port_pins(4), "atmega1284p": port_pins(4),
    "at90usb1286": port_pins(6),
    "atmega1280": port_pins(12), "atmega2560": port_pins(12),
    "sam3x8e": port_pins(4, 32),
    "pru": beaglebone_pins(),
    "linux": {"analog%d" % i: i for i in range(8)}, # XXX
}`
`Arduino_analog_standard = [
    "PC0", "PC1", "PC2", "PC3", "PC4", "PC5", "PE2", "PE3",   //replaced PE0 PE1 by PE2 PE3
]`
`Arduino_from_mcu = {
    "atmega168": (Arduino_standard, Arduino_analog_standard),
    "atmega328": (Arduino_standard, Arduino_analog_standard),           //I'm not sure of the need
    "atmega644p": (Sanguino, Sanguino_analog),
    "atmega1280": (Arduino_mega, Arduino_analog_mega),
    "atmega2560": (Arduino_mega, Arduino_analog_mega),
    "sam3x8e": (Arduino_Due, Arduino_Due_analog),
}`

In /avr/gpio.c
`static const uint8_t adc_pins[] PROGMEM = {
#if CONFIG_MACH_atmega168 || CONFIG_MACH_atmega328
    GPIO('C', 0), GPIO('C', 1), GPIO('C', 2), GPIO('C', 3), 
    GPIO('C', 4), GPIO('C', 5), GPIO('E', 2), GPIO('E', 3),       //replaced ('E', 0) and ('E', 1) by ('E', 2) and ('E', 3)
#elif CONFIG_MACH_atmega644p || CONFIG_MACH_atmega1284p
    GPIO('A', 0), GPIO('A', 1), GPIO('A', 2), GPIO('A', 3),
    GPIO('A', 4), GPIO('A', 5), GPIO('A', 6), GPIO('A', 7),
#elif CONFIG_MACH_at90usb1286
    GPIO('F', 0), GPIO('F', 1), GPIO('F', 2), GPIO('F', 3),
    GPIO('F', 4), GPIO('F', 5), GPIO('F', 6), GPIO('F', 7),
#elif CONFIG_MACH_atmega1280 || CONFIG_MACH_atmega2560
    GPIO('F', 0), GPIO('F', 1), GPIO('F', 2), GPIO('F', 3),
    GPIO('F', 4), GPIO('F', 5), GPIO('F', 6), GPIO('F', 7),
    GPIO('K', 0), GPIO('K', 1), GPIO('K', 2), GPIO('K', 3),
    GPIO('K', 4), GPIO('K', 5), GPIO('K', 6), GPIO('K', 7),
#endif
};`

KevinOConnor, 2017-12-09:
On Sat, Dec 09, 2017 at 08:22:07AM +0000, pamalofeev wrote:
> Hello!
> Firstly, I could not use arduino pins names, for example ar2, analog1 .. But it's not scary, atmega names like PC1, PC0 work fine, except for two. These are the ADC6 and ADC7 pins, they do [...]

Thanks.  I think it should be fixed as of commit 4a5801bb (both
arduino mappings and support for PE0/1 or analog5/6).  Unless I'm
missing something, it shouldn't be necessary to change the
src/avr/gpio.c code as using PE0/1 should be fine.

-Kevin

closed 2017-12-15 by KevinOConnor