Klipper3d/klipper #56: MCU 'mcu' shutdown: Rescheduled timer in the past

khseal, 2017-12-11:
Lowered the speed of printing in the slicer and got this error.
[klippy (2).log](https://github.com/KevinOConnor/klipper/files/1549567/klippy.2.log)

KevinOConnor, 2017-12-12:
On Mon, Dec 11, 2017 at 02:09:19PM -0800, Sergey wrote:
> Lowered the speed of printing in the slicer and got this error.

Do you get successful prints at higher speeds?

Is there a g-code file that regularly causes this failure?  If so,
please attach the g-code file.

I looked at the log and it looks like the micro-controller has very
high interrupt load, but I don't know why that would be.  Just for
good measure, you should grab the latest code, reflash the
micro-controller, and see if the issue persists.

-Kevin

khseal, 2017-12-12:
Unfortunately, I erased the file g-code.
I printed at a speed of 60mm/s normally. Decreased the speed to 50 mm/s received this error twice in at the beginning of printing.
I'll try to print something else and if there is a problem, I'll send you a log and an g-code.

KevinOConnor, 2017-12-12:
On Tue, Dec 12, 2017 at 11:32:34AM -0800, Sergey wrote:
> I tried another model. The error is the same ...

Something very odd is occuring.  Can you run the following and report
what it says:  avr-gcc --version

Can you run the following and provide the resulting out.tar.gz file:
cd ~/klipper ; make clean ; make ; tar cfz out.tar.gz ~/klipper/out/

Can you also perform the following steps:

sudo service klipper stop
make clean
rm /tmp/klippy.log
make flash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
sudo service klipper start

Then wait 20 seconds, connect to klipper in octoprint, type M112 in
the Octoprint terminal window.  Then attach the resulting
/tmp/klippy.log file.

> A similar problem arose in pamalofeev

If this issue occurred on another machine, could you gather the
klippy.log file from it?

-Kevin

khseal, 2017-12-12:
>  A similar problem arose in pamalofeev
If this issue occurred on another machine, could you gather the
klippy.log file from it?
Pamalofeev is another user. He later will show his log.

> Something very odd is occuring.  Can you run the following and report
what it says:  avr-gcc --version
> Can you run the following and provide the resulting out.tar.gz file:
cd ~/klipper ; make clean ; make ; tar cfz out.tar.gz ~/klipper/out/

I collect the firmware for the AVR on one laptop. I'm printing on another laptop(Old laptop have very old avr-gcc).Perhaps because of this problem? 
I will attach the files a little later.

pamalofeev, 2017-12-13:
Yes, I confirm, there are errors at some speeds. I did for max_z_velocity 6mm / s and everything worked fine. Then I made 5mm / s and there were errors. For 4mm / s there are no errors. I attach logs for different maximum speeds for z.
[5klippy.log](https://github.com/KevinOConnor/klipper/files/1556301/5klippy.log)
[6klippy.log](https://github.com/KevinOConnor/klipper/files/1556302/6klippy.log)
[4klippy.log](https://github.com/KevinOConnor/klipper/files/1556303/4klippy.log)

KevinOConnor, 2017-12-13:
On Wed, Dec 13, 2017 at 04:51:44PM +0000, pamalofeev wrote:
> Yes, I confirm, there are errors at some speeds. I did for max_z_velocity 6mm / s and everything worked fine. Then I made 5mm / s and there were errors. For 4mm / s there are no errors. I attach logs for different maximum speeds for z.

You're hitting a bug separate from this issue.  There is logic to
round the homing speed to an even multiple of steps that was
implemented in commit 002dc0df, but it seems that commit does not work
when the homing speed for the Z axis (default of 5mm/s) is greater
than or equal to the max_z_velocity.  I'll put together a fix, but for
now you should be able to work around it by explicitly specifying a z
homing_speed strictly lower than the max_z_velocity.

The original issue appears to be extremely high interrupt/timer load
for no explainable reason.  It does not appear to be related to
homing.

-Kevin

KevinOConnor, 2017-12-13:
On Tue, Dec 12, 2017 at 08:51:35PM +0000, Sergey wrote:
> I will attach the files a little later.

Does the problem persist after you flashed the micro-controller again?

Can you provide a log containing just an M112?  (In the octoprint
terminal tab, type FIRMWARE_RESTART, wait 20 seconds, type M112, then
send the klippy.log file.)

Do you know if this issue is a regression?  (Did it used to work a few
versions ago and stopped working with a recent code pull?)

-Kevin

khseal, 2017-12-13:
> Do you know if this issue is a regression?  (Did it used to work a few
versions ago and stopped working with a recent code pull?)

I recently started using a Klipper. Therefore I can not answer you this question.

> Does the problem persist after you flashed the micro-controller again?

I'll try to check in the next time.

> Can you provide a log containing just an M112?  (In the octoprint
terminal tab, type FIRMWARE_RESTART, wait 20 seconds, type M112, then
send the klippy.log file.)
Here this log
[klippy.log](https://github.com/KevinOConnor/klipper/files/1557133/klippy.log)

khseal, 2017-12-14:
Nothing changed after update firmware. Bug also occurs at lower speeds.

KevinOConnor, 2017-12-16:
On Thu, Dec 14, 2017 at 04:04:49AM -0800, Sergey wrote:
> Nothing changed after update firmware. Bug also occurs at lower speeds.

I'm not sure what is occurring.  I've looked through the logs and they
seem to indicate the AVR chip is under high irq/timer load.
Unfortunately, I don't see why that would be.  Your config looks
pretty standard and the host software isn't reporting anything odd.

The log indicates the ADC readings are showing wild fluctuations in
delivery time, the log contains "Resetting prediction variance"
messages, and the error is a "Reschedule timer in the past" - all are
consistent with high timer/irq load.  Unfortunately, nothing in the
log of recently sent mcu commands nor the config would indicate
anything that could cause such a high load.

I put together a debugging code branch to output more data on the AVR
scheduling.  It requires a reflash of the mcu:

git fetch ; git checkout origin/work-timerdebug-20171216
make flash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

Send the klippy.log file if you can reproduce the error with the above
code.

-Kevin

khseal, 2017-12-16:
I'll try to check in the near future.

khseal, 2017-12-16:
Strange things. Printing on the firmware for debugging was successful.
Log of successful printing.
[klippy.log](https://github.com/KevinOConnor/klipper/files/1564712/klippy.log)

khseal, 2017-12-18:
I think the problem is gone. I'll try different speeds.

KevinOConnor, 2017-12-20:
On Wed, Dec 13, 2017 at 04:51:44PM +0000, pamalofeev wrote:
> Yes, I confirm, there are errors at some speeds. I did for max_z_velocity 6mm / s and everything worked fine. Then I made 5mm / s and there were errors. For 4mm / s there are no errors. I attach logs for different maximum speeds for z.

FYI, I believe this particular issue ("timer too close" errors on
homing when the max_z_velocity is greater than the homing_speed)
should be fixed now (commit 1d21bf66).

-Kevin

khseal, 2017-12-21:
I checked the high speed (100mm/s). There error "timer too close"
Here is the log

KevinOConnor, 2017-12-21:
On Thu, Dec 21, 2017 at 11:53:43AM +0000, Sergey wrote:
> I checked the high speed (100mm/s). There error "timer too close"
> Here is the log 

It looks like there is a defect in the processing of "shutdown" events
on the AVR platform (specifically, interrupts could be incorrectly
enabled by longjmp).  This doesn't explain your failures, but it could
cause the debugging to be misleading.

Your latest log has similar symptoms to the early ones (high latency
in thermistor readings).  It's possible this is due to a host issue
instead of a micro-controller issue.  I'm wondering if perhaps your
host machine is sleeping the USB bus to save power?  If so, it could
be sleeping too long (200+ milliseconds) and causing these problems.

In any case, you should grab the latest code to fix the shutdown issue
(commit d303e556).

-Kevin

khseal, 2017-12-22:
> Your latest log has similar symptoms to the early ones (high latency
in thermistor readings).  It's possible this is due to a host issue
instead of a micro-controller issue.  I'm wondering if perhaps your
host machine is sleeping the USB bus to save power?  If so, it could
be sleeping too long (200+ milliseconds) and causing these problems.

I do not know how Linux works with USB. The laptop is powered by a power supply.
Perhaps there is a USB high load in old notebook. I have a web camera connected to USB. I'll soon replace the camera.
I need to look for software that measures the delay of USB under Linux.
P.S.
Recently, my all USB interface is cursed. On other devices, there are problems with USB that I have never met. =)

khseal, 2017-12-26:
Bad news. Bug alive...
[klippy.log](https://github.com/KevinOConnor/klipper/files/1587568/klippy.log)

KevinOConnor, 2017-12-28:
On Tue, Dec 26, 2017 at 08:10:55AM -0800, Sergey wrote:
> Bad news. Bug alive...
> [klippy.log](https://github.com/KevinOConnor/klipper/files/1587568/klippy.log)

This log shows similar behavior to past logs - erratic ADC times
followed by a shutdown.  This log would generally indicate it is not
related to USB - it seems to be something causing high timer/irq load
on the micro-controller.  I'm not sure what's going on.  The problem
seems localized to your environment, but I don't see anything special
about your mcu config.

I've updated the work-timerdebug-20171216 branch with more debugging.
I'd recommend running with that branch for any future tests:

git fetch ; git checkout origin/work-timerdebug-20171216 ; make flash ...

-Kevin

khseal, 2017-12-28:
> I've updated the work-timerdebug-20171216 branch with more debugging.
I'd recommend running with that branch for any future tests:

It was a strange thing. The debug version did not stop at low speed.
I will continue testing for the version version.

khseal, 2017-12-29:
I try test work-timerdebug-20171216. Again bug leave.  In debug version of this error disappears.

KevinOConnor, 2018-01-08:
> On Sat, Jan 06, 2018 at 11:03:06PM +0000, Sergey wrote:
> I have a new bug. Here is the log
> klippy.log
> Twice printed and an error occurred twice.

This log shows the same symptoms as previous logs - high variance in
ADC response times.  The debugging further indicates that the time it
takes to read the ADC values is highly variable - normally a reading
(with 8 oversamples) takes about 8 milliseconds to complete.  However,
we see in various places in the log where the cycle is taking over
250ms (see the "#output: lst" lines).

> I'll try to roll back to the old version.
> It seems the error is due to the poor power of the AVR controller.

I'm inclined to agree that this is something specific to this
particular chip or setup.

-Kevin

khseal, 2018-01-08:
It seems that after the usb web camera has been disconnected, the error has gone.
I tried to change the power of the microcontroller and burned completely the control board. =( 
It is better not to do anything when you have not slept much.
Ordered by Arduino Due.

closed 2018-01-17 by KevinOConnor