Klipper3d/klipper #58: Support for Simultaneous two ramps board

NeelD87, 2017-12-15:
Is it possible to drive two boards simultaneously to as to support more axis.
XYZ on one board and UV on other?

KevinOConnor, 2017-12-15:
On Fri, Dec 15, 2017 at 12:30:43AM -0800, NeelD87 wrote:
> Is it possible to drive two boards simultaneously to as to support more axis.
> XYZ on one board and UV on other?

I don't know what UV is, but yes - in general it is possible to add
additional micro-controllers to control additional stepper motors.
See docs/Features.md, the mcu section of config/example-extras.cfg,
and config/generic-replicape.cfg as an example.

Implementing motion across multiple mcus has been unit tested, but
could use additional testing in real world usage.

-Kevin

hg42, 2017-12-15:
I did a first print yesterday with two 8bit boards (RAMPS + Melzi derivate).
One controlled XYZ and one did E and all thermistors.
This worked very well.

closed 2017-12-19 by KevinOConnor