Klipper3d/klipper #59: Klipper requires rehoming after end of print job.

JackBradach, 2017-12-16:
After a print, I try to move the print head (using Octoprint's UI) but the terminal spits up, eg:
Recv: !! Must home axis first: 105.483 111.374 10.000 [62024.281]

It seems to immediately forget its position as soon as a print job ends.  This becomes a problem if I need to move the head out of the way before clearing the bed; the extruder may not have a clear path 'home'.

KevinOConnor, 2017-12-16:
On Fri, Dec 15, 2017 at 07:58:16PM -0800, Jack Bradach wrote:
> After a print, I try to move the print head (using Octoprint's UI) but the terminal spits up, eg:
> Recv: !! Must home axis first: 105.483 111.374 10.000 [62024.281]
>
> It seems to immediately forget [...]

See "Why can't I move the stepper before homing the printer?"  in the
FAQ.  In short, update Octoprint and/or slicer settings so that the
head is moved out of the way before disabling the motors (or change
them to not disable the motors).

-Kevin

geodave810, 2017-12-19:
I had a similar issue.  Using Marlin firmware with octoprint, I could move the Z axis up a few mm's after a failed print so it would be above the binder clips when homing the x & y.  I could also issue a G01 Z5 code after a finished print with Marlin. I looked in the FAQ & I see in Octoprint/Settings/GCODE Scripts it has the M84 for when a job is canceled.  I do not have klipper installed at the moment, but it seems like it had the same behavior after a successful job & there is not gcode in that section. How come the Marlin firmware works with the M84 in place & klipper does not?

KevinOConnor, 2017-12-19:
On Tue, Dec 19, 2017 at 06:08:34AM -0800, David wrote:
> I had a similar issue.  Using Marlin firmware with octoprint, I could move the Z axis up a few mm's after a failed print so it would be above the binder clips when homing the x & y.  I could also [...]

I'm not sure I understand your question.

After a successful print job, the slicer usually emits the M84 (or
equivalent M18) - see the slicer settings.

Marlin (at least when I last used it) will happily drive the print
head into a wall or bed if one accidentally commands it to - it
doesn't check positions after an M84.

Head crashes were a huge frustration for me when I was tuning my first
printer.  I like having some basic sanity checks in the firmware I
use.

-Kevin

geodave810, 2017-12-19:
Ok. I understand now.  Thanks for the clarification.  You are right the Slic3r does have the M84  at the end G-Code.

closed 2017-12-27 by KevinOConnor