Klipper3d/klipper #61: Linking fans to operate in tandem

sapell, 2017-12-17:
Do you have any plans to implement support for e.g. dual fan part cooling?
I'm using Klipper with Replicape and have dual fans for part cooling with both fans connected to a dedicated fan connector. Even better if there is a way to do it currently that I'm unaware of.

dragonnn, 2017-12-17:
I don't understand? Why not just connecting fans in parallel

BlackStump, 2017-12-17:
The Replicape has 4 fan connections on the board. (B3 ver)
so in config you can add a header as such to enable 2 cooling fans set to operate when extruder reaches 50 degrees

[heater_fan my_cooling_fan1]
pin: replicape:power_fan1
heater: extruder
heater_temp: 50.0

[heater_fan my_cooling_fan2]
pin: replicape:power_fan2
heater: extruder
heater_temp: 50.0

you can change heater: extruder to heater: bed if you prefer tho I have not tested that.
and 
heater_temp: to any temp you want the cooling fans to come on.
hope this helps

sapell, 2017-12-17:
Thanks for the replies!
Actually the two fans are not for heaters but instead part cooling fans (operated by M106) attached to the end effector with two cooling shrouds. Sure, the fans could be connected in parallel, but since there are 4 outputs for fans, why not utilize them. Also I believe it's better to have the fans on separate outputs rather than loading one output with two fans.

BlackStump, 2017-12-17:
Then use this but Kevin can chip in If I am wrong.
[fan]
pin: replicape:power_fan1
pin: replicape:power_fan2

sapell, 2017-12-17:
Thanks for the tip! Must try that tomorrow.

KevinOConnor, 2017-12-18:
On Sun, Dec 17, 2017 at 01:32:55PM -0800, Simo Apell wrote:
> Actually the two fans are not for heaters but instead part cooling fans (operated by M106) attached to the end effector with two cooling shrouds. Sure, the fans could be connected in parallel, but [...]

There is currently no way to define two print cooling fans.  It would
require host software changes to enable that.  Alternatively, one
could change the hardware to control both fans from the same power
source.

Let me know if you're up for testing new software - I think I can put
together the software changes if you want to pursue that.

-Kevin

BlackStump, 2017-12-18:
I can't do any testing for another month or so my replicape/printers are in storage until I move to a more permanent location.
Hopefully " Sapell" can test as having the option to use all 4 fan connector as thermo fans or gcode fans would be  great.   Thanks Kevin.

sapell, 2017-12-18:
Thanks for the reply Kevin! I can test new software, it's just that I can only run shorter tests in the evenings due to work and small kids.
- Simo Apell

KevinOConnor, 2017-12-19:
I've updated the code to add this feature (commit 1b075059).  See the
"multi_pin" feature in config/example-extras.cfg.  In brief, add a new
"[multi_pin my_fan]" section with "pins=mypin1,mypin2,..." defining
your fan pins.  Then change your "[fan]" section to read
"pin=multi_pin:my_fan".

-Kevin

sapell, 2017-12-19:
Brilliant! I'll give it a try soon. Thank you very much Kevin!

- Simo

sapell, 2017-12-19:
Works beautifully, superior job!

Issue resolved, closing ticket.

closed 2017-12-19 by sapell