Klipper3d/klipper #62: M18 not turning off motors with Replicape

sapell, 2017-12-17:
I can't get the M18 g-code to respond expectedly, the steppers do not disable.
I have a Replicape configured with Klipper and attached is my config.
Klipper was cloned from master branch yesterday.

[printer.cfg.zip](https://github.com/KevinOConnor/klipper/files/1566107/printer.cfg.zip)

KevinOConnor, 2017-12-18:
On Sun, Dec 17, 2017 at 09:41:32PM +0000, Simo Apell wrote:
> I can't get the M18 g-code to respond expectedly, the steppers do not disable.
> I have a Replicape configured with Klipper and attached is my config.
> Klipper was cloned from master branch yesterday.

To report an issue, please include the klippy.log file as described in
docs/Contact.md.  If Klipper isn't producing an error message during
the event, then be sure to issue an M112 prior to collecting the
klippy.log file (as described in the above document).

-Kevin

sapell, 2017-12-18:
Sorry about forgetting the log. I started a fresh log, issuing first G28, then M18 and shortly after issued M112. The log is attached. The steppers are firmly holding regardless of the M18.
- Simo

[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/1568524/klippy.log.gz)

sapell, 2017-12-18:
Seems that the previous log file is incomplete, here's a new one. Also issued M84 after M18 which should have the same function.
[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/1569272/klippy.log.gz)

KevinOConnor, 2017-12-19:
On Mon, Dec 18, 2017 at 07:51:31PM +0000, Simo Apell wrote:
> Seems that the previous log file is incomplete, here's a new one. Also issued M84 after M18 which should have the same function.

On the Replicape, Klipper disables the steppers by reducing the
stepper motor driver reference voltage to zero.  Thus, the stepper
enable line is still on, but the maximum current should be zero.  As I
understand it, this is the only practical way to disable the steppers
on the Replicape (as the replicape shift registers can't really be
changed at run-time because there is no way to update them
atomically).

The log you sent indicates the reference voltage on the steppers was
commanded to a non-zero value during the homing operation and then
commanded back to zero on the M18 command.  So, everything looks
normal in the log.

Are the steppers able to freely move prior to the G28?

Do heaters and fans work properly - can you both enable them and turn
them off?

Were you able to move the motors after issuing the M112 command?

What replicape revision and hardware are you using?

-Kevin

sapell, 2017-12-19:
Curiously the C-tower (Z-stepper) is free to move during boot for a while, but A and B towers are 'locked'. Once Klipper gets loaded, all the steppers are locked.

The steppers were/are not free to move prior to the G28.

Heater and fan control works as expected.

The steppers are held locked even after M112.

My replicape is of B3 revision, other relevant hardware is BeagleBone Black, revision of which I can't easily check as it's under the Replicape and both are under my build plate, but surely its revision C and guessing C3 more exactly as it's failry recently acquired.

For the record, I have not carried out the 'Rev. B stepper noise during boot' HW fix yet, as it seemed unnecessary with the Redeem firmware. Redeem handles the M18/M84 properly even without the fix. [More details here](http://wiki.thing-printer.com/index.php?title=Replicape_rev_B#Hardware_details). I was thinking to take a peek how the M18/M84 is implemented in [Redeem](https://bitbucket.org/intelligentagent/redeem/src/0129bd0204f7/redeem/?at=develop).

jakep82, 2018-01-01:
I'm bench testing a Replicape and experiencing the same issue, but it's a bit more nuanced.  After issuing an M84, the motor becomes much easier to move, but it's clearly not completely off.  I'm on the Replicape Slack board, and I found a little info there.  Quoting the relevant info 

> "I believe enables are done over the serial interface (SPI). Step/dir are driven directly"

 and 

> "Look in the class redeem.Steppers.Stepper_00B3 for the enable/disable functions. X/Y/Z use different address than E/H (the shift register state is set via spi)."

sapell, 2018-01-05:
Thanks jakep82! I'm also in the Replicape Slack, but as I've sort of given up on Redeem I'm not active there. I'll have to look at the Redeem code at some point if I can make any sense of it, have very little experience on the C++ syntax. BTW, I have now implemented the HW fix for the stepper noise during boot, but it has no effect on this issue as it shouldn't, but silenced the steppers during boot.

jakep82, 2018-01-05:
I have a Rev B3A which includes the fix for the noise on boot issue.  I'm glad because I don't have soldering skills to do it myself.

sapell, 2018-01-06:
@jakep82 That's weird as I also have B3A but had the noise issue. It wasn't too hard to do, actually found an easier mounting method for the resistors than in the Replicape documentation.

KevinOConnor, 2018-01-09:
I've uploaded a new branch ( work-replicape-20180108 ) with code to enable / disable the replicape steppers dynamically.  This branch should also support static_pwm_output pins (issue #104).  I don't have a good way to test this, so let me know your results - success or failure.

git fetch ; git checkout origin/work-replicape-20180108 ; sudo service klipper restart

sapell, 2018-01-09:
Great, thanks a lot Kevin! Will try the branch as soon as possible.

jakep82, 2018-01-09:
I only had a few minutes to play with it tonight, but everything appears to be working as expected. The steppers are now completely off until activated, and turn off after an M84.  The static pwm fan also stays on even when everything else is shut off.

sapell, 2018-01-09:
I tested the branch and can confirm that the M18 and M84 now properly disable the steppers, closing this issue. However, for me the static pwm fan turns off when issuing M18 or M84 as opposed to how it works for jakep82. This is not necessarily an issue and good even if static pwm fan is used for cooling electronics, namely the stepper drivers as I do, no cooling needed for inactive stepper drivers.

Was the intention that the static pwm fan responds to M106 and M107 when stepper activity is not overriding the static pwm fan control, that's how it works for me?

KevinOConnor, 2018-01-09:
The static pwm turning off on an M18 is an error in the test code.  I'll look into fixing it.

Lets leave this issue open until we get the code merged into the master branch.

KevinOConnor, 2018-01-10:
I've updated the work-replicape-20180108 branch with code that should better handle static_pwm_output pins.  The disable steppers on M112 is not yet implemented.

git fetch ; git checkout origin/work-replicape-20180108 ; sudo service klipper restart

sapell, 2018-01-10:
I can confirm that now the static pwm stays on even after M18 or M84.

KevinOConnor, 2018-01-11:
On Wed, Jan 10, 2018 at 10:17:37PM +0000, Simo Apell wrote:
> I can confirm that now the static pwm stays on even after M18 or M84.

Thanks.  I've pushed the current support to the master branch.  Lets
leave this issue open until the disable on M112 can be addressed.

-Kevin

sapell, 2018-02-23:
Has the disable on M112 been addressed in a past commit or is it still on the do-to list? I have not actively tested the behavior after M112. Also I have switched back to Arduino/RAMPS from Replicape, was it so that the motor disabling issue after M112 only applies to Replicape?

KevinOConnor, 2018-02-23:
On Fri, Feb 23, 2018 at 09:56:10AM +0000, Simo Apell wrote:
> Has the disable on M112 been addressed in a past commit or is it still on the do-to list? I have not actively tested the behavior after M112. Also I have switched back to Arduino/RAMPS from Replicape, was it so that the motor disabling issue after M112 only applies to Replicape?

It's still on the todo list.  It requires mcu to be able to generate
an spi message transmission on shutdown (which will be necessary for
the tmc2130 as well).  It is only an issue for the replicape - as it
is specific to the pca9685 chip.

-Kevin

KevinOConnor, 2018-05-07:
FYI, this should be fixed as of commit f70fefa0.  However, I don't have a good way of testing this, so please let me know the results.

closed 2018-05-14 by KevinOConnor