Klipper3d/klipper #64: should limit errors stop a print?

hg42, 2017-12-19:
I got an error like "this move would extrude too much" (something like this was expected while testing M221 S1000 !).

Thoughts:
* ideally the processor could fix that by slowing down, but it would either need look ahead or some error while deaccelerating has to be accepted. The slow down could be simply managed by changing the speed_factor which I am currently testing.
* the exceeding value could be limited, which may give a quality error, but there is a chance that it is not noticable. In case of extrusion it is not unlike a skipping stepper.

In general:
* should limits (or other errors) be corrected or worked around or should they completely stop a print?
* the strategy for such errors could be different from user to user. Some need a perfect result and want to restart the print, others might accept reduced quality in these cases. Some might hate if a 10 days print is stopped at the last day...
* there could be a weight for each error which adds to a score. If the score reaches a limit it could result in a stop
* the duration of the print could raise the limit for this score

BlackStump, 2017-12-19:
I think this is the section you need to adjust ....you will find in example.cfg
#max_extrude_cross_section:
#   Maximum area of the cross section of an extrusion line (in
#   mm^2). This setting prevents excessive amounts of extrusion during
#   relatively small XY moves. If a move requests an extrusion rate
#   that would exceed this value it will cause an error to be
#   returned. The default is: 4.0 * nozzle_diameter^2

I had issues with this section when trying to use my purge routine maybe the default should be raised?

hg42, 2017-12-19:
thx, but sorry, this is not my problem :-)

I forced an extrusion that is insane while testing my implementation of M221. I used a flow of 1000%...

This issue is more about how something like this should be handled, where the title tells half of the story.

KevinOConnor, 2017-12-29:
I don't think there is a "general" answer to this type of question.  The appropriate way to handle an error depends on the error.  For example, if a thermistor reports the heater is above max_temp, then the heater should be turned off and an error raised.  In contrast, if a move is requested at a speed higher than the maximum speed then we should just limit the speed to the maximum.

I'd say the behavior for exceeding max_extrude_cross_section is correct - the check is there to prevent silly errors that could potentially harm the machine (like accidentally issuing an absolute E distance when in a relative E mode) - in that case, I think the only sensible thing is to let the user know the command was in error.

If there are some errors that you don't think should be errors, please open an issue with the log and gcode containing the error along with a description of what you think should have been done instead.

hg42, 2018-01-01:
So max_extrude_cross_section should lie far beyond the physical limit?

I think, for such limits there are generally two possibilities:
1. a limit chosen carefully below the physical limit, to actually prevent bad effects, ideally by adjusting the calculation to obey the limit so there would be no negative effect on the print, or if that is not possible or too much effort, by simply limiting this operation as a protection, giving some effect in the print result, but no damage.
2. a limit generously chosen above the physical limit preventing insane commands to be executed at all.

May be if there is such a distinction, the name should express that. As a non-native English speaker I can only guess.

Limits like these:
```
max_velocity
max_accel
```
are respected by calculation, but others, while having similar naming will (may be unexpectedly) stop the print:
```
min_temp
max_temp
max_extrude_cross_section
position_max (?)
```
I would suggest a naming like "stop_max_temp" or "fatal_max_temp" or "error_max_temp" to be more explicit about their meaning.
Or as an alternative, the other way, "limit_max_velocity", "limit_max_acceleration"...

closed 2018-01-11 by KevinOConnor