Klipper3d/klipper #66: Internal error on command:"G28"

sapell, 2017-12-19:
I get this error every now and then.

Running Klipper on BeagleBone Black (C) and Replicape (B3).
Log: [klippy.log.gz](https://github.com/KevinOConnor/klipper/files/1573323/klippy.log.gz)

- Simo Apell

sapell, 2017-12-19:
Seems to occur when using Octoprints homing button, which issues G91, G28 Z0 and G90 in succession. Does not occur when issuing only G28 from Octoprint terminal.

KevinOConnor, 2017-12-20:
On Tue, Dec 19, 2017 at 01:30:12PM -0800, Simo Apell wrote:
> I get this error every now and then.
> 
> Running Klipper on BeagleBone Black (C) and Replicape (B3).

That's unexpected.  What version of Linux did you install on the
Beaglebone (cat /etc/debian_version)?

As a test, could you change HOMING_STEP_DELAY at the top of
klippy/homing.py to:

HOMING_STEP_DELAY = 0.000001

And see if the problem goes away?

-Kevin

sapell, 2017-12-20:
I have Debian 9.3 installed according to /etc/debian_version.
The version has probably bumped a notch as a result of 'apt upgrade', as I'm sure I downloaded the latest available image through https://beagleboard.org/latest-images, but the latest there is currently 9.2.

Changed the HOMING_STEP_DELAY to 0.000001, at least right after restart it not throwing the error. Albeit, yesterday the errors started to occur after more than hour of use and manual movements through Octoprint. I'll leave the ticket still open for a while, if the error shows up again.

sapell, 2017-12-22:
It seems that the issue was due to the mjpg-streamer loading thd Beaglebone too heavily, too eager webcam settings. So sort of barking up the wrong tree here. Closing the issue.

closed 2017-12-22 by sapell