Klipper3d/klipper #68: suggestion: layers, modules, plugins

hg42, 2017-12-21:
I assume users of Klipper will be mostly power users and they will have very different goals.

I think, this can be solved by
* using strict layers
* clear protocol definitions between layers
* modules (plugins, add-ons, ...)
* a hierarchical configuration system that allows to access every parameter in every module (and documentation belongs to these modules).

This allows alternative implementations of layers and modules.
Users can create modules.
Users can choose from modules.
Modules that are not used do not disturb functionality.

This might be a bit difficult. E.g. gcodes like M220 and M221 can be implemented as module, but some need support in a lower layer that implements the feature.
I guess this could be solved by a registration system that inserts postprocessing between the layers.
This would be easy if the modules only communicate through channels. Then you could insert processing modules.

KevinOConnor, 2017-12-29:
I certainly attempted to make the code modular.  It's clear that more work is necessary - for example there are variants in homing, temperature sensors, and bed leveling probes that could use further abstractions.  Better documentation and support for "plugins" would also be a big plus.

I don't think github issues are the best way to raise very general topics like the above.  I don't see a time when consensus will be found on such a general topic, and that leaves the issue open indefinitely.  Perhaps the mailing list is a better fit.

hg42, 2017-12-31:
ok, so lets close this, I still did not subscribe to the emailing list, but I will if it's your favorite communication channel

closed 2017-12-31 by hg42