Klipper3d/klipper #70: enhancement: Reading Steinhart Hart coefficients for thermistors from config

gdachs, 2017-12-23:
I have calculated the Steinhart Hart coefficients for my unknown thermistors. I used a K-Type probe and some math for this. I would like to have a possibility to enter this values into the klipper configuration.
I don't think I can enter these values in the Sensors dictionary  in heater.py, because I have no name for the thermistors.
I could try to do it myself and could send you a pull request, but I would like to know before how you would like it. 

One way could be to add sensors to the config that could be added to the builtin sensors dictionary before first use, so that they could be referenced later in the config.

Another possibility is to have a new config option:
sensors_steinhart_hart_coefficient = <t1>,<r1>,<t2>,<r2>,<t3>,<r3>
This setting would overwrite the sensor_type value and would set the class to thermistor.
This is how smoothieware is doing it, I don't know which is better.

KevinOConnor, 2017-12-24:
On Sat, Dec 23, 2017 at 07:42:07PM +0000, gdachs wrote:
> I have calculated the Steinhart Hart coefficients for my unknown thermistors. I used a K-Type probe and some math for this. I would like to have a possibility to enter this values into the klipper [...]

I hadn't thought of that approach, but it makes sense.

> Another possibility is to have a new config option:
> sensors_steinhart_hart_coefficient = <t1>,<r1>,<t2>,<r2>,<t3>,<r3>
> This setting would overwrite the sensor_type value and would set the class to thermistor.
> This is how smoothieware is doing it, I don't know which is better.

I'm not sure which is better either.  I'd lean towards the second
approach.  Though, I'd suggest something like "sensor_type =
custom_thermistor" and then six "custom_thermistor_temperature1 =
123.4", "custom_thermistor_resistance1 = 345.6", etc. fields.

Thanks,
-Kevin

gdachs, 2017-12-24:
I will try to implement it, if I find the time, but I have to warn you. I am a very experienced C/C++ programmer, but Python I can nearly read only. So it might get more work to review my code than to do it yourself. 
As I told some seconds ago on IRC, I currently have only an old melzi board for tests and my new printer looks like that: https://www.thingiverse.com/thing:2641247, the 2. picture, so I will need some time to do it.

hg42, 2017-12-24:
I think the modular approach is more attractive.
I would even put each sensor in a file.
Then these files could be exchanged by users and simply dropped into a directory.

gdachs, 2017-12-24:
hg42: with modular approach you mean my first option? 
Generally it is a good idea, but for me it would be useless, as I just don't know which thermistor they have put into my silicone heater and e3d clone. The beta values I got were of no help. The temperature I got with the e3d clone was 30°C off, when I used the beta value.
So what should I put into this sensor file, that could help other users?

hg42, 2017-12-25:
@gdachs yes, I mean the first option, to create a new sensor (e.g. [temperature_sensor custom] in your case).
I think the work is nearly the same, but the outcome is better with creating a new sensor like this.

gdachs, 2017-12-25:
Yes, you are right. I start to like this more. Would allow to add all types of currently known sensors. Maybe the builtin sensors should be removed even and should go into the config.

KevinOConnor, 2018-02-25:
I'm going to close this as it doesn't appear that anyone is actively working on developing it.

closed 2018-02-25 by KevinOConnor