Klipper3d/klipper #72: Can't establish connection to printer "The klippy host software is attempting to connect"

chrzr, 2017-12-27:
Hello there!

After reading about Klipper on Hackaday today, I decided that I want to install it on my Raspberry Pi 2/Octoprint/CR-10 combo. I really like the idea of utilizing the spare computational power of the RPi for more than just sending gcode and displaying the UI!

The installation went just fine and I also managed to flash the firmware to the printer's SMELZI board without any error, using the "avrdude -p atmega1284p -c arduino -P /dev/ttyUSB0 -U out/klipper.elf.hex" command provided in the generic-melzi.cfg file.
Unfortunately, I am not able to establish the connection between the host and the MCU, as I only get the following message:

"The klippy host software is attempting to connect.  Please retry in a few moments.
Printer is not ready"

I already checked the printer.cfg file and changed the port from /dev/ttyUSB0 to /dev/ttyACM1, but that also did not work. I also tried the reboot and firmware_reboot command, as well as rebooting octoprint and the RPi without any success.
Is there anything else I could try, or is the SMELZI simply not supported yet?

Thanks in advance!

chrzr

KevinOConnor, 2017-12-27:
On Wed, Dec 27, 2017 at 12:20:42AM +0000, chrzr wrote:
> Hello there!
>
> After reading about Klipper on Hackaday today, I decided that I want to install it on my Raspberry Pi 2/Octoprint/CR-10 combo. I really like the idea of utilizing the spare [...]

To report an issue, please include the klippy.log file as described in
the docs/Contact.md file.  Also, please test if the steps in "When I
restart my micro-controller the device changes to /dev/ttyACM1" helps
with your printer.

-Kevin

chrzr, 2017-12-27:
Thanks for the quick response! Sorry for not including the logfile!
I've found the error ... *facepalm*
I overlooked the "baud" setting in the printer.cfg, as I thought it would just take the same setting as set in Octoprint. So it just defaulted to 200000 instead of 115200. After setting it in the printer.cfg the communication works! :)

KevinOConnor, 2017-12-27:
Glad to hear you found the issue.

closed 2017-12-27 by KevinOConnor

TylonHH, 2017-12-29:
I run into same issue because in the corexy sample was no baud setting. I added it and it works now.
suggestion: Shouldnt the specific printer example provide all setting?

KevinOConnor, 2017-12-29:
On Fri, Dec 29, 2017 at 10:15:56PM +0000, TylonHH wrote:
> I run into same issue because in the corexy sample was no baud setting. I added it and it works now.
> suggestion: Shouldnt the specific printer example provide all setting?

The default in the micro-controller config and in the host config is
250000.  That's the recommended baud rate.  Does that rate not work on
your hardware?

-Kevin