Klipper3d/klipper #73: PT100 with amplifierboard on RUMBA board

kimkejser, 2017-12-27:
Is it possible to use a PT100 with an E3D amplifier board, i couldn't find any description in the documentation here on GitHub.

KevinOConnor, 2017-12-27:
On Wed, Dec 27, 2017 at 11:18:03AM +0000, kimkejser wrote:
> Is it possible to use a PT100 with an E3D amplifier board, i couldn't find any description in the documentation here on GitHub.

It would need some changes to the code.  As an initial step you could
add the following to the Sensors table in klippy/heater.py:

"PT100 INA826": { 'class': Linear, 't1': 100., 'v1': 1.52, 't2': 300., 'v2': 2.29 },

Then one could declare a sensor of "PT100 INA826" in the config.
However, this will have some temperature inaccuracies at the high
temperature range (temp > 500).

The volts and temperatures are from the wiki page at:
https://wiki.e3d-online.com/E3D_PT100_Amplifier_Documentation#Output_Characteristics

-Kevin

kimkejser, 2017-12-27:
Ok, thank you for the fast reply. I'll give it a whirl tomorrow.
-K

kimkejser, 2018-01-03:
It works great. Tested it yesterday.
I just added the info to the Sensors table as you described. 
Thanks :-D

closed 2018-01-03 by kimkejser